libplctag/libplctag #94: PC Cpu OverLoad

rafaelkrause, 2019-03-30:
Hi Guys,

After a switch from version 1.5 to 2.0 (or even 1.6) I got pc cpu overload. The lib get from 15% to 25% of my cpu, even Whem I am not reading or Writing to PLC.

You guys have seem that in yours test?

I compiled the lib on VS 2015 and 2017.

As test, I changed the tag_rw.c to not Finish at end of program, just to check if is something with my program, but I got same results.

this what a I changed on ttag_rw.c test program

`    
    if(tag) {
        plc_tag_destroy(tag);
    }

    printf("press enter");

    printf("Time: %.3fms\n",(double)(clock() - count) / CLOCKS_PER_SEC);
    printf("Done\n");
    getchar();  
    return 0;`

rafaelkrause, 2019-03-30:
Just updating .... I downloaded another libtag.dll compiled from "mesta1/libplctag-csharp" ( It suppose to be the 2.0 version.) and re-run the test.
Now the cpu drop down to 1% to 2% (that's OK).
So i guess it's something with my compilation, but I really don't know what.

Some suggest ?

kyle-github, 2019-03-30:
Hmm, this does not sound good.   I have not seen problems with high CPU unless the client program is doing something bad.   Tag_rw should not be doing that.  Thanks for this report.   I would like to make sure that there is not a latent problem in the library!

Is the high CPU something that is repeatable?   Can you try downloading the source and recompiling the latest 2.0 version to see if you get the same thing?   If it is at all repeatable, then there may be something I am doing in the library that is a problem.

Are you compiling the library in DEBUG or RELEASE mode?   When you run CMake, you will automatically get... RELEASE mode.  I think.    I wonder if there is something happening depending on the mode?

I am traveling for work through next week, so I will not be in front of my machines with Windows VMs (I do all my work on the library on Linux) until after then.

rafaelkrause, 2019-04-10:
Hi Kyle, thanks for the awnser.

I did another tests with version 2, from a compiled dll and works fine.

I really does not know if is something on my vs configuration that generate this errors. I compiled using cmake in realease. I guess is something related to my VS configuration.

  Unfortunately i can another make more tests rigth now (busy in a machine comissioning) so, I will use the pre compiled dll. As soon as get free i will try to compile again.

Really thanks for the lib

Rafael

I

kyle-github, 2019-04-11:
I will try to compile it this weekend using the latest stable 2.0 and see if I can repeat the problem.

kyle-github, 2019-05-26:
I was never able to see this issue.   It sounds like something did not get compiled correctly.   However, I still need to figure out how that happened so that I can update the instructions!

kyle-github, 2019-06-02:
Can you still reproduce this with any of the later versions?

kyle-github, 2019-07-02:
Please reopen this if you see it again!

closed 2019-07-02 by kyle-github