libplctag/libplctag #96: Failed to read SubProgram variable tag

shouye0928, 2019-04-26:
I failed to read SubProgram variable tag,  SubProgram name is S030, variable name is z_TestDint[22],  and use S030:z_TestDint[22] as tag name, but failed to call plc_tag_create API function. Is this tag name correct?

shouye0928, 2019-04-27:
How to read or write subprogram variable?

kyle-github, 2019-04-27:
See issue #29.

You were very close.   You use it like this:

PROGRAM:S030.z_TestDint[22]

You need the word "PROGRAM" first, then a colon, ":", then the program name, "S030", then a dot, ".", then the tag name.

Please try that.

kyle-github, 2019-04-27:
Also note that you need to allow reads in the permissions!

shouye0928, 2019-04-28:
Kyle, thanks for the quick response, I will try it and post the results.

shouye0928, 2019-04-28:
Use PROGRAM:S030.z_TestDint[22], it works well, thanks.

kyle-github, 2019-05-26:
Closing if there are no other comments.

closed 2019-05-26 by kyle-github