libplctag/libplctag #97: Getting the list of tags from Logix controller

afiya-g, 2019-05-24:
Hello Kyle,

Does current version supports now to get the tag list from Logix PLC?
 I'm using libplctag lib for my client project, and the requirement is to get the AB controller tag list.
 If the current version is supporting, can you please let know how to do it.

Regards,
 Syed.

kyle-github, 2019-05-26:
No the current version does not support this yet.   It has been requested multiple times, so I probably should port my test code that was working on a different version of the library.

afiya-g, 2019-05-27:
Kyle,

Thanks much for the early reply, any tentative date you are planning to release the requested support.

Regards,
Syed

kyle-github, 2019-05-27:
Not at this time.   I have very limited to time to work on the library right now and I prioritize bug fixes over features at all times.

kyle-github, 2019-05-27:
The initial attempt at this was done in the new_ab branch.   However, that branch has a bunch of other non-working code in it (does not compile).   It was one of my failed attempts to add multi-request packets.

However, most of the tag listing code is probably still more or less valid.  I will see what I can do today (I have a few hours) to try to get it into the main branch.

The attached document shows how to do the tag listing process (and several other things).
[1756-pm020_-en-p.pdf](https://github.com/kyle-github/libplctag/files/3224079/1756-pm020_-en-p.pdf)

In the new_ab branch, you will need to start with the example program, list_tags.   

The problems with this branch are legion:

- I started modifying how the tag string worked in an ill-advised fashion.   Too many changes and it would have broken all existing code badly.
- I was trying to set up the multiple-request support, but in the wrong place in the code so it became very unwieldy.
- I was trying to move to a different locus of control in the threading and that was not completed, so it is kind of half-baked with remains of the old control flow.
- I was refactoring the code into specific PLC types directly (still needs to be done, but not as a single step).

In short I was doing too much at once.   With the current 2.0 code structure, this should be feasible to revisit and complete.

kyle-github, 2019-05-27:
I was able to get almost complete in the time I had today.   It almost works.   If you want to look at the code, it is on the "tag_list" branch.   I have some off-by-one errors and a few minor clean ups to get done before it is ready for more testing.

kyle-github, 2019-06-01:
I think this is almost complete.  I should finish this over the weekend!  I am testing on Windows now.

kyle-github, 2019-06-02:
This has been released in version 2.0.17.

closed 2019-06-02 by kyle-github