libplctag/libplctag #98: Reading mulitple tags in one go from the Logix controller

afiya-g, 2019-05-24:
In my project requirement, the multiple tags needs to be read from the controller in one go, the list of tags should go to the controller to be read, the controller should response back the request for the multiple tags. The other libraries supports this command using connected type.  I would like to know how to do it using libplctag.

kyle-github, 2019-05-26:
The library will do this automatically.  Please look at the "async" example.   The library will automatically pack multiple requests into one request packet to the PLC and unpack the results.   You do not need to make a list separately.  Just create all the tag elements as you would normally and then read them all at once using the asynchronous calls (with a timeout of 0).  

If you use different threads for each tag, then you may not get the same level of packing.   Once a request has been sent to the PLC, no more sub-requests can be added to it.  They will be collected into the next request packet.   

Also note that the library will attempt to negotiate a much larger packet size with the target PLC.   It can use up to 4002 bytes against a ControlLogix-class PLC.

The request packing is only implemented for ControlLogix/CompactLogix PLCs.

kyle-github, 2019-05-26:
Please reopen this if you have trouble!   Thanks for using the library.

closed 2019-05-26 by kyle-github