libplctag/libplctag #99: It is taking around 45ms to read the tag from Logix controller

afiya-g, 2019-05-24:
It is taking around 45ms to read the tag from Logix controller. 
Please suggest to improve the performance to read it within 12ms.

I'm using simulator to read the tag value, when I run the simulator on the same Windows 10 PC it could able to read within 12ms, however when I run the simulator on the other windows 10 PC it takes around 45ms to read.

kyle-github, 2019-05-26:
This sounds like a networking problem.  What is the network path between the PC running the library and the PC running the simulator?

Are you using a name for the remote PC or a bare IP address?

afiya-g, 2019-05-27:
Kyle,

Thanks much for your early reply.
Both PC's are running on Windows 10 OS and it is on peer-to-peer IP based address.

Regards,
Syed.

kyle-github, 2019-05-27:
Can you test ping times between the PCs?

Are you testing the repeated transaction time between the two PCs or is it just one shot?   I.e. if you run something like the tag_rw program, the time will include the TCP and other connection overhead and is not representative of the time required to do a transaction with the PLC.  If you use another test where you repeatedly read/write the PLC without closing the TCP connection, then the connection overhead will be lost in the noise.

TCP connection overhead can be fairly large.   The CIP overhead of establishing a connection is also fairly large (at least one back-and-forth packet handshake).

If you add the debug flag to the tag string, you will get output that shows some of the internal times.  Add "&debug=3" to the tag string.

I have a fairly bad network and I usually usually see response times around 7-11ms.   However, connection set up overhead can be very high.   If I am connecting to a PLC/5, it can be almost 200ms.  So if I read a tag once, I would see something like 220ms.  If I read it twice, I would see 240ms, etc.

kyle-github, 2019-06-02:
Any update on this?

fluisgirardi, 2019-07-04:
I have used the old libplctag under Linux with reads ~10ms. But the same program with same hardware (PC and PLC) on Windows takes ~30ms. I don't know why, but appears to be an OS issue..

fluisgirardi, 2019-07-04:
Maybe you are reading these tags on your simulator on 12ms because of the loopback interface, that is a software interface, and because of this, it's many times more fastest than a network card...

kyle-github, 2019-07-15:
I generally see around 10-15ms on a local 100Mbps/1Gbps LAN (not a great network) and lower if I am on a good 1Gbps talking to a modern L8x ControlLogix CPU.

fluisgirardi, 2019-08-15:
Hahaha @kyle-github using a cannon to kill a mosquito? The best PLC that I have tested is a CompactLogix L36. And with this PLC, I got ~10ms under Linux and ~30ms under Windows under a 100Mbps network. Maybe PC hardware interferences?

kyle-github, 2019-09-07:
I am going to close this for now.  Please reopen if you are still having problems.

closed 2019-09-07 by kyle-github