libplctag/libplctag #100: How to read and write slc505 string file?

petereagle, 2019-05-29:
Hi KyLe,
I want read few string tags from slc505. The value hold in these tags are:
ST9:0 = 2019_05_28 ,ST9:1 = 08:22:34 , ST9:2 = 2201 , . . . , ST9:10 = 125psi

With modified string.c , i only received empty string as shown [ string 0 ( - 2xxxxxx..... char) = '' ].
Please advice me how to rectify this issue, thank you.

kyle-github, 2019-05-30:
Can you please share your tag string?

What version of the library are you using?   Windows?  Linux?

petereagle, 2019-05-30:
Thanks for reply, Mr.Kyle
i used Version2.0 library on ubuntu 16.04 LTS in windows 10. 
Sorry, the "String tag" here is referring to slc cpu string file address , not logix platform cpu symbolic tag.   
  
Grab string.c file from your example make minor changes on Tag Path, 
and try read a first string tag address ST9:0

#include <stdio.h>
#include <stdlib.h>
#include "../lib/libplctag.h"
#include "utils.h"

#define TAG_PATH "protocol=ab_eip&gateway=192.168.111.100&cpu=SLC&elem_size=1&elem_count=1&name=ST09:0&debug=2"
#define ELEM_COUNT 1
#define ELEM_SIZE 1
#define DATA_TIMEOUT 5000

int main()
{
    int32_t tag = 0;
    int rc;
    int i;

    /* create the tag */
    tag = plc_tag_create(TAG_PATH, DATA_TIMEOUT);

    /* everything OK? */
    if(tag < 0) {
        fprintf(stderr,"ERROR: Could not create tag!\n");
        return 0;
    }

    if((rc = plc_tag_status(tag)) != PLCTAG_STATUS_OK) {
        fprintf(stderr,"Error setting up tag internal state. Error %s\n", plc_tag_decode_error(rc));
        plc_tag_destroy(tag);
        return 0;
    }

    /* get the data */
    rc = plc_tag_read(tag, DATA_TIMEOUT);
    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d: %s\n",rc, plc_tag_decode_error(rc));
        plc_tag_destroy(tag);
        return 0;
    }

    /* print out the data */
    for(i=0; i < ELEM_COUNT; i++) {
        int str_size = plc_tag_get_int32(tag,(i*ELEM_SIZE));
        char str[ELEM_SIZE] = {0};
        int j;

        for(j=0; j<str_size; j++) {
            str[j] = (char)plc_tag_get_uint8(tag,(i*ELEM_SIZE)+j+4);
        }
        str[j] = (char)0;

        printf("string %d (%d chars) = '%s'\n",i, str_size, str);
    }

    /* we are done */
    plc_tag_destroy(tag);

    return 0;
}
 
Hope information provided to you is correct.

kyle-github, 2019-05-31:
Can you change the debug=2 to debug=4 and capture the output?

Also, I am fairly sure that the elem_size=1 and elem_count=1 are incorrect.   The elem_size field must be at least 2 (and must be a multiple of 2 generally) for PLC/5, SLC and MicroLogix.

petereagle, 2019-05-31:
Okay, tag path change as shown and log file attached.
#define TAG_PATH "protocol=ab_eip&gateway=192.168.111.100&cpu=SLC&elem_size=2&elem_count=1&name=ST09:0&debug=4"
#define ELEM_COUNT 1
#define ELEM_SIZE 2
#define DATA_TIMEOUT 5000

[logfile.txt](https://github.com/kyle-github/libplctag/files/3239494/logfile.txt)

kyle-github, 2019-06-01:
Hi,

Interesting.   There isn't an error here.   Here's the return packet (edited for clarity):

> unpack_response:1378 Got single response packet.  Copying 57 bytes unchanged.
> unpack_response:1432 Unpacked packet:
> unpack_response:1433 00000 6f 00 21 00 00 14 02 01 00 00
> unpack_response:1433 00010 00 00 02 d3 07 13 00 00 00 00
> unpack_response:1433 00020 00 00 00 00 00 00 00 00 01 00
> unpack_response:1433 00030 02 00 00 00 00 00 b2 00 11 00
> unpack_response:1433 00040 cb 00 00 00 07 3d f3 45 43 50
> unpack_response:1433 00050 21 4f 00 00 d3 01 00
> unpack_response:1442 Done.
> plc_tag_read:734 elapsed time 55ms

The last bytes, `4f 00 00 d3 01 00`, are the response.   I'll break them down:
`4f` - this is the original function code, `0f` with the "complete" bit set.
`00` - status, zero means no error.
`00 d3` - sequence ID of this request, returned to us.
`01 00` - data.   This is probably the count of number of bytes of the string length?

So, something seems odd here.   You are only getting 2 bytes, but I think that is all you asked for.   Normally this would be the length of the string, but one byte seems low.  Is the string really short?

Try changing elem_count to 4 and see if we get part of the string.   So your elem_size should be 2 and your elem_count should be 4.   That should get 8 bytes of the string and perhaps we can determine what the internal format is from that.   The request is only asking for 2 bytes.

Different Allen-Bradley PLCs handle strings in different ways, so I wonder if we have a different format here.  I can only test on PLC/5 and I know that recently a user of the library found that MicroLogix did not support the exact same set of commands and returns responses differently.   In this case, it looks like the command to the PLC is working fine, and some data is being returned, but the data seems odd.

You can test different tag strings by using the tag_rw example.   Then you do not have to recompile each time you change something.  Use the -t param to set the data type to uint8.

petereagle, 2019-06-04:
Hi Mr.Kyle,
Thanks to your data bytes break down explanation. 
After tweaks around the program, i managed get data :
string 0 (82 chars) = '1032547698103254769810325476981032547698103254769810325476981032547698103254769810'
string 1 (26 chars) = 'BADCFEHGJILKNMPORQTSVUXWZY'

Which PLC string tag value hold :
ST9:0 = 0123456789012345678901234567890123456789012345678901234567890123456789012345678901
ST9:1 = ABCDEFGHIJKLMNOPQRSTUVWXYZ

Problem encountered.
1.   Compare string 0 and ST9:0 value, allocated value for high byte and low byte are wrong. Same 
      issue for string 1 and ST9:1
2.  elem_count can not set more than 2, else "  ERROR: Unable to read the data! Got error code -33: PLCTAG_ERR_TOO_LARGE "

Here attached program and debug file for better understanding. Let me know how to resolve
these issue. Many thanks in advance.
[Logger_500.c.txt](https://github.com/kyle-github/libplctag/files/3254430/Logger_500.c.txt)
[logfile_20190605.txt](https://github.com/kyle-github/libplctag/files/3254431/logfile_20190605.txt)

kyle-github, 2019-06-05:
How strange!  It looks like each pair of characters is swapped:

string: 01 23...
output: 10 32...

Could strings be stored as INT (16-bit) in big-endian format???

Try to invert the order in which you read each pair of bytes.   It looks like you will need to treat each pair of characters as swapped.

```c
for(j=0; j<str_size; j++) {
        if(j % 2 == 0) {
            str[j] = (char)plc_tag_get_uint8(tag,(i*ELEM_SIZE)+j+4+1);
        } else {
            str[j] = (char)plc_tag_get_uint8(tag,(i*ELEM_SIZE)+j+4);
        }    
        str[j] = (char)0;
}
```

Also try to see what happens when you have a string with an odd number of characters in it.

Have you been able to read other kinds of tags correctly?  I wonder if this byteswapping extends to other type.

This is a very interesting result!

kyle-github, 2019-06-05:
Also... I see that you are skipping the first four bytes.   Is that the string size word?   Some systems use a 2-byte count followed by up to 82 bytes of string data and others use a 4-byte count word followed by the string data.

Micro800-series PLCs seem to use a single count byte followed by up to 255 bytes of string data.

petereagle, 2019-06-05:
Hi Mr.Kyle,

> Also... I see that you are skipping the first four bytes. Is that the string size word?

Yes, you are right. They are string size words. Actually for slc500 platform, 2 bytes is enough
to hold the string size.

> Some systems use a 2-byte count followed by up to 82 bytes of string data and others use a 4-byte count word followed by the string data.

Reference to SLC500 Instruction Set Reference manual page 10-3. 
State that, the string lengths must be between 0 and 82 bytes for each element

With your code help, i modified it to skip that 4 bytes string size as shown:

        for(j=0; j<str_size; j++) {
                if(j % 2 == 0) {
                    str[j] = (char)plc_tag_get_uint8(tag,(i*ELEM_SIZE)+j+2+1);
                } else {
                    str[j] = (char)plc_tag_get_uint8(tag,(i*ELEM_SIZE)+j+1);
                }
        }
                str[j] = (char)0;
        

> Also try to see what happens when you have a string with an odd number of characters in it.
Have you been able to read other kinds of tags correctly? I wonder if this byte swapping extends to other type. 

From what i tried out, look like plctaglib was predefined the string data to 82 characters binding 
with string size 4 bytes for each element size. Therefore, on Tag path elem_size i set it to 84; no 
other value.
         
Regardless odd / even number of characters length, as long user string characters in PLC string 
element is not larger than 82. The plctaglib will work fine.

Here is the result i got.
string 0 (82 chars) = 
     '0123456789012345678901234567890123456789012345678901234567890123456789012345678901'

string 1 (82 chars) = 'ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz`~!@#$%^&*()_-=+{}[]|:;'",<.>/'
 
Unfortunately i can read 2 elem_count only successfully. It will error on "Unable to read the data! Got error code -33: PLCTAG_ERR_TOO_LARGE" if more count you request. 

Is there way to rectify this issue?   

Thank you Mr.Kyle

xinthose, 2019-06-05:
These are my working tested code written in c++:

```
std::vector<std::string> plctag::read_tag_str(int tag_num, int timeout, int element_size, int element_count)
{
	boost::mutex::scoped_lock lock(mutex); // prevent multiple threads

	int status = 0;
	std::vector<std::string> svec;
	bool debug = false;

	if (debug)
		INFO << "started";

	try
	{
		// 1. read data
		status = plc_tag_read(tag.at(tag_num), timeout); // timeout (ms)
		if (status != PLCTAG_STATUS_OK)
		{
			ERROR << "tag.at(" << tag_num << ") >> plc_tag_read >> error = " << plc_tag_decode_error(status);
			throw - 2;
		}
		// 2. format data
		for (int i = 0; i < element_count; i++)
		{
			/// method 1
			int str_size = plc_tag_get_int32(tag.at(tag_num), (i * element_size));
			status = plc_tag_status(tag.at(tag_num));
			if (status != PLCTAG_STATUS_OK)
			{
				ERROR << "tag.at(" << tag_num << ") >> plc_tag_get_int32 >> error = " << plc_tag_decode_error(status);
				throw - 3;
			}

			char char_str[83] = {0};
			int j = 0;

			for (j = 0; j < str_size; j++)
			{
				char_str[j] = (char)plc_tag_get_uint8(tag.at(tag_num), ((i * element_size) + j + 4));
				status = plc_tag_status(tag.at(tag_num));
				if (status != PLCTAG_STATUS_OK)
				{
					ERROR << "tag.at(" << tag_num << ") >> plc_tag_get_uint8 >> error = " << plc_tag_decode_error(status);
					throw - 4;
				}
			}
			char_str[j] = (char)0;	 // null terminate char array
			std::string str(char_str); // convert char array to string
			svec.push_back(str);

			/// method 2
			/*char str_data[str_data_size];
			int num_strings = plc_tag_get_size(tag.at(tag_num)) / element_size;
			int str_index;

			for(int i = 0; i < num_strings; i++)
			{
				// get the string length
				int str_len = plc_tag_get_int32(tag.at(tag_num), i * element_size);

				// copy the data
				for (str_index = 0; str_index<str_len; str_index++)
				{
					str_data[str_index] = (char) plc_tag_get_uint8(tag.at(tag_num), ((i * element_size) + 4 + str_index));
				}

				// pad with zeros
				for (; str_index < str_data_size; str_index++)
				{
					str_data[str_index] = 0;
				}
				std::string str(str_data);	// convert char array to string
				svec.push_back(str);
			}*/
		}
	}
	catch (const std::exception &e)
	{
		ERROR << "e.what() = " << e.what();
		throw - 1;
	}

	if (debug)
		INFO << "ended";

	return svec;
}

void plctag::write_tag_str(int tag_num, int timeout, int element_size, std::vector<std::string> svec)
{
	boost::mutex::scoped_lock lock(mutex); // prevent multiple threads

	int status = 0;
	bool debug = false;

	if (debug)
		INFO << "started";

	try
	{
		// 1. set the data
		for (size_t i = 0; i < svec.size(); i++)
		{
			int str_len = 0, str_index = 0;
			int base_offset = i * element_size;
			char *str = (char *)svec.at(i).c_str();

			// a.  set the length
			str_len = strlen(str);
			if (debug)
				INFO << "base_offset = " << base_offset << "; str_len = " << str_len;
			status = plc_tag_set_int32(tag.at(tag_num), base_offset, str_len);
			if (status != PLCTAG_STATUS_OK)
			{
				ERROR << "tag.at(" << tag_num << ") >> plc_tag_set_int32 >> error = " << plc_tag_decode_error(status);
				throw - 2;
			}

			// b.  copy the data
			for (str_index = 0; str_index < str_len; str_index++)
			{
				status = plc_tag_set_uint8(tag.at(tag_num), (base_offset + 4 + str_index), str[str_index]);
				if (status != PLCTAG_STATUS_OK)
				{
					ERROR << "tag.at(" << tag_num << ") >> plc_tag_set_uint8 >> error = " << plc_tag_decode_error(status);
					throw - 3;
				}
			}

			// c.  pad with zeros
			for (; str_index < str_data_size; str_index++)
			{
				status = plc_tag_set_uint8(tag.at(tag_num), (base_offset + 4 + str_index), 0);
				if (status != PLCTAG_STATUS_OK)
				{
					ERROR << "tag.at(" << tag_num << ") >> plc_tag_set_uint8 >> error = " << plc_tag_decode_error(status);
					throw - 4;
				}
			}
		}

		// 2. write the data
		status = plc_tag_write(tag.at(tag_num), timeout); // timeout (ms)
		if (status != PLCTAG_STATUS_OK)
		{
			ERROR << "tag.at(" << tag_num << ") >> plc_tag_write >> error = " << plc_tag_decode_error(status);
			throw - 5;
		}
	}
	catch (const std::exception &e)
	{
		ERROR << "e.what() = " << e.what();
		throw - 1;
	}

	if (debug)
		INFO << "ended";

	return;
}
```

kyle-github, 2019-06-06:
Thanks, @xinthose!

I missed the question earlier why only two strings could be read.   The answer is that this uses the PCCC protocol and its maximum packet size is about 250 bytes.   With the header and other overhead, that only leaves enough room for two strings.

There may be a way to use multiple packets on the backend to get entire tags with PCCC, but it is a lot more difficult (and seems to be PLC-specific) than with the CIP protocol that ControlLogix and CompactLogix PLCs use.

Now that I have separated out the different PCCC-speaking PLC types, I may be able to work on this.

petereagle, 2019-06-06:
Hi Mr.Kyle,
Appreciate, if you can sort out PCCC protocol to have multiple tags read/write for slc platform.

Hello Mr.xinthose 
Thanks to you too.

kyle-github, 2019-06-06:
It will be difficult to do.   I do not have a SLC 500 or a MicroLogix PLC.  I only have a PLC/5.

petereagle, 2019-06-11:
Hi Mr.Kyle,
To work around with 2 string tags I packed all PLC data into long string with comma separator. 
Perform data parse once received in string tags.
It's work!!!

petereagle, 2019-06-13:
Hi Mr.Kyle,
Thanks for your support and afford.

closed 2019-06-13 by petereagle