libplctag/libplctag #101: Tag listing not working on Compact Guard Logix

ManuelJm, 2019-07-05:
HI,
I've tried the tag listing example on a compact guardlogix and i always get a time out error (-32)
I've increased the time out up to 15minutes but still no answer to get the listing from the plc.

Tracked the code a bit and it seems it never reaches the tag reading part in the internals of the library.

Using the simple example to get a single tag works ok.

kyle-github, 2019-07-05:
Thanks for using the library!

Can you please add &debug=4 to the tag string and capture the output?  This will provide more information for debugging.

I wonder if the GuardLogix protects the tags from being listed?

Hopefully I can tell more about the problem once I see the output.  Thanks for reporting this!

ManuelJm, 2019-07-07:
Hi Kyle, I'll test the debug option and let you know what the output is.
At least the individual tag query works so seems that for one tag query nothing is protected.

ManuelJm, 2019-07-08:
Hi, Kyle this is the output with the debug option.

D:\libplctag\bin\Release>list_tags 192.168.1.130 1,0
Using tag string: protocol=ab-eip&gateway=192.168.1.130&path=1,0&cpu=lgx&name=@tags&debug=4
2388-06-07 08:09:53.945 thread(1) tag(0) INFO find_tag_create_func:82 Matched protocol=ab-eip
2388-06-07 08:09:53.946 thread(2) tag(0) INFO tag_tickler_func:145 Starting.
2388-06-07 08:09:53.950 thread(1) tag(0) INFO ab_tag_create:142 Starting.
2388-06-07 08:09:53.954 thread(1) tag(0) INFO rc_alloc_impl:99 Starting, called from ab_tag_create:149
2388-06-07 08:09:53.957 thread(1) tag(0) INFO rc_alloc_impl:118 Done
2388-06-07 08:09:53.959 thread(1) tag(0) DETAIL rc_alloc_impl:123 Returning memory pointer 000002642EE84010
2388-06-07 08:09:53.964 thread(1) tag(0) DETAIL ab_tag_create:157 tag=000002642EE84010
2388-06-07 08:09:53.967 thread(1) tag(0) DETAIL ab_tag_create:213 Setting up Logix tag.
2388-06-07 08:09:53.969 thread(1) tag(0) DETAIL get_tag_data_type:480 Done.
2388-06-07 08:09:53.972 thread(1) tag(0) DETAIL session_find_or_create:211 Starting
2388-06-07 08:09:53.975 thread(1) tag(0) DETAIL session_find_or_create:235 Creating new session.
2388-06-07 08:09:53.977 thread(1) tag(0) INFO session_create_unsafe:478 Starting
2388-06-07 08:09:53.980 thread(1) tag(0) DETAIL session_create_unsafe:480 Warning: not using passed port 44818
2388-06-07 08:09:53.984 thread(1) tag(0) INFO rc_alloc_impl:99 Starting, called from session_create_unsafe:482
2388-06-07 08:09:53.987 thread(1) tag(0) INFO rc_alloc_impl:118 Done
2388-06-07 08:09:53.989 thread(1) tag(0) DETAIL rc_alloc_impl:123 Returning memory pointer 000002642EE84230
2388-06-07 08:09:53.994 thread(1) tag(0) INFO match_dhp_node:112 Bad syntax in DH+ route.  Expected DH+ channel identifier (A/2 or B/3)
2388-06-07 08:09:53.998 thread(1) tag(0) DETAIL cip_encode_path:216 Found regular routing. Conn path length=1
2388-06-07 08:09:54.002 thread(1) tag(0) INFO match_dhp_node:112 Bad syntax in DH+ route.  Expected DH+ channel identifier (A/2 or B/3)
2388-06-07 08:09:54.006 thread(1) tag(0) DETAIL cip_encode_path:216 Found regular routing. Conn path length=2
2388-06-07 08:09:54.010 thread(1) tag(0) DETAIL cip_encode_path:291 ioi_size before 6
2388-06-07 08:09:54.012 thread(1) tag(0) INFO cip_encode_path:308 Done.
2388-06-07 08:09:54.014 thread(1) tag(0) DETAIL add_session_unsafe:319 Starting
2388-06-07 08:09:54.017 thread(1) tag(0) DETAIL add_session_unsafe:336 Done
2388-06-07 08:09:54.019 thread(1) tag(0) INFO session_create_unsafe:581 Done
2388-06-07 08:09:54.022 thread(1) tag(0) INFO session_init:597 Starting.
2388-06-07 08:09:54.024 thread(1) tag(0) DETAIL thread_create:609 Starting.
2388-06-07 08:09:54.026 thread(1) tag(0) DETAIL thread_create:643 Done.
2388-06-07 08:09:54.026 thread(3) tag(0) INFO session_handler:960 Starting thread for session 000002642EE84230
2388-06-07 08:09:54.028 thread(1) tag(0) INFO session_init:611 Done.
2388-06-07 08:09:54.032 thread(3) tag(0) DETAIL session_handler:967 in SESSION_OPEN_SOCKET state.
2388-06-07 08:09:54.035 thread(1) tag(0) DETAIL session_find_or_create:281 Done
2388-06-07 08:09:54.037 thread(3) tag(0) INFO session_open_socket:626 Starting.
2388-06-07 08:09:54.040 thread(1) tag(0) DETAIL ab_tag_create:299 using session=000002642EE84230
2388-06-07 08:09:54.043 thread(3) tag(0) DETAIL socket_create:832 Starting.
2388-06-07 08:09:54.045 thread(1) tag(0) INFO ab_tag_create:314 Done.
2388-06-07 08:09:54.048 thread(3) tag(0) DETAIL socket_create:851 Done.
2388-06-07 08:09:54.050 thread(1) tag(0) INFO plc_tag_create:458 tag set up elapsed time 0ms
2388-06-07 08:09:54.052 thread(3) tag(0) DETAIL socket_connect_tcp:871 Starting.
2388-06-07 08:09:54.055 thread(1) tag(0) DETAIL add_tag_lookup:1818 Starting.
2388-06-07 08:09:54.059 thread(1) tag(0) DETAIL add_tag_lookup:1838 Found unused ID 11
2388-06-07 08:09:54.062 thread(1) tag(0) DETAIL add_tag_lookup:1858 Done.
2388-06-07 08:09:54.057 thread(3) tag(0) DETAIL socket_connect_tcp:920 Found numeric IP address: 192.168.1.130
2388-06-07 08:09:54.064 thread(1) tag(11) INFO plc_tag_create:476 Returning mapped tag ID 11
2388-06-07 08:09:54.069 thread(3) tag(0) DETAIL socket_connect_tcp:1008 Done.
2388-06-07 08:09:54.070 thread(1) tag(11) INFO plc_tag_create:478 Done.
2388-06-07 08:09:54.075 thread(1) tag(11) INFO plc_tag_read:662 Starting.
2388-06-07 08:09:54.077 thread(1) tag(11) INFO tag_read_start:202 Starting
2388-06-07 08:09:54.073 thread(3) tag(0) INFO session_open_socket:643 Done.
2388-06-07 08:09:54.079 thread(1) tag(11) INFO build_tag_list_request_connected:394 Starting.
2388-06-07 08:09:54.082 thread(3) tag(0) DETAIL session_handler:986 in SESSION_REGISTER state.
2388-06-07 08:09:54.088 thread(3) tag(0) INFO session_register:656 Starting.
2388-06-07 08:09:54.086 thread(1) tag(11) DETAIL session_create_request:2347 Starting.
2388-06-07 08:09:54.091 thread(3) tag(0) DETAIL send_eip_request:1685 Starting.
2388-06-07 08:09:54.096 thread(3) tag(0) DETAIL send_eip_request:1698 Sending packet of size 28
2388-06-07 08:09:54.099 thread(3) tag(0) DETAIL send_eip_request:1699 00000 65 00 04 00 00 00 00 00 00 00
2388-06-07 08:09:54.099 thread(3) tag(0) DETAIL send_eip_request:1699 00010 00 00 00 00 00 00 00 00 00 00
2388-06-07 08:09:54.099 thread(3) tag(0) DETAIL send_eip_request:1699 00020 00 00 00 00 01 00 00 00
2388-06-07 08:09:54.094 thread(1) tag(11) INFO rc_alloc_impl:99 Starting, called from session_create_request:2349
[... cut ...]
2388-06-07 08:09:59.416 thread(3) tag(0) DETAIL session_handler:1065 in SESSION_UNREGISTER state.
2388-06-07 08:09:59.418 thread(3) tag(0) INFO session_unregister:733 Starting.
2388-06-07 08:09:59.421 thread(3) tag(0) INFO session_unregister:737 Done.
2388-06-07 08:09:59.423 thread(3) tag(0) DETAIL session_handler:1075 in SESSION_CLOSE_SOCKET state.
2388-06-07 08:09:59.426 thread(3) tag(0) INFO session_close_socket:746 Starting.
2388-06-07 08:09:59.428 thread(3) tag(0) INFO session_close_socket:754 Done.

But no tags are listed on the command line.

kyle-github, 2019-07-09:
Thanks for the logs!   This definitely helps.   I think I see the base of the problem, but not quite how to fix it yet.   This is a bug.   There are at least two problems:

1) the error of the result size being larger than expected is not getting propagated back to the client (the user of the library).   So you end up timing out.

2) (this is the real problem) the buffer size allocated is much smaller than the result.   The code checks for this case, so no damage is done like a crash.   But it is not correctly sizing the result buffer.

I will be working on a fix.  Thank you for finding this!

kyle-github, 2019-07-09:
I think I have fixed this.  Please try version 2.0.18.

ManuelJm, 2019-07-09:
Hi Kyle, now it seems worked now i get the list of tags.
Thanks a lot.
By the way tag listing is supported only for logix cpus?

kyle-github, 2019-07-10:
Currently there is only support for Logix (ControlLogix and CompactLogix) CPUs.   It may be possible to list tags for older types of PLCs (PLC/5, SLC 500 etc.) but I have not seen how to do it.   The Micro8xx series also may not support tag listing, but I do not have one and cannot check.

Thanks again for the bug report!

closed 2019-07-10 by kyle-github