libplctag/libplctag #102: it is possible use this on ESP or Arduino?

MrTora, 2019-07-11:
at least read or write 1 variable from arduino it is posible or need to adapt the lib for arduino libs?

kyle-github, 2019-07-14:
I'd like to get a port done but right now I suspect that the biggest problem is going to be memory use.   I allocate multiple 4K buffers (one per tag and one per PLC).   That adds up quickly.

I am not sure that Arduino would ever work unless there is significantly more memory.   ESP might work.   Some of those chips have a few MB of RAM.

The second problem is that the library uses threading internally.   The protocol is naturally very asynchronous, but also very stateful.  Platforms without some form of threading would be difficult.   It would be less of a port and more of a rewrite from scratch.

Hopefully I will get some time to experiment with some of these smaller systems.  

I would be happy to look at any patches you might have!

kyle-github, 2019-07-14:
Note that this is a duplicate of #68.

closed 2019-07-14 by kyle-github

trevm8, 2019-07-17:
I am also extremely interested in using this library with the ESP-32, This would open up a whole host of low cost automation products.

MrTora, 2019-07-17:
yes!, i'm gonna test this on raspberry pi, maybe using the raspberry as a gateway from Ethernet/IP to Ethernet WIFI ESP.

trevm8, 2019-07-17:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/kyle-github/libplctag/issues/102?email_source=notifications\u0026email_token=AMDTMKYSJ4ANQLKM7FL2DXDP753ORA5CNFSM4IBWFDPKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD2GNXZI#issuecomment-512547813",
> "url": "https://github.com/kyle-github/libplctag/issues/102?email_source=notifications\u0026email_token=AMDTMKYSJ4ANQLKM7FL2DXDP753ORA5CNFSM4IBWFDPKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD2GNXZI#issuecomment-512547813",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

I used a python based library last year and had tags read and write to a compactlogix so that is already possible. Taking the raspberry pi out if the equation is a much better prospect as a raspberry pi with SD card has a short operational life in comparisonSent from Samsung Mobile on O2

kyle-github, 2019-07-18:
I have run the library on an RPi 2B a while ago under Raspbian.  It worked fine.   What operating systems are available for the ESP32?

On the other issue someone proposed NuttX and I am also looking at MyNewt.

kyle-github, 2019-07-18:
Please continue discussion on issue #68.