libplctag/libplctag #105: PLC IO Tree

Jskom, 2019-08-12:
A coworker and I are able to read and write tags directly from the PC.  We do not even have anything setup in the IO tree to make this work.  Now, it seems as though the PC is setup as a PLC.  I am trying to see if I can add an ENBT card to the PLC IO tree (Slot 1) and then add a PLC to that in slot 0.  There are some parameters like ENBT version which I can only select "A".  Also, revision which I have as the default 5 but disabled keying.  We are trying to set it up so that the PLC can read some tags that reside in the PC.  Chassis size and rack connection are also parameters i am asked to set.  I guess my question is what does this look like on the PLC, IO Configuration Tree side?

Thanks

acidtonic, 2019-08-12:
Basically to summarize we used tag writing functionality from the PC as a client...

Now we wish to have the PC serve up tags as a server, with an AB Logix PLC configured to connect to it via an IO Tree entry to browse/read/edit the tags offered from the PC.

Do you know how that configuration would look on the AB side?

kyle-github, 2019-08-14:
The library does not support being used as a server.   It is only a client.   

You could try the test program lgx_sim in the tests directory.   It is very primitive and I use it to test the library when I do not have access to a PLC.

If you can use Python, there is a library, cccpo, that supports both client and server use.   I have used it in the past to do tests.   It is quite good.

As to how to configure the PLC to talk to cccpo, I am not sure.   We use direct explicit messages or produced/consumed tags and those are fairly straightforward via message instructions.   If your PLC already has a network module (I strongly suggest using an EN2T instead of an ENBT), then I think you can just set up a message instruction.

Jskom, 2019-08-15:
Thanks for the quick response.  In either way I will need to configure a device in the PLCs hardware tree that is representative of the PC.  If I do it through a 1756-EN2T card then I have to setup the PC like as a PLC.  If I setup the PC as a generic ethernet module or bridge then I have to setup using assembly instances.  I don't think that is the route you were heading with this.  Any chance you can take a screen shot of the PLC side message instruction setup and IO tree?  I would need to know the message type and service type. Also, you have to point to something in the hardware tree.  Do you setup the PC as a generic ethernet module or PLC.  If PLC, which PLC and version.  Any help with all of this would be great.

kyle-github, 2019-08-16:
Unfortunately, I have never set up the PLC to poll from the PC.  Only the other way around.  I know that the library will not pretend to be a PLC, so what are you trying to aim the real PLC at?

With produced/consumed tags, we did something similar to what you describe by adding the other PLC into the IO tree.   But I think that requires the whole CIP protocol running on both ends to allow you do easily link up tags.

I found the following link:

https://plcgurus.net/producer-consumer-messaging-explicit/

It describes how to set up both produced/consumed tags and explicit messaging (which is what the library uses).

kyle-github, 2019-09-07:
Have you had any luck with this?

kyle-github, 2019-10-05:
Closing for now.  Please reopen this if you get another chance to test!

closed 2019-10-05 by kyle-github