libplctag/libplctag #107: Read Groups and Threads

thevinge, 2019-08-19:
Hi @kyle-github ,

I noticed that you mentioned something about a "readgroup" in one of the other questions in here. I was wondering, if you had an example of this?
Furthermore, is it possible to use threads with your library? I have tried to use multiple threads to call plc_tag_read, however, doing so, only makes the read time slower. Do you have an example of how to read the plc tags as fast as possible? (I have already looked at the async example, that comes with the library, but it is only using one thread)

xinthose, 2019-08-19:
You could look at my C++ wrapper.  I use `boost::mutex::scoped_lock lock(mutex);` for my `plctag` operations.  Create threads from `int main` to perform operations on the tags created in `initialize_tags`.

thevinge, 2019-08-20:
I will take a look at your wrapper. @xinthose, have you come across a "readGroup" before? I will try to find the question, where it was mentioned in the first place.

xinthose, 2019-08-20:
@thevinge No, I have not.  That sounds useful.  My use case is not SCADA or data harvesting, but application specific (i.e. close/open values, read flow, etc.).

kyle-github, 2019-08-20:
Hi, sorry for the delay.   Thanks for using the library!

TL;DR: using async mode tends to give slightly better performance with smaller numbers of tags.

The "readgroup" idea was one that I was looking at for a while to solve the problem of bundling requests into one data packet to the PLC.  I ended up not using it as it provided much more complexity for the internals of the library without any obvious benefit.   I implemented it in multiple ways and was not happy with the results.

More advanced Allen-Bradley PLCs (ControlLogix and CompactLogix) support bundling multiple requests (read and write) together into one packet to the PLC and one response from the PLC.  Since the round trip packet time between a PC and the PLC tends to be around 10ms or so, bundling has a huge impact on performance.

Older versions of the library used an undocumented "feature" of these PLCs by queuing up multiple requests in the PLCs' internal network buffers.   However, it seems that the buffer size is not consistent across multiple types of network module (ENBT, EN2T, built-in L80 series etc.) and some CIP-based PLCs like the Micro800 series do not even support unconnected mode.   

Using this feature was best done using a lot of threads to queue the results.   There was a fair amount of internal complexity in the library to do automatic retries of requests that got dropped and there were still edge cases that did not work right.  It also interfered with attempts to make the library auto-disconnect and auto-reconnect (which it does now).

I found a new public (finally!) document from AB that showed how to bundle requests and the 2.0 version of the library does this automatically when talking to a PLC that supports it (so far only ControlLogix and CompactLogix).   I replaced the informal buffer queuing approach with this approach.

All you need to do is use tags normally.   The library will **automatically** bundle the requests and separate out the responses internally.  Nothing is visible at the user level except much increased throughput and lower average response times when you use a lot of tags.   There is nothing you need to change in the tag string.   Combined with an earlier change where the library negotiates with the PLC to support a larger packet size (4000 bytes instead of 500 bytes), this allows you to read dozens of tags in one request-response interaction with the PLC.

You can use multiple threads or async operations within one or a small number of threads.  I find that the async methods uses fewer resources but we use both in production.   Performance seems to be slightly better with the async method as that tends to do things in batches.

Internally, the library has a thread running for each PLC it is talking to.  That thread looks at its queue of requests and bundles all that it can into one request to the PLC.   By using async mode, the batches of requests tend to all come at once and the library has a little more to work with.  Using threads tends to spread out the requests and then they may not get packed as much.

Let me explain a bit more.   The library will attempt to pack all outstanding requests into one bundle if it can.  Suppose that you have four threads reading tags.   Each thread sends requests every 100ms.   What you can see is:

```
Thread 1 -> queue tag1 request.
Thread 2 -> queue tag2 request.
PLC thread -> bundle tag1 and tag2 requests and send bundle1 to PLC.
... wait 3 ms while other threads run ...
Thread 3 -> queue tag3 request.
Thread 4 -> queue tag4 request.
... wait for ~7ms for the response to come back ...
PLC thread -> unpack response1 from the PLC.
PLC thread -> bundle tag3 and tag4 requests and send bundle2 to PLC.
Thread 1 -> notice that there is a response and process it.  Set status.
Thread 2 -> notice that there is a response and process it.  Set status.
... wait ~10ms ...
PLC thread -> unpack response2 from the PLC.
Thread 3 -> notice that there is a response and process it.  Set status.
Thread 4 -> notice that there is a response and process it.  Set status.
```

If you have a lot of tags (dozens) and threads, generally you will get pretty good packing.  If you do not have that many tags, I see the above pattern a lot.

If you use the async mode, things tend to work out differently:

```
Thread 1 -> queue tag1 request.  Queue tag2 request.  Queue tag3 request.  Queue tag4 request.
PLC thread -> bundle tag requests and send bundle1 to PLC.
... wait for ~10ms for the response to come back ...
PLC thread -> unpack response1 from the PLC.
Thread 1 -> notice that there are response and process them.  Set status on tags.
```

It is not perfect.   You can have Thread 1 above get preempted at some point before it queues all the requests.   The window is much smaller than in the multi-threaded case.   At least in practice I see a noticeable difference in average read latency.  It is not a huge difference, but I usually can measure it.

The async.c example shows this in operation by doing the operations against many tags in a sort of batch mode.  It creates them all in a batch.  Then waits for the creates to succeed.  Then it triggers a read on all the tags in a batch.  Then waits for them to succeed.  Etc.

You can use threads and the library supports this directly in two ways:
1. each API call is internally thread safe.
2. the library provides plc_tag_lock() and plc_tag_unlock() API calls to lock an internal mutex.

[... cut ...]
For best performance, I now suggest using a minimal number of threads and using the async calls (using a timeout value of 0).   The async and data logger examples show this.   

Generally the only reason to use threading now is if it makes your code cleaner or simpler.  The library is very much IO-bound.   Even under fairly heavy load (reading 300 tags) it generally uses less than 1-2% of the CPU on my systems.   I have run the library on a Raspberry PI and the load there is also quite low.

I hope this helps!

kyle-github, 2019-09-01:
Can I close this?

thevinge, 2019-09-04:
> 
> 
> Can I close this?

Hi Kyle, sorry for my absence. Yes, you can close the issue now.

closed 2019-09-04 by thevinge