libplctag/libplctag #108: Downloadable Builds

timyhac, 2019-08-31:
Hi Kyle - would you consider supplying downloadable builds for this project?
I see that a few people have mentioned it in passing, but not directly (#89 , #76 , #74 ).

kyle-github, 2019-08-31:
Hi, yes this has been brought up.   I need to figure out how integrate a CI system that will build these automatically.   It is clearly much requested!

timyhac, 2019-08-31:
Great! - I would love to help out here and was doing a little research on how this could work.

[Azure Pipelines has a free tier](https://azure.microsoft.com/en-au/blog/announcing-azure-pipelines-with-unlimited-ci-cd-minutes-for-open-source/#targetText=Starting%20today%2C%20Azure%20Pipelines%20provides,open%20source%20project%20for%20free.&targetText=Many%20of%20the%20top%20open,list%20is%20growing%20every%20day.) for open source projects, with integrations into Github. I will try to get a linux x86 build working using this and report back to see if you are interested in pursuing this channel. If I haven't reported back in 1 week then I have given up or forgotten (don't want to leave you hanging).

kyle-github, 2019-09-01:
Ooh, nice.  I was starting to look at GitHub's built in CI system, "GitHub Actions."  Not sure I have a clue yet.   I signed up for the Beta test.

timyhac, 2019-09-01:
Turns out this is quite easy.

1/ Connect the repo to Azure pipelines: https://github.com/marketplace/azure-pipelines
2/ Set up a build pipeline
3/ Edit the pipeline script to be something like this: https://github.com/timyhac/libplctag/blob/master/azure-pipelines.yml
This is based on your build notes: https://github.com/kyle-github/libplctag/blob/master/BUILD.md

I didn't go any further because I wasn't sure how you want to package the binaries, but if you download the "build" artifact, they are all there. I didn't test them, but I assume they work because there were no errors.

I'll try to get a windows build pipeline working too.

timyhac, 2019-09-01:
Yep the windows one works as well: https://github.com/timyhac/libplctag/blob/master/azure-pipelines-1.yml

I didn't actually test the binary built on the linux build pipeline, but I did just now test the windows dll and it works!

kyle-github, 2019-09-02:
Wow! Thanks! I will have to read through this.  I signed up for the Beta test of GitHub's "Actions" which seem to be linked to Azure (not surprising as Microsoft owns GitHub now).

kyle-github, 2019-09-07:
Thanks!  I was able to create an Azure pipeline and make it build.   But I need to figure out how to connect that back to GitHub so that the build artifacts are downloadable.

timyhac, 2019-09-08:
Awesome Kyle - I've just had a play with the "Releases" tab, and it doesn't seem as straightforward as the builds what its doing (I like the idea of having a yaml file that defines the pipeline), but it was easy enough to get going. This is the result:
https://github.com/timyhac/libplctag/releases/tag/v1234-test

From the build pipeline, click Release
[image]

It will create a Stage for you, leave it as Empty Job.
[image]

Once you have done that, you have to create a task associated with Stage 1 and add a "GitHub release" task.
[image]
[image]

There is some info [here](https://docs.microsoft.com/en-us/azure/devops/pipelines/tasks/utility/github-release?view=azure-devops) on this task.

To get it running I had to override the Tag source, but I think you would probably want to leave on Git tag, that way it will use the version tag on the git repo, and I'm pretty sure it won't make a release unless there is a tag of some sort.

If there are automated tests that can be done without connecting to a PLC you would put a stage in for that.

----

So cool!

kyle-github, 2019-09-08:
Interesting.   I am following along, but much slower than you :-)

I got two build pipelines to run, building both Ubuntu and Windows.   I am trying to figure out how to make the artifacts show up in GitHub.   So far, no luck.  I was able to get the build status showing on the README page though.

Thanks for pushing me in this direction.  This is going to be helpful for many users if I can get the artifacts figured out.

kyle-github, 2019-09-08:
I have been fiddling with the CMake configuration to make things a bit cleaner.   Right now, the binaries were all dropped in a big pile in the middle of the build directory along with everything else.   In trying to clean up the archive parts, I decided to make the build process work a little more cleanly.   I have this working on Ubuntu and now I will be working on Windows.

kyle-github, 2019-09-09:
Thanks to all your help, I was able to get this more or less working.   I still do not have the artifacts get attached to the GitHub release automatically, but they are built correctly.   Mostly. 

Right now I think the biggest problem may be CMake.   I cannot figure out how to clean up the artifacts that are built.   I think I might need to filter these out a lot more.   That might be possible in Azure Pipelines too.   Not sure yet.

I use GitHub to do the releases right now.   So I am not doing native Git tags.   However, I think I will start to look into that, then I can set up a release pipeline in Azure.

This also gives me access to a macOS build machine.   Not directly, but it might be close enough to get the library working on a Mac.

Again, thank you so much for suggesting this!

timyhac, 2019-09-09:
Awesome! Yes I had some issues getting the "Github release" task working but managed to eventually after just trying a whole bunch of different things with no particular direction. I'm hoping that the pipelines will eventually become a little more intuitive.

In the publish task I'm pretty sure you can publish only the specific binaries you want. I guess it depends on how you want to do it, clean it up with CMake, or pick out the binaries with pipelines. Maybe if you make one publish task per binary that you want to publish?

Yeah having access to a macOS machine (for free) is pretty neat - can't believe they're giving it away!

timyhac, 2019-09-09:
P.S. this isn't selfless at all - I'm putting together a cross-platform [.NET wrapper](https://github.com/timyhac/libplctag.NET), so having the binaries published by you was my true end-game 😈

timyhac, 2019-09-14:
Heya @kyle-github - the releases are looking great! Thanks for doing this.

Let me know if I can help out with the github release task, otherwise we can consider this one closed?

kyle-github, 2019-09-20:
There are two .Net wrappers.  Mesta Automation has one.   You might want to see if one of those works well, even as a starting point.

kyle-github, 2019-09-20:
I have to look at releases more closely.  I use GitHub's tagging rather than "pure" Git, so my work flow is different than Azure Pipelines expects.   It still might be feasible.

I'll close this one.   I need to think about whether I want an automated tagging system or not.   I really want it to attach the build artifacts, so I will experiment as I get time (I have a huge project right now that is going to keep my busy for at least the next six months or so).

closed 2019-09-20 by kyle-github