libplctag/libplctag #110: Any plans to support protocol=ab_vbp  ?

dirtyb15, 2019-09-11:
Just curious, we use rsemulate5000 to test.
Thanks!

kyle-github, 2019-09-12:
Hi, thanks for the interest in the library!

I am not sure what vbp is :-(  Do you have some docs?   I know that some people have managed to get the emulator (or maybe it was SoftLogix) working with the library.   But the AB software set up is tricky.   Is that what you are trying to do?

kyle-github, 2019-10-05:
Since you so kindly created the LabView wrappers, can I close this now?  I have searched and not found any references to the back plane protocol.

Note that the actual hardware apparently uses some form of ControlNet or DeviceNet (CIP over a non-Ethernet bus) on the back plane.   At least that is what I see speculated on the Internet!

dirtyb15, 2019-10-07:
Yes, please close.
thanks!
Dave

closed 2019-10-07 by dirtyb15