libplctag/libplctag #111: Disabling automatic disconnect and recoonect feature released as part of v2.0.10

AravindPavate, 2019-09-20:
Hi All, Is there any way to disable the automatic disconnect and reconnect feature as we are facing issue when there is delay in communication. Also it would help if there is way to report the connection status after auto disconnect to the wrapper level and if there is way to increase the idle timeout from 5 sec to more?

Reference: https://github.com/kyle-github/libplctag/releases/tag/v2.0.10

kyle-github, 2019-09-21:
Hi Aravind,

Right now, there is a way to make the disconnect/reconnect time shorter, but not longer.   
It sounds like you need a few changes:

- disable automatic connect and disconnect.
- report the connection status on a tag.
- change the idle timeout.

Is that correct?

I will have to experiment a bit to see what the PLC allows (at least my PLCs) for the idle time.   It is part of the overall set up of a connection for sessions that use connected mode.   There is an underlying limit (from what I have read) but I do not know if it is possible to change that limit in the PLC.   I believe that it is higher than the current 5 seconds.   However, there is an upper limit and I am not sure what it is.  I do not think it is all that long.  Perhaps less than one minute?

Can you describe more about what you are doing that needs these changes?

AravindPavate, 2019-09-23:
> Hi Aravind,
>
> Right now, there is a way to make the disconnect/reconnect time shorter, but not longer.
> It sounds like you need a few changes:
>
> * disable automatic connect and disconnect.
> * report the [...]

Hi Kyle, 

Thanks for your response. You right on the changes we are looking for 
 
> * disable automatic connect and disconnect. ( as in Version 2.0.9)

if the above is not possible then the following option are required 
> * report the connection status on a tag.
> * configurable idle timeout. 

As of now we sew the disconnection issue due to delay in communication (>5 sec) couple of times due to ActiveMQ overload. So we want to have control over disconnection either by disabling auto disconnect feature or by configuring the idle timeout and reporting to higher layers on the auto disconnection to handle it gracefully.  

Let me know if you have more questions and which of our above requirements can be met.

Thanks & Regards,
Aravind V Pavate

kyle-github, 2019-09-24:
Hi Aravind,

Thanks for the information.  I want to dig a little deeper.

You stated that you need to turn off the idle detection because something in your application occasionally pauses long enough to trigger it?  

Assuming that this is correct, what is the problem with dropping the connection?   The PLC will drop the connection on its side (in a very annoying way) in about five seconds.   I just try to do it first so that a clean shutdown of the connection and clean up of the PLC's resources is guaranteed.   The PLC usually only has the ability to support between 32 and 64 TCP connections (perhaps the newer ones go to 128).   So dropping a connection you are not using is a good habit.

The library will reconnect automatically when you next try to access any tag on that PLC.   Usually this only takes a few tens of milliseconds.   Is that delay the problem?   Or is something else happening?

The reason I am asking is that I want to make sure that you are not seeing reconnection failures which would result in a 5-second back off before retrying.   If you are seeing this 5-second back off then it is possible that the PLC is dropping the connection first. 

If the PLC drops the connection on its own, it can be painful to recover.   I have heard of cases where the TCP connection is dropped by the PLC but the OS on the PC is not notified.   There are several cases reported by a few users where the PC could not reconnect without restarting the PC!   Or the application suddenly quits in this case with an uncaught signal.

I _think_ I have worked around all these issues (at least on Linux) but if you have a case where this is still happening, then this is an urgent bug!

I can certainly change the idle handling to turn off idle connection dropping but then that may recreate the problems I mention above.  I can also change the code so that a longer idle period is requested from the PLC.   Based on what I read, the PLC will likely drop the connection sometime between 30 seconds and two minutes no matter what I request though.

Can you please describe the behavior you see with your application when there is a pause in the data flow and the idle drop kicks in?

If you add "&debug=3" to the string you use to create the tag, you may be able to capture more information about what is happening inside the library and I may be able to determine if there is an unknown condition or a bug here.

AravindPavate, 2019-09-25:
> Hi Aravind,
>
> Thanks for the information. I want to dig a little deeper.
>
> You stated that you need to turn off the idle detection because something in your application occasionally pauses long enough [...]

closed 2019-09-25 by AravindPavate

AravindPavate, 2019-09-25:
Hi Kyle,

Thanks for your reply. I think I missed to mention one portion here is, for testing purpose we have created PLC Simulator Application (mocking the real PLC) which acts as a server for incoming request from the client application. Because the PLC Simulator Application is Synchronous once the disconnection happens it is just reporting "Remote client has closed the connection" and not trying to reconnect further. So we are planning the to change the PLC Simulator to listen asynchronously to incoming requests. Which we feel resolve the issue with this.

We are yet to test our New Application with Libplctag Library with actual PLC Hardware and if we face this disconnect issue there, then we will pursue this issue further. So closing this issue at this time. Thanks for all your response and time.

kyle-github, 2019-09-26:
Hi Aravind,

Please let me know if the new simulator works.

The libplctag library has a very simple form of simulator in the lgx_sim program.   You can look at that code to see if any is useful to you.