libplctag/libplctag #112: Unable to read tag from multi-dimensional array

vavruska, 2019-10-01:
I am trying to read a DINT tag value from a multidimensional array MinorFaultBitsHistory[0,0] and am getting an error -19 (PLCTAG_ERR_NOT_FOUND). I can retrieve values from a one dimensional array just fine.

I turned on debug=3 (debug=1 did nothing). Stripped what I believe to be the relevant parts.

2019-09-01 13:19:14.964 thread(1) tag(21) INFO find_tag_create_func:82 Matched protocol=ab_eip
2019-09-01 13:19:14.964 thread(1) tag(21) INFO ab_tag_create:142 Starting.
2019-09-01 13:19:14.964 thread(1) tag(21) INFO rc_alloc_impl:99 Starting, called from ab_tag_create:149
2019-09-01 13:19:14.964 thread(1) tag(21) INFO rc_alloc_impl:118 Done
2019-09-01 13:19:14.964 thread(1) tag(21) WARN check_tag_name:720 parse of CIP-style tag name MinorFaultBitsHistory[0,0] failed!
2019-09-01 13:19:14.964 thread(1) tag(21) INFO ab_tag_create:306 Bad tag name!
2019-09-01 13:19:14.964 thread(1) tag(21) WARN plc_tag_create:453 Error PLCTAG_ERR_BAD_PARAM while trying to create tag!

kyle-github, 2019-10-02:
Hi, thanks for using the library.

The library currently does not handle multi-dimensional tags in the format that you used.  Try "MinorFaultBitsHistory[0][0]" using separate array indexes.

I am going to mark this as an enhancement request because it has come up from time to time!

vavruska, 2019-10-02:
I made the changes as suggested and it works great for getting and setting.  
Thanks for a clean, easy to use library.

kyle-github, 2019-10-06:
I updated the library code and released version 2.0.23 with fixes for this.   It should support both ways of specifying multi-dimensional arrays now.   Thanks again for reporting this!

closed 2019-10-06 by kyle-github