libplctag/libplctag #115: Listing of plc tags has some issues

ManuelJm, 2019-10-09:
Hi, while trying to recovery a list of plc tags you only get that tags inside the scope of the main program other tags are not listed (the scope are listed but not their tags), also there is no a way to list then althought you can read then proviving the scope in the tag path.

Is there any plan to improve that? Or maybe i'm missing something with the tag listing.

kyle-github, 2019-10-09:
Hi, thanks for using the library!

The current code does not read program tags.   I have found some code in pylogix which does, but it has not been a priority to add that to the library.   It does not look particularly difficult but, as with everything Allen-Bradley, will require experimentation.

The syntax I was thinking of was `PROGRAM:<prog name>.@tags` to keep somewhat aligned with what is there now.

I take patches :-)

My current priorities are:

- bug fixes.
- formalize my testing process and make it part of the CI system as much as possible.
- get the library packaged for Debian and possibly NuGet.
- get the library buildable on Android.
- reduce the memory use to work on embedded systems.
- more developer UX improvements (such as program tag listing and showing the tags in the format needed to create a tag).

If there is a great need for this by others, I will rearrange my priorities to work on this.

I should probably publish the priorities to get more feedback.

Thanks for bringing this up!

kyle-github, 2019-10-09:
I set up a poll in issue #116, so please vote!

kyle-github, 2019-10-13:
I was looking over the code for pylogix to get some ideas and notices that Dustin implemented program tag listing.   It turned out to be fairly simple, so a couple of hours of coding and testing and this is now implemented in v2.0.25!

closed 2019-10-13 by kyle-github

ManuelJm, 2019-10-13:
Cool, what syntax is used?

PROGRAM:<prog name>.@tags ?

kyle-github, 2019-10-13:
Oops, yes!

**Controller tags:** `protocol=ab-eip&gateway=10.206.1.39&path=1,4&cpu=lgx&name=@tags`

**Program tags:** `protocol=ab-eip&gateway=10.206.1.39&path=1,4&cpu=lgx&name=Program:MainProgram.@tags`

The latter gets the MainProgram tags.

ManuelJm, 2019-10-14:
Thanks I'll test it and let you know.

ManuelJm, 2019-10-15:
Hi Kyle, the feature seems to work well. Nice work.