libplctag/libplctag #116: Poll: project priorities

kyle-github, 2019-10-09:
I only have so much time and due to the nature of this kind of library, there are not many people who are providing patches.  So it is mostly just me working on the core library.

I have been setting priorities based on what interests me and my direct customers, but I think I need to change that based on comments I have had recently.

So here is a poll, please let me know what you would be most interested in!

/polls "Debian package" "NuGet package" "Android port" "Embedded" "UX Feature: program tag listing" "UX Feature: sample web-based tag explorer" "Add Modbus TCP" "Add Siemens protocol" "Add support for PCCC/DF1-based PLCs other than PLC/5 via DH+ bridges"

If you have something else that is not on the list above, please add it to the comments on this issue.

I will run the poll for a couple of weeks, perhaps to the end of October and see what priorities come out of this.   

Note that I am quite hardware limited.  I have a few ControlLogix CPUs and a PLC/5.   I have a DHRIO link between one of the ControlLogix systems and the PLC/5 as well as Ethernet on the PLC/5.  Based on past experience, this means that my testing has holes.   If you can donate old hardware or you can provide online access to it, that helps greatly in my ability to add new features for old PLCs.

Please let me know what is important to you!

kyle-github, 2019-10-10:
Hmm, there are supposed to be two more entries!  At least on Chrome I am only seeing 8 entries.

kyle-github, 2019-10-10:
It looks like options with "/" in them may break the poll app.   Should I do this again?

xinthose, 2019-10-10:
Oh yah, you're right.  What are the other two options?

kyle-github, 2019-10-10:
One was for DF1/PCCC PLC support (PLC/5, SLC, etc.) via a DH+ bridge.  Right now, only PLC/5 is supported.

xinthose, 2019-10-10:
OK.  Don't most Siemens PLC's support modbus too?  I know Schneider Electric's Modicon PLC's do.

Twoshrubs, 2019-10-10:
I used to use LibNoDave with the PLCSoftwareToolbox wrapper for building Siemens applications. Funnily enough at the place I was doing it I use to work with a guy called Dave, the fun i used to have in meetings talking about Dave Connections.

I really need to get out more!

Yes all modern forward thing PLC manufacturers support Modbus.. erm apart from Rockwell :D

xinthose, 2019-10-10:
The MicroLogix 1400 Series B has Modbus built in, free of charge, but not the others.  You have spend $1k+ to add Modbus to a CompactLogix :(

kyle-github, 2019-10-11:
I moved the poll to SurveyMonkey.   It was not possible to modify this poll and still keep the results.

kyle-github, 2019-10-11:
To the comments above about Modbus, so if I support Modbus, I'll get most Siemen's PLCs "for free"?   That would make Modbus a much more interesting option!

Twoshrubs, 2019-10-14:
Yes, last time I used Siemens (about 6yrs ago) I was using Modbus to talk with a non-Siemens servo drive.. that was using TIA Portal. 
I remember back in my airport days (about 12yrs ago) that someone said you had to buy the license to use modbus with Siemens but that was using Step7. looking around the interweb, I dont think this is still the case. Plus with Modbus you can talk to all the Modicon/Schneider PLC's and alot of off the shelf equipment talks in Modbus.
So either Modbus TCP or Profinet is good in my book.

kyle-github, 2019-11-03:
The poll is closed.

The next priority is Siemens S7 protocol.  After that is embedded.

closed 2019-11-03 by kyle-github