libplctag/libplctag #117: Help CompactLogix1769 L30ERMS

rafaelkrause, 2019-10-10:
Hi,

Any one have tested communication with CPU CompactLogix1769 L30ERMS ? Which Rack and Slot are used on configuration.

Thanks in advance

BR
Rafael

xinthose, 2019-10-10:
My tag path looks like `protocol=ab_eip&gateway=10.0.0.3&path=1,0&cpu=lgx` etc.  I have my CPU in slot zero, first rack,

From the [wiki](https://github.com/kyle-github/libplctag/wiki/API#creating-a-tag): 

> path=1,0 - Required for LGX, Optional for PLC/SLC/MLGX IOI path to access the PLC from the gateway. This example translates as BACKPLANE->SLOT 0.

kyle-github, 2019-11-11:
Can I close this?   Thanks for any feedback.

rafaelkrause, 2019-11-12:
Yes for sure !!! It was a problem with firewall tunnel etc... Thanks for the help

kyle-github, 2019-11-13:
OK.  Thanks.

closed 2019-11-13 by kyle-github