libplctag/libplctag #118: Memory leak on tag listing and possible erroneous tag creation

ManuelJm, 2019-10-18:
HI Kyle, when creating a tag there is a memory leak that arises in some cases, on the function
setup_tag_listing also were not sure if after returning the calling function will behave ok or not.

Please check up this modified code, we couldn't mark it better using the github mechanism, hope its clear enough, look almos at the end of the function, we've added a comment prior to returning the tag not found error.

int setup_tag_listing(ab_tag_p tag, const char *name)
{
    char **tag_parts = NULL;

    pdebug(DEBUG_DETAIL, "Starting.");

    /* Check for a match with just "@tags" for a controller tag listing. */
    if(str_cmp_i(name, "@tags") == 0) {
        /* controller tag listing. */
        pdebug(DEBUG_DETAIL, "Tag is a controller tag listing request.");

        /* fall through to the last part to set up the tag. */
    } else if(str_length(name) >= str_length("PROGRAM:x.@tags")) {
        tag_parts = str_split(name, ".");

        /* check to make sure that we have at least one part. */
        if(!tag_parts) {
            pdebug(DEBUG_INFO, "Tag is not a tag listing request.");
            return PLCTAG_ERR_NOT_FOUND;
        }

        /* check that we have exactly two parts. */
        if(tag_parts[0] != NULL && tag_parts[1] != NULL && tag_parts[2] == NULL) {
            /* we have exactly two parts. Make sure the last part is "@tags" */
            if(str_cmp_i(tag_parts[1], "@tags") != 0) {
                mem_free(tag_parts);
                pdebug(DEBUG_INFO, "Tag is not a tag listing request.");
                return PLCTAG_ERR_NOT_FOUND;
            }

            if(str_length(tag_parts[0]) <= str_length("PROGRAM:x")) {
                mem_free(tag_parts);
                pdebug(DEBUG_INFO, "Tag is not a tag listing request.");
                return PLCTAG_ERR_NOT_FOUND;
            }

            /* make sure the first part is "PROGRAM:" */
            if((tag_parts[0][0]) != 'P' && (tag_parts[0][0]) != 'p') {
                mem_free(tag_parts);
                pdebug(DEBUG_INFO, "Tag %s is not a tag listing request.", name);
                return PLCTAG_ERR_NOT_FOUND;

            }

            if((tag_parts[0][1]) != 'R' && (tag_parts[0][1]) != 'r') {
                mem_free(tag_parts);
                pdebug(DEBUG_INFO, "Tag %s is not a tag listing request.", name);
                return PLCTAG_ERR_NOT_FOUND;
            }

            if((tag_parts[0][2]) != 'O' && (tag_parts[0][2]) != 'o') {
                mem_free(tag_parts);
                pdebug(DEBUG_INFO, "Tag %s is not a tag listing request.", name);
                return PLCTAG_ERR_NOT_FOUND;
            }

            if((tag_parts[0][3]) != 'G' && (tag_parts[0][3]) != 'g') {
                mem_free(tag_parts);
                pdebug(DEBUG_INFO, "Tag %s is not a tag listing request.", name);
                return PLCTAG_ERR_NOT_FOUND;
            }

            if((tag_parts[0][4]) != 'R' && (tag_parts[0][4]) != 'r') {
                mem_free(tag_parts);
                pdebug(DEBUG_INFO, "Tag %s is not a tag listing request.", name);
                return PLCTAG_ERR_NOT_FOUND;
            }

            if((tag_parts[0][5]) != 'A' && (tag_parts[0][5]) != 'a') {
                mem_free(tag_parts);
                pdebug(DEBUG_INFO, "Tag %s is not a tag listing request.", name);
                return PLCTAG_ERR_NOT_FOUND;
            }

            if((tag_parts[0][6]) != 'M' && (tag_parts[0][6]) != 'm') {
                mem_free(tag_parts);
                pdebug(DEBUG_INFO, "Tag %s is not a tag listing request.", name);
                return PLCTAG_ERR_NOT_FOUND;
            }

            if((tag_parts[0][7]) != ':') {
                mem_free(tag_parts);
                pdebug(DEBUG_INFO, "Tag %s is not a tag listing request.", name);
                return PLCTAG_ERR_NOT_FOUND;
            }

            /* we have a program tag request! */
            if(!cip_encode_tag_name(tag, tag_parts[0])) {
                mem_free(tag_parts);
                pdebug(DEBUG_WARN, "Tag program listing, %s, is not able to be encoded!", name);
                return PLCTAG_ERR_BAD_PARAM;
            }
        } else 
		{
			**_// mem free is required here or memory leak will be produced!!_**
			mem_free(tag_parts);
                        pdebug(DEBUG_INFO, "Tag is not a tag listing request.");
            _**// this error code is returned but when is processed by calling function get_tag_data_type
            // seems its not handled, could you confirm this is ok?**_
            return PLCTAG_ERR_NOT_FOUND;
        }
    } else {
        pdebug(DEBUG_INFO, "Tag is not a tag listing request.");
        return PLCTAG_ERR_NOT_FOUND;
    }

    if(tag_parts) {
        mem_free(tag_parts);
    }

    tag->tag_list = 1;
    tag->elem_type = AB_TYPE_TAG_ENTRY;
    tag->elem_count = 1;  /* place holder */
    tag->elem_size = 1;

    pdebug(DEBUG_DETAIL, "Done.");

    return PLCTAG_STATUS_OK;
}'

kyle-github, 2019-10-18:
Hmm, looks like it could be a leak to me too!  I will test when I get back to my home lab.

Thanks for finding this!

Once I get a good block of time, perhaps during the holiday season, I am going to figure out how to run a fuzzer against the library.  I am sure that there are some bugs waiting to be found.

kyle-github, 2019-10-19:
Thanks for finding this!

ManuelJm, 2019-10-19:
We've found this using Visual Studio Leak detector if you are interested.

ManuelJm, 2019-10-19:
> Closed #118 <https://github.com/kyle-github/libplctag/issues/118> via
> 1d7f2d5

Just to make sure , the tag producing this error is a tag name reported by
the sae plc using the tag listing, as i understand this error is produced
when
the tag is malformed.
How this could be possible if its the same name used by the plc?
I'll do something with this tag to see it we can read it or not but could
be that this is hiding another problem.

ManuelJm, 2019-10-22:
Hi Kyle, attached is a csv file generated from a compact guard logix plc.

Some of the tags on the list could not be readed correctly by libplctag
(tag is created erroneously or it detects the tag name as misleading).

The first tags that fails is :

TAG A_Paro_Emergencia1100 BOOL (Class := Safety, RADIX := Decimal, Constant
:= false, ExternalAccess := Read/Write)

Could you check them?

kyle-github, 2019-10-22:
Hi Manuel,

Unfortunately, I do not see an attachment :-(

Can you please send it again?   Thank you very much for continuing to test the edge cases!

Perhaps this should be a separate issue if the tags are not able to be read correctly?   Is that what you are writing above?

ManuelJm, 2019-10-22:
File was attached but probably it was banned by email server

here are the csv lines

remark;"CSV-Import-Export"
remark;"Date = Mon Jul 8 08:11:43 2019"
remark;"Version = RSLogix 5000 v30.00"
remark;"Owner = W7 virtual"
remark;"Company = "
0.3
TYPE;SCOPE;NAME;DESCRIPTION;DATATYPE;SPECIFIER;ATTRIBUTES
TAG;;POINT_IO:1:I;"";"AB:1734_IB8S_Safety2:I:0";"";"(Class := Safety
ExternalAccess := Read/Write)"
TAG;;POINT_IO:1:O;"";"AB:1734_IB8S:O:0";"";"(Class := Safety
ExternalAccess := Read/Write)"
TAG;;POINT_IO:2:I;"";"AB:1734_OB8S_Safety1:I:0";"";"(Class := Safety
ExternalAccess := Read/Write)"
TAG;;POINT_IO:2:O;"";"AB:1734_OB8S:O:0";"";"(Class := Safety
ExternalAccess := Read/Write)"
TAG;;POINT_IO:3:C;"";"AB:1734_DI8:C:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
ALIAS;;POINT_IO:3:I;"";"";"POINT_IO:I.Data[3]";"(RADIX := Binary
ExternalAccess := Read/Write)"
TAG;;POINT_IO:4:C;"";"AB:1734_DO8_NoDiag:C:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
ALIAS;;POINT_IO:4:I;"";"";"POINT_IO:I.Data[4]";"(RADIX := Binary
ExternalAccess := Read/Write)"
ALIAS;;POINT_IO:4:O;"";"";"POINT_IO:O.Data[4]";"(RADIX := Binary
ExternalAccess := Read/Write)"
TAG;;POINT_IO:I;"";"AB:1734_5SLOT:I:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;POINT_IO:O;"";"AB:1734_5SLOT:O:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev04:C;"";"AB:ETHERNET_MODULE:C:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev04:I;"";"AB:ETHERNET_MODULE_SINT_32Bytes:I:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev04:O;"";"AB:ETHERNET_MODULE_SINT_28Bytes:O:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev02:C;"";"AB:ETHERNET_MODULE:C:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev02:I;"";"AB:ETHERNET_MODULE_SINT_32Bytes:I:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev02:O;"";"AB:ETHERNET_MODULE_SINT_28Bytes:O:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;CR30:C;"";"AB:440C_CR30:C:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;CR30:I;"";"AB:440C_CR30_p2I4:I:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;CR30:O;"";"AB:440C_CR30:O:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev05:C;"";"AB:ETHERNET_MODULE:C:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev05:I;"";"AB:ETHERNET_MODULE_SINT_32Bytes:I:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev05:O;"";"AB:ETHERNET_MODULE_SINT_28Bytes:O:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev03:C;"";"AB:ETHERNET_MODULE:C:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev03:I;"";"AB:ETHERNET_MODULE_SINT_32Bytes:I:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev03:O;"";"AB:ETHERNET_MODULE_SINT_28Bytes:O:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev06:C;"";"AB:ETHERNET_MODULE:C:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev06:I;"";"AB:ETHERNET_MODULE_SINT_32Bytes:I:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev06:O;"";"AB:ETHERNET_MODULE_SINT_28Bytes:O:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;UR_1:I;"";"_0577:UR_8129B9EA:I:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;UR_1:O;"";"_0577:UR_F1551D4B:O:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev01:C;"";"AB:ETHERNET_MODULE:C:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev01:I;"";"AB:ETHERNET_MODULE_SINT_32Bytes:I:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;dev01:O;"";"AB:ETHERNET_MODULE_SINT_28Bytes:O:0";"";"(Class := Standard
ExternalAccess := Read/Write)"
TAG;;ACK;"";"DINT";"";"(Class := Standard RADIX := Decimal Constant := false
ExternalAccess := Read/Write)"
TAG;;ACK_MEM;"";"BOOL";"";"(Class := Standard RADIX := Decimal Constant :=
false
ExternalAccess := Read/Write)"
TAG;;ALARM_PIC1;"";"DINT";"";"(Class := Standard RADIX := Decimal Constant
:= false
ExternalAccess := Read/Write)"
TAG;;A_Barrera1300;"";"BOOL";"";"(Class := Safety RADIX := Decimal Constant
:= false
ExternalAccess := Read/Write)"
TAG;;A_Barrera1301;"";"BOOL";"";"(Class := Safety RADIX := Decimal Constant
:= false
ExternalAccess := Read/Write)"
TAG;;A_Paro_Emergencia1100;"";"BOOL";"";"(Class := Safety RADIX :=
Decimal Constant
:= false
ExternalAccess := Read/Write)"

kyle-github, 2019-10-23:
Thanks!

Which tags have problems reading?  All of them in the list?   Most do not appear to be actual tags, but are special system data.   For instance "UR_1:I" might not work.

That said, "A_Paro_Emergencia1100" seems like a normal BOOL tag.   What is the tag string you use to read it?   Can you try to read with that string and add "&debug=4" to the tag string.  This will cause the library output (a lot!) debugging info.   If you could please capture that output and attach it to this bug, I can hopefully help!

I will reopen this bug for now to keep it active.

ManuelJm, 2019-10-23:
Yes , "A_Paro_Emergencia1100" is the one with the issue there are more i'll
try to collect a small list.
We will do the test and send you the info.
Thanks Kyle!

kyle-github, 2019-10-30:
Have you had a chance to check on this?

kyle-github, 2019-12-02:
Please open another ticket if you can recreate the problem with the BOOL tags.  Thanks for the great bug reports!

commit 1d7f2d58 (2019-10-19): Fixed memory leak on illformed tag listing name.  Closes #118.
closed 2019-12-02 by kyle-github