libplctag/libplctag #121: Question: Can inputs be read from Micrologx1100?

cp26, 2019-11-05:
Attempting to read whole word of I:0.0.  Is there support for this or a certain format for the tag?

kyle-github, 2019-11-06:
Hi,

Thanks for using the library.   You can read individual words (tested on PLC/5, so hopefully it will work for MicroLogix!) with the tag name "I1:x" where X is the word to read.  I tested with I1:0 and I1:1 and both work.   I1:0.0 does not work.   The request seems to be formatted correctly, but the PLC returns an error.   I am not sure if that is an actual bug or a specific difference of the PLC/5.

Here is my test tag string:

"protocol=ab_eip&gateway=10.206.1.38&cpu=plc5&elem_count=1&elem_size=2&name=I1:0&debug=4"

I used the tag_rw utility to read the input data file.

Please let me know if that works for you!

cp26, 2019-11-06:
Kyle,

Thanks for the reply. So the ML1100 displays the first input word as I0:0.  In order to read this one I had to use I1:0.

closed 2019-11-06 by cp26