libplctag/libplctag #122: Can't access a tag's bits when addressed like "TagStructure.TagName[1].1" on L8

howchmo, 2019-11-07:
I am putting together a simulation of control panels and I can't read and write things like:
     "From_Interface.OCC_Buttons[2].4"

I have used other libraries that do allow this addressing scheme.

howchmo, 2019-11-07:
So, I found a convenience function in the C# wrapper to do the writing.  I was able to create my own wrapper to do the reading.

kyle-github, 2019-11-08:
Hi,

Your example uses structures which have some challenging behavior for single bit/boolean fields.  That aside, the library is not encoding the above example.  It errors out.   So it looks like I need to figure out how that should be encoded.   It seems like it should work (except that AB is fully of bad surprises).

I am going to mark this as a bug.

kyle-github, 2019-11-10:
I have looked up how some other libraries handle this and they all do the masking locally rather than using some special encoding in the tag to do this.   This is dangerous because when you go to write such a tag, you will write the entire underlying DINT (or whatever the underlying type is) which can lead to overwriting other bit values that might have changed in the PLC.

I have not found any protocol documentation about handling bits outside of explicit boolean arrays or BOOL tags.

I will keep looking.   I know that the PLC/5, SLC, and MicroLogix family have a specific protocol operation to do bit-oriented operations.   It seems like the Logix-class PLCs should have such a thing too!

kyle-github, 2019-11-11:
Note that this is a special case of #82.

kyle-github, 2019-11-21:
I have done some more research on this (i.e. looking at all the other EIP libraries!) and it looks like there is not a way to do this in the protocol directly using the CIP READ and WRITE functions!   There is a function to do read-modify-write requests for integral types (bytes, words, double words etc.) but I cannot find any examples of it.  It might be time for trial and error.

I am going to call this an enhancement request rather than a bug.   It still needs to be fixed.

kyle-github, 2020-04-04:
This is now underway.   I found an updated document on AB's site that has an example of how to send the CIP Read-Modify-Write command.

closed 2020-04-05 by kyle-github

howchmo, 2020-04-05:
> Closed #122 <https://github.com/kyle-github/libplctag/issues/122> via
> b7657b7

That is awesome.  I had written my own wrapper that reads the whole byte
before writing to one bit to assure this was not a problem.  Since my
process was the only client, it was not an issue.

kyle-github, 2020-04-05:
This is now ready for pre-release.  Check in a few minutes and I should have that up with all the binaries from the Azure CI pipeline.

kyle-github, 2020-04-05:
Please let me know if the new API works for you.  I am not all that happy with it.   It seems less clean than the rest.

kyle-github, 2020-04-05:
I am working on different behavior with the API.  I like it a lot better.  Now you can use any of the get/set uint8, uint16... int8, int16... functions (as long as the size is less than or equal to the size of the underlying tag).   If the bit is set, reading any equal or smaller size will give you a "1" result.   Otherwise you will get "0".   The get/set bit functions still work to get individual bits in integral tags.

commit b7657b7d (2020-04-05): API version 2.1, fix #122.