libplctag/libplctag #123: Possible Memory Leak

cp26, 2019-11-20:
Using latest libplc within past week. 
**OS**: The test was performed on Ubuntu but I have also seen this on armv5te linux. 
**PLC**: ML1100
**Attached FILES**: debugPLC.txt (debug=3) and valgrindPLC.txt (valgrind)

I am seeing a memory leak confirmed with Valgrind on the call to plc_tag_read  This occurs when the connection to the PLC is broken, typically disconnecting Ethernet cable.  Attaching debug=3 report that shows Ethernet disconnect. 

**Function Calls:**
On program start call plc_tag_create

I have a timer that calls a read function every second that does the following: 
(based on async example)
1. call plc_tag_read, if not ok or not pending then exit
2. do check status while tag status NOT ok, exit if timeout occurs 
3. if make it here then get values 

I am not calling destroy tags in this scenario since I am continuously reading, which I think is acceptable?

I am seeing total requests in queue count increase, can you elaborate on what that is?  

Thanks in advance!
[debugPLC.txt](https://github.com/kyle-github/libplctag/files/3870191/debugPLC.txt)
[valgrindPLC.txt](https://github.com/kyle-github/libplctag/files/3870192/valgrindPLC.txt)

kyle-github, 2019-11-21:
Hi,

Thanks for using the library and creating such a well documented issue!

There are a couple of things going on, but I cannot tell where the problem lies for sure.   Are you calling plc_tag_destroy() or plc_tag_abort() on the tags after noting the timeout?

Unfortunately, the log with debug=3 isn't quite enough for me to tell what is happening.  There are two possible paths that could be the problem here:

1) calling plc_tag_read() again while a request for that tag is in flight is actually queuing another read.   There is code in place to prevent that, but it is possible that there is a regression in the code. 

2) the thread on the back end that is handling the queued requests is stuck in a state where the queued requests are not checked for being aborted and thus are not being freed.   There is a specific set of steps to do this check and purge and I wonder if this is an edge case where that code is not called.

Requests are shared between multiple threads and thus are protected by atomic locks and are reference counted for memory safety.   So if the session thread (that talks to the PLC) is in a state where it is not purging aborted requests then those requests will stay in the queue and will not be freed.

One thing that is interesting is how little memory is still unfreed at the end when checking with Valgrind (that is the tool I use too).   704 bytes is only two requests and that is a lot lower than the queue size was.  This leads me to believe that the problem is more of a delay in freeing than in a permanent leak since almost the entire queue was cleaned up on program exit.   That is why I suspect possibility 2 above.

I am away from my home office right now, but when I get back I will try to duplicate this.   Thanks to your description, it looks like it is fairly straightforward!

Thanks for a great bug report.  I wish they were all as clear as this!

Best,
Kyle

cp26, 2019-11-21:
Great feedback.  

I am not calling the abort function.  I am going to try that now and test again.

cp26, 2019-11-21:
Well that was an easy fix it appears.  Added plc_tag_abort if timeout occurs and seems to have fixed it.

kyle-github, 2019-11-21:
Thanks for checking that.   Interesting.   That actually seems to indicate that option 1 above may be the real problem.   You should not be able to start a new read on a tag that already has an operation in flight.   That's not what I expected!

There are a number of interleaved timeouts and other things happening here.   At some point the back end decides that it has not received a response in too long and times out (independent from the front end).   IIRC, that is about 2.5 seconds.  It will then fail upward and eventually try to close the TCP socket to the PLC and attempt to set everything back up again (this works to cover a large number of problems in a simple and clean way).   The default retry time is 5 seconds.  

I definitely need to tighten up the documentation on async mode.   In that mode, the library gives you pretty much complete ownership of everything.   In synchronous mode, if there is any kind of error, including a timeout, the library will call abort before you even get a return value.   There is more going on that just a bit more control.

Note that with MicroLogix processors, there is no way to do request bundling into single packets.   So there is no particular performance gain from using async mode.   It does allow you to use fewer threads so if your hardware is resource-poor, it still makes sense.

What platform is using the ARM processor?   I have tested the library on a Raspberry Pi (v2 IIRC) in the past.

cp26, 2019-11-21:
I have a web gui where you can add tags for micrologix or compactlogix so I just like to keep the same code structure.  The device resources are also somewhat limited so the fewer threads is a plus too. 

I am using Linux version 2.6.39  (gcc version 4.2.2). Compiling with a ELDK toolchain.  

I am seeing another issue when running this on a OpenWRT OS compiled for uclibc.  This is probably different so I will close this and open a new issue.

closed 2019-11-21 by cp26