libplctag/libplctag #124: ERROR: Incoming ID not valid

cp26, 2019-11-21:
Using latest libplc 
**OS**: The test was performed on OpenWRT
**PLC**: ML1100
**Attached FILES**: debugPLC.txt (debug=4)

On start of program the plc_tag_create is called for all tags.
I have a timer that calls a read function every second that does the following:
(based on async example)
1. call plc_tag_read, if not ok or not pending then exit
2. do check status while tag status NOT ok, abort tags and exit if timeout occurs
3. if make it here then get values

FWIW, this error does not occur on Ubuntu.

If I execute the program with the PLC disconnected I receive this error: `thread(0002) 2019-10-21 23:40:19.701 ERROR to_tag_index:1496 Incoming ID is not valid! Got 0`
This eventually leads to a segfault. 

 If the PLC is connected and then later disconnected this error does not occur.  
[debugPLC.txt](https://github.com/kyle-github/libplctag/files/3877282/debugPLC.txt)

Any suggestions for where to focus in for an error?

kyle-github, 2019-11-22:
This looks like a much older version of the library.   The formatting of the logs does not have the tag ID in it in the prefix to each debug statement.  Did you install it a long time ago?

Please make sure that the library version that is being called is newer than this.  It looks a lot like what would happen if you took new client code and called the old library with it.

Is it possible that you have an older version of the library installed ahead of the new one in PATH or LD_LIBRARY_PATH?

cp26, 2019-11-22:
Okay, so you are exactly right.  I did not put the new library in the proper directory in my SDK.
This has resolved the segfault and the error prints referencing the not valid id.  It seems I am right back where I was  on issue #123. However, the abort tags does not seem to free the memory on the  OpenWRT platform.   The memory creep occurs when starting with PLC disconnected. 

I do not currently have a way to run Valgrind on this platform but maybe this is related to the points you made in #123 .

kyle-github, 2019-11-25:
There seem to be two problems:
1) unfreed memory when the library is shut down.
2) increasing queue depth when the connection is dropped.

I have been poking at this most of the day.   I think I was able to duplicate at least one of the problems you mentioned in #123: a leak in some cases when the library was shut down.  I was only able to duplicate this when I shut down the whole program while a request was in flight.  If that is the only case, this is not dangerous since the OS will clean up memory.  I am not convinced that this is the only case though.   Does this match your observations?

The increasing queue depth problem is proving to be a little trickier.  I am still away from my test bed, so I cannot test it with real PLCs by yanking the Ethernet cable.   I went through the code and tweaked a few lines in the code that starts operations.   

This lives in the _**fix_123**_ branch.  Until I get a chance to test it better I do not want to release this to master.   Could you please check this out and see if this solves at least the unfreed memory problem on program termination?   The fixes for the second part, the queue depth, are just a handful of lines of code, but spread out across many files and they need a lot of regression testing for me to feel comfortable releasing them.

I had to add another error code, PLCTAG_ERR_BUSY, to handle the case that another operation is tried while one is in flight.   That is part of why I do not want to release this without a lot of warning and testing.  It will effectively be a slight API change.

kyle-github, 2019-11-25:
Again, thanks for the great documentation on these issues.  I probably should break these out into two bugs.

kyle-github, 2019-11-25:
BTW, there are two new test programs: async_stress and busy_test.   The first tests multiple async tags at once and just aborts and retries on any error during reading.  It runs until killed with ^C which was about the only way I could consistently kill the program with a request in flight.   The second intentionally calls plc_tag_read() twice on a tag in a row and validates that PLCTAG_ERR_BUSY is returned.

cp26, 2019-11-25:
> There seem to be two problems:
>
> 1. unfreed memory when the library is shut down.
> 2. increasing queue depth when the connection is dropped.
>
> I have been poking at this most of the day. I think I was [...]

In the case of testing on the OpenWRT.  I am observing increasing memory by running `ps` in the terminal.  This is only seen when starting with the PLC disconnected.   

I have not seen the unfreed memory on the OpenWRT when I stop the program with ctrl-c like I did before using the plc_tag_abort on Ubuntu.  But I also do not have Valgrind working on OpenWRT.  So I am not observing mem increase on program close currently.

So I THINK from I can gather there is only one issue at this point related to the mem creep which seems to be tied back to the que increase.  

But if you are seeing something different I can remove the plc_tag_abort and duplicate the memory issue on Ubuntu. And then see if the _**fix_123**_ resolves it?

kyle-github, 2019-11-25:
Calling plc_tag_abort() is still necessary.  But you should see the queue depth stop increasing if the other tweaks worked.
However, this is the part that I cannot test easily right now.
I would appreciate it if you can your existing tests on the new branch to see if things look better.

Best,
Kyle

cp26, 2019-11-25:
Using library from _**fix_123**_  branch.  That does not appear to have worked.  It looks like the que continues to increase. 

If I might make a suggestion, it would be nice to see the library version printed in the debug at start to be certain of version.

Let me if I can test anything else.  Thank you for your efforts!

snippet of debug=4, start w PLC disconnected: 

```
2019-10-25 17:35:37.102 thread(4) tag(11) INFO plc_tag_read:676 Starting.
2019-10-25 17:35:37.103 thread(4) tag(11) INFO tag_read_start:162 Starting
2019-10-25 17:35:37.103 thread(4) tag(11) DETAIL session_create_request:2347 Starting.
2019-10-25 17:35:37.103 thread(4) tag(11) INFO rc_alloc_impl:99 Starting, called from session_create_request:2356
2019-10-25 17:35:37.104 thread(4) tag(11) INFO rc_alloc_impl:118 Done
2019-10-25 17:35:37.104 thread(4) tag(11) DETAIL rc_alloc_impl:123 Returning memory pointer 0xd7fc0
2019-10-25 17:35:37.105 thread(4) tag(11) DETAIL session_create_request:2370 Done.
2019-10-25 17:35:37.105 thread(4) tag(11) DETAIL session_add_request:859 Starting. sess=0x5c7b0, req=0xd7fc0
2019-10-25 17:35:37.106 thread(4) tag(11) INFO session_add_request_unsafe:824 Starting.
2019-10-25 17:35:37.106 thread(4) tag(11) INFO session_add_request_unsafe:843 Total requests in the queue: 697
2019-10-25 17:35:37.109 thread(4) tag(11) INFO session_add_request_unsafe:845 Done.
2019-10-25 17:35:37.110 thread(4) tag(11) DETAIL session_add_request:865 Done.
2019-10-25 17:35:37.111 thread(4) tag(11) INFO tag_read_start:276 Done.
2019-10-25 17:35:37.112 thread(4) tag(11) INFO plc_tag_read:754 Done

 status: 1

2019-10-25 17:35:37.113 thread(4) tag(12) INFO plc_tag_read:676 Starting.
2019-10-25 17:35:37.115 thread(4) tag(12) INFO tag_read_start:162 Starting
2019-10-25 17:35:37.115 thread(4) tag(12) DETAIL session_create_request:2347 Starting.
2019-10-25 17:35:37.116 thread(4) tag(12) INFO rc_alloc_impl:99 Starting, called from session_create_request:2356
2019-10-25 17:35:37.121 thread(4) tag(12) INFO rc_alloc_impl:118 Done
2019-10-25 17:35:37.121 thread(4) tag(12) DETAIL rc_alloc_impl:123 Returning memory pointer 0xd8140
2019-10-25 17:35:37.123 thread(4) tag(12) DETAIL session_create_request:2370 Done.
2019-10-25 17:35:37.123 thread(4) tag(12) DETAIL session_add_request:859 Starting. sess=0x5c7b0, req=0xd8140
2019-10-25 17:35:37.125 thread(4) tag(12) INFO session_add_request_unsafe:824 Starting.
2019-10-25 17:35:37.125 thread(4) tag(12) INFO session_add_request_unsafe:843 Total requests in the queue: 698
2019-10-25 17:35:37.127 thread(4) tag(12) INFO session_add_request_unsafe:845 Done.
2019-10-25 17:35:37.131 thread(4) tag(12) DETAIL session_add_request:865 Done.
2019-10-25 17:35:37.132 thread(4) tag(12) INFO tag_read_start:276 Done.
2019-10-25 17:35:37.133 thread(4) tag(12) INFO plc_tag_read:754 Done
2019-10-25 17:35:37.134 thread(4) tag(13) INFO plc_tag_read:676 Starting.
2019-10-25 17:35:37.135 thread(4) tag(13) INFO tag_read_start:162 Starting
2019-10-25 17:35:37.136 thread(4) tag(13) DETAIL session_create_request:2347 Starting.
2019-10-25 17:35:37.136 thread(4) tag(13) INFO rc_alloc_impl:99 Starting, called from session_create_request:2356
2019-10-25 17:35:37.140 thread(4) tag(13) INFO rc_alloc_impl:118 Done
2019-10-25 17:35:37.141 thread(4) tag(13) DETAIL rc_alloc_impl:123 Returning memory pointer 0xd82c0
2019-10-25 17:35:37.142 thread(4) tag(13) DETAIL session_create_request:2370 Done.
2019-10-25 17:35:37.143 thread(4) tag(13) DETAIL session_add_request:859 Starting. sess=0x5c7b0, req=0xd82c0
2019-10-25 17:35:37.144 thread(4) tag(13) INFO session_add_request_unsafe:824 Starting.
2019-10-25 17:35:37.144 thread(4) tag(13) INFO session_add_request_unsafe:843 Total requests in the queue: 699
2019-10-25 17:35:37.146 thread(4) tag(13) INFO session_add_request_unsafe:845 Done.
2019-10-25 17:35:37.146 thread(4) tag(13) DETAIL session_add_request:865 Done.
2019-10-25 17:35:37.147 thread(4) tag(13) INFO tag_read_start:276 Done.
2019-10-25 17:35:37.151 thread(4) tag(13) INFO plc_tag_read:754 Done
```

kyle-github, 2019-11-26:
Yeah, a queue depth of ~700 is clearly broken unless you are using 700 tags (I have tested with 1000+, so it is possible).

I thought of the version output that you mention last night too.  One of the challenges is determining where to put the output.   The first debugging outputs are actually never printed because it is not until the "debug=X" part of the tag attributes is parsed that the debug level gets set.   So if I printed it out on library start up as a debugging output you'd never see it.

I worked around this by outputting the version as a string if you set the debug level to 4 or greater.   Can you pull the fix_123 branch again and see if you see "libplctag version 2.0.28." in the output, at the beginning?

kyle-github, 2019-11-26:
I want to try to reproduce your test case on Wednesday   Can you provide more details, please?  

I do not have a system running OpenWRT or even a RPi right now.  Unfortunately, it seems like that is a factor here.

Do you start out with the PLC unplugged?

cp26, 2019-11-26:
> Yeah, a queue depth of ~700 is clearly broken unless you are using 700 tags (I have tested with 1000+, so it is possible).
>
> I thought of the version output that you mention last night too. One of [...]

I am only reading 3 tags. 

The library print out works,  that  is great and will help to confirm right lib.  

`libplctag version 2.0.28.
2019-10-26 14:23:49.692 thread(1) tag(0) INFO find_tag_create_func:82 Matched protocol=ab_eip
2019-10-26 14:23:49.693 thread(1) tag(0) INFO ab_tag_create:144 Starting.
2019-10-26 14:23:49.694 thread(1) tag(0) INFO rc_alloc_impl:99 Starting, called from ab_tag_create:151`

cp26, 2019-11-26:
> I want to try to reproduce your test case on Wednesday Can you provide more details, please?
> 
> I do not have a system running OpenWRT or even a RPi right now. Unfortunately, it seems like that is a factor here.
> 
> Do you start out with the PLC unplugged?

Absolutely, what details would be most helpful?

Yes, I am starting out with the PLC disconnected in this test.

kyle-github, 2019-11-27:
Can you describe what you do in your code, or provide stripped down code so that I make sure that I am doing the exact same thing?   I have limited time over the holiday (US holiday), so I want to make sure that I am as efficient as possible!

Also a full capture of all the output from the first couple of seconds of running would help.   You can attach files to issues in GitHub.

cp26, 2019-11-27:
Okay. I have stripped down the code to only contain a timer that calls a function to read 3 tags.  I will have to test in the morning to confirm it functions as before and then I will post along with debug output.

cp26, 2019-11-27:
Attached is the debug of the first few seconds of running the binary. Also attached is the code  that reads 3 tags every second. 

I am seeing a (-29) return code on the float tag for some reason when the printf on status is uncommented, but not seen on Ubuntu. 

[Main.txt](https://github.com/kyle-github/libplctag/files/3898054/Main.txt)

[debugPLC.txt](https://github.com/kyle-github/libplctag/files/3898073/debugPLC.txt)

kyle-github, 2019-11-28:
Thanks for the test program.  I got it to compile and was able to replicate the queue increase.   It was a stupid error on my part.   The new aborted request purging function was called outside the main while loop :-(  Because I do call this function elsewhere, but only when doing packed 
requests (ControlLogix only), I was not seeing it. 

With that fixed, I get a maximum queue depth of 3.   As soon as the main program calls plc_tag_abort, the queue depth drops.   Due to the asynchronous nature of the thread running 
the session and the thread(s) in the client application, this could be higher but should not be more
than 2x the number of threads in this kind of scenario.  And that condition would last no more than
about 1 millisecond (at least on Linux).

The fix_123 branch has been updated with this latest update.

Thanks for your patience with this!

cp26, 2019-11-28:
Awesome!  I was able to test it some and it appears to have resolved the memory creep issue. I need to do some more testing when I'm back beside my test devices but so for so good.  

I appreciate you working to fix this.

Now, I am seeing something that I did not experience before. It is not related to the new library since the old one appears to do it as well.  But I am getting a -29 return code when checking the status of float tags.  The test code I provided above produces this when the return code is printed.  Maybe there is an issue with the Micrologix 1100 PLC, I am not sure right now. I will download a new program to the ML1100 and see.

kyle-github, 2019-11-28:
Good to hear that this is mostly working.   

You are getting a -29 (PLCTAG_ERR_REMOTE_ERR) back.   There are several places that could come from.   Generally this is a place where I have not translated the remote error to a library error.  In all cases it means that the PLC returned an error to the library.

There are three main places that this could happen:

1. In the status response for the EIP wrapper.   That is generally fairly unusual.
2. In the status response of the CIP wrapper for the PCCC command.   This can happen if something is malformed in the path.
3. In the PCCC response itself.  PCCC is the equivalent of the DF1 protocol and has its own status.  This is the most likely.

I got some errors in my testing because on my PLC/5, the data file N7 is only one word long.   So N7:0 works, but N7:1 does not.   But if you are not getting this error on Ubuntu then this is less likely to be the problem.

Hmm, what is the CPU in your OpenWRT box?   Is it big or little-endian?   Perhaps there is something that regressed.  It has been a while since I had a working big-endian VM (I used MIPS 32-bit with Debian) so it is possible that something broke and was not caught.

This is happening just for the floats?   Have you tried any other data files than N and F?

cp26, 2019-11-28:
This is actually not happening on Ubuntu so seems specific to my hardware.  It is possible I had error in my toolchain when compiling the libplctag library since this is a new error for me. 

I do not have my OpenWRT unit accessible now but I am testing on my arm5vte(i think little-endian but need to confirm) unit right now.  I did also notice this on my OpenWRT box at one point earlier today and it is using a ARM926EJ-S (little-endian). 

I tested on N and those still work.

Here is the debug=4, reading one F8:0 tag, if it helps to identify the cause:

[debugPLC.txt](https://github.com/kyle-github/libplctag/files/3899765/debugPLC.txt)

kyle-github, 2019-11-29:
Hmm, you uncovered another bug in the PCCC code.   The error translation was not using the right value.   It looks (PCCC errors are not easy to find translated) like this is actually, "Illegal command or format."   So, my guess is that something is not right with the command being used.  I updated the fix_123 branch with better (or at least different) error decoding.

Can you try to change the PLC type to "PLC5" just for the float case?

Other people have had trouble with the MicroLogix.  It seems to be slightly different than the other PCCC-based PLCs for some reason.   Unfortunately, I do not have one to test with.

Could there be another problem?   Does it have a F8 data file?   Some of the smaller PLCs are really limited and I wonder if some data files are missing.

cp26, 2019-11-29:
I changed the cpu type to PLC5, still seeing the -29 response. Attaching debug.  

The strange thing is the float reads have been working for me before for quiet some time.   

I thought something might be an issue on the PLC. I re-downloaded my program to it but still the same problem. 

[debugPLC.txt](https://github.com/kyle-github/libplctag/files/3905991/debugPLC.txt)

Attaching image of the data file.  
[image]

cp26, 2019-11-29:
I just retested using the latest release branch and then tested the fix_123 branch.  The -29 error does not occur in the latest release branch, but does in fix_123.  Now, it is possible that I made error in the latest compilation of the fix_123 branch library.  So I will double check that.

kyle-github, 2019-11-29:
The error still seems to be 0x10.   Which might be "Illegal command or format."  Maybe.   The problem is that the errors for PCCC wrapped in CIP wrapped in EIP are not obviously documented.  At least as far as I can tell.

Apart from that, there is something odd here.  The PCCC commands for the previous two log files are both for PLC/5 PLCs.   Something is not right here...

OK.  I think I found part of the problem.   If you choose SLC as the PLC type, there was a refactoring error (introduced relatively recently) that used the PLC/5 type.   I added better PCCC error decoding too and that does seem to pinpoint the error, at least somewhat.

I have added some additional debugging output to make sure that the PLC type is clearly shown.  The code path that determines the PLC type did not output anything to the debug logs in the right places to determine exactly which PCCC sub-type was being selected.  Now it does.

There is another version of the fix_123 branch with these fixes in it.   I clearly need to find a way to get a MicroLogix!

BTW, using "slc" as the PLC type is currently the same as choosing "micrologix" but it may not be in the future.   The code paths differ partly today.

I know that the packet sizes for the two types of PLC do differ and that will be a factor when I get to handling tags larger than request/response packets for PCCC PLCs.  The protocol commands supported by PLC/5, SLC and MicroLogix differ though the differences between the last two are small.   

I tried with "micrologix" as the PLC type to make sure that the right packets are created in my testing (which fails against my PLC/5) and it looked good according to the DF1 manual.  I suggest using "micrologix" as the PLC type instead of "slc" as that will guarantee that if I have to make MicroLogix-specific changes you will get them.

Thanks again for your patience and help debugging this!

kyle-github, 2019-11-29:
I am still confused as to why Ubuntu would be different than OpenWRT.  That definitely should not happen if the same version of the library is being used.

kyle-github, 2019-11-29:
I am also confused why the N data files worked.  There are some strange things that AB did with MicroLogix protocol implementation.

kyle-github, 2019-11-29:
All these regressions you are finding is making me rethink my priorities for the next few months.  I need to come up with some way to run real regression tests against realistic data responses.  Since I cannot put PLCs in the Azure data centers (I use Azure pipelines for building), I am going to need some other way to test.

This is going to delay S7 support, but I cannot have these kind of regressions so I need to have a strong framework to test and integrate that into some CI system.   Maybe simple packet reflection and matching would work...

cp26, 2019-11-30:
> I am also confused why the N data files worked. There are some strange things that AB did with MicroLogix protocol implementation.

Sorry about the confusion here.  I had updated my Ubuntu library to the fix_123.  I seem to always be testing for multiple hardware sets and forget to update libraries before testing.

cp26, 2019-11-30:
Okay, the latest fix_123 looks to have fixed it!  Floats are reading again using cpu=slc, or cpu=micrologix.  

I typically have access to microlgix, compact logix, and control logix so let me know if I can help test anything.

kyle-github, 2019-11-30:
Thanks for testing this!  I am going to run through more regression tests this weekend and if it looks OK, I'll make a release.

I am starting to work on a regression test suite that does not depend on having a PLC.  So I would like to ping you once it is ready to get packet dumps from MicroLogix.  I have a ControlLogix (several) and a PLC/5.   

I am working on a very simple packet matching system that does no real interpretation of the packets themselves.   I think I can use it with appropriate packet dumps from various PLC types to make sure that I can more easily test for regressions.   It is very time consuming now and without all the types of PLC, it is difficult to make sure that I do not break systems I do not have.

Thanks very much for all your help in debugging and fixing the issues with the MicroLogix and the queue depth problem!

kyle-github, 2019-12-02:
I did an official release with the fixes tested above.  I did additional tests with Valgrind covering the following scenarios:

1. the library connects to the PLC and the PLC is taken offline.  Then the application is brought down.
2. the PLC is offline when the application is started but is later brought online when the application is still running.
3. the application is up and running and the PLC is taken offline and brought back online.

I tested these three scenarios against a PLC/5 reading F8:0, against a ControlLogix using unpacked requests and against a ControlLogix using packed requests.

In all cases the queue depth is bounded.   Note, not zero or a few requests.  I saw up to 33 requests in flight in my tests using async_stress.  This is a new example program that I built after testing with the code you sent.  It does a streamlined form of the same kind of testing.    The queue depth is going to be a function of the timeout duration and the number of tags you are trying to read.   You may want to measure it using the async_stress tool and capturing the logs.

The root cause is that the background thread has some spots in its execution where it is doing blocking IO of some sort.  Particularly when opening a socket or doing DNS lookups.  During these times, the thread is not able to purge the queue.

I am rethinking some of my priorities for the next couple of months.  The queue depth issue is a big one.   Without any way to really bound it, the library will not work at all in smaller embedded systems.   Depending on the version and type of the PLC, the request buffers can be up to about 4k in size.  So 33 already represents a fairly significant amount of memory.   

I have a design in mind to fix the problem, but it is going to take a complete inversion of control flow during request times.   It will reduce the number of total requests in memory to two fixed buffers or about 8k.   The unfortunate part here lies in the amount of code that I will need to change.

I have also started work on a test harness to make sure that regressions become much harder to do against PLCs I do not have.   I hope I can reach out to you in the next couple of months to get some packet dumps for specific scenarios from MicroLogix PLCs.

kyle-github, 2019-12-02:
This is fixed in v2.0.28.

closed 2019-12-02 by kyle-github