libplctag/libplctag #128: using Wrapper for C# and I read slowly.

vidaconcerveza, 2019-12-12:
Hi, 
I just made HMI System as C# application with plctag.dll.
(PLC is Controllogix series.)

I just need to read 40 bit tags to show On/Off status each 100ms,
(Eventhough totally use 100 tags, but I just need to read tags of active page.) 
but It takes more than 3~5 sec to read all tags. 

and I quit and execute the program it works well.
but, if I quit and execute when it works properly, It works bad.
I don't know what it happens.

I dispose when I shut down the program also.
(Do I have to remove tags?)

When I started to develop this project in my laptop, it had not happened.

and now I try to Client's PC and it happens.
( https://github.com/vidaconcerveza/HumanMachineInterface )

if you have any clue, or comments, Please Tell me.

Thanks.

kyle-github, 2019-12-13:
Thanks for using the library!

To restate your problem: sometimes when you run your application, it reads all 40 BOOL tags quickly.  When restart, it takes many seconds to read.  If it takes a long time to read, then on restart it works quickly.   Is that the behavior you see?

I need a bit more information to guess what might be the problem.   Can you please use one of the sample programs to reproduce the problem?

I tried this with the data_dumper example program.  I used the attached config file to read 50 BOOL tags every 100ms.   On my bad network, it always succeeds in less than 50ms.   I did not see any variation or extraordinarily long times after multiple restarts of the program.

[data_dumper2.conf.zip](https://github.com/kyle-github/libplctag/files/3958979/data_dumper2.conf.zip)

What version of the library are you using?   What OS?

Can you get the problem to reproduce when you use the tag_rw tool to read one tag?   If you can, please add a debug=4 attribute to the tag string and capture the output.   This will generate a lot of debugging output and may allow me to determine what the problem is.

Thanks!

vidaconcerveza, 2019-12-13:
Thanks for your response.
Exactly it happenes like your restatement. 

I state the spec. and some information bellow.

1. Programming Language : C# ( Wrapper by Mesta Automation)
2. OS : Windows10 Pro(x64)
3. Library version : I'm not sure cause I got it from Wrapper. I'll try to update to ver2.0
4. PLC : AB Controllogix

I also couldn't see any slow when I was developping in the office (used debug mode)
and now I'm facing this issue in the site PC.

now I don't have PLC so I have to try at the site next monday. 

I'll do some test as your words. but I'm not sure cause I haven't used Cpp or C. 

Thanks a lot

kyle-github, 2019-12-14:
Did you build the library yourself or get it premade somewhere?   If you can, try getting the latest version from GitHub.   The latest release has both source and Windows 10 x64 binaries.

When the program is running slowly, do you see every read take about 5 seconds or do you see a slow read and then a fast one?

Can you share some of your code?   There are some internal timeouts that are about 5 seconds.  If there is a problem connecting to the PLC, it will only retry about every 5-10 seconds.   

If you are using async mode, you should make sure that you call plc_tag_abort (I am not sure what the C# wrapper calls it) if you get an error or you decide to time out the request.

kyle-github, 2019-12-14:
The library calls can be used in two ways: 

1. in synchronous mode.   Each call to plc_tag_read() or plc_tag_write() will wait until either a response from the PLC is received or the passed timeout is reached without a response.   The timeout argument is the number of milliseconds to wait.   To use synchronous mode, make sure the timeout is not zero.

2. in asynchronous mode.   This lets your application logic decide how to handle timeouts and errors.  To use asynchronous mode, pass 0 as the timeout value.

If you are using synchronous mode and you read all of the tags (you said about 40) in one thread, it might take several seconds depending on your network.   In synchronous mode, the library sends a request to the PLC, waits for the response, processes it then returns to your program.   Because of this, you will get very poor throughput if you use only one thread.

The best way to get better performance is to use asynchronous mode.   However, if that is difficult the next best method is to use synchronous mode with a thread per tag.  I do suggest using asynchronous mode.   I can easily read 1000 tags many times per second using asynchronous mode.

Even if you use asynchronous mode, make sure that you do not use long sleeps in any loops.   I strongly suggest using very short sleeps such as 1-2 milliseconds.   If your PC and PLC are linked with a reasonable 100Mbps wired network (not WiFi), you should see response times from the PLC in 10-20ms or perhaps less.   One common thing I see in code is having a loop that checks status and in that loop waiting 50 or more milliseconds each time.   Since the PLC will almost certainly respond in less than that, this kind of wait time just makes the performance worse.

There is example code in async.c, hopefully close enough to C# to be useful!  

I hope this helps!

vidaconcerveza, 2019-12-14:
Thanks for your advise.
I'm not sure that you have seen my code on my repository,
so I attach the code bellow.

Anyway, as you said, I use just one thread to read/get tag values using synchronous mode. 
(Even though I made Asynchronous socket communication.)

I'll try to edit the code as Async mode like your advise first.

Thanks for your advise and I'll let you know the result.

vidaconcerveza, 2019-12-15:
I attached a part of my code to read/get value.

--------------------------------------------------------------------------------------------------------
~
for (int i = 0; i < 22; i++)
{
    rc = plcClient.ReadTag(tags_status[i], DataTimeOut);
    if (rc != Libplctag.PLCTAG_STATUS_OK)
    {
       Logger("PLC", $"에러:'{tags_status[i].Name}' CANOT READ THE TAG. Error Code : {rc} - 
       {plcClient.DecodeError(plcClient.GetStatus(tags_status[i]))}\n")
    }
    temp_Status[i] = plcClient.GetBitValue(tags_status[i], -1, DataTimeOut);
}
~
UIUpdate_MccStatus(temp_Status_Flag, temp_Status, temp_Manual_Flag, temp_Manual);                            
--------------------------------------------------------------------------------------------------------

when I Checked, each "for statement" takes almost 1 second.
and I have 2 for-statement to get the tag value.

and UIUpdate method is just to control UI and It doesn't take time.

vidaconcerveza, 2019-12-17:
Didn't work Asynchronous mode as well.
Actually I made a simple TestApp(Console) to check it and It also show same things. 

while My HMI is running, 
I turn off and turn on several times the Test App, 
and when the test App is fast, HMI is also fast. and When the test App is Slow, HMI is also slow.

If you have interest you can see my project code bellow, eventhough it is previous version.

https://github.com/vidaconcerveza/HumanMachineInterface.git

Thanks.

xinthose, 2019-12-17:
Why don't you use C++ instead?

kyle-github, 2019-12-17:
Thanks for the code.   

Looking at the code, it looks like 1 second is not a bad time for the for loop.   If you have 22 tags and it takes 1 second to read all of them, that is about 50ms per tag.  If your network is not good, that is not a surprising time.   Your program is not using asynchronous mode, so your performance will not be very good.  If you read all these tags in asynchronous mode you should see the response from the PLC in less than 50ms for all the tags at once.

The most interesting result here is that running two separate programs at the same time gives the same results in each program.  This sounds like a problem with your client PC, your network, the PLC, or your PLC network.   Running two programs at the same time with so little load should result in both programs showing different delays but they both slow down at the same time.   That strongly implies that the problem is outside the programs.

I suggest looking at your network to see what might be wrong with it.  I have seen something a bit like this when there was a network loop (multiple paths from one node to another).  Normally those are stopped by the network switches.  Do you perhaps have multiple computers on the network with the same IP address?

vidaconcerveza, 2019-12-19:
> Why don't you use C++ instead

Even in Programming world, I’m new. 
Still C++ is a kind of Main language, 
I’m using C# cause of UI work.

Thanks!

vidaconcerveza, 2019-12-19:
> Thanks for the code.
>
> Looking at the code, it looks like 1 second is not a bad time for the for loop. If you have 22 tags and it takes 1 second to read all of them, that is about 50ms per tag. If [...]

The site is a Power plant, and they assigned IP address.
I tried async method to update, it didn’t work as well.
The project term was too short, about 2 weeks. So we decided to use Rslinx, it works.

But I’ll try to find a solution for the future work.
Thanks a lot for your support, kyle.
I’ll visit again when I have question

kyle-github, 2019-12-22:
Thanks for trying.   If you get a chance to replicate the problem, I would like to help debug it.   

Please reopen this when you can!

closed 2019-12-22 by kyle-github