libplctag/libplctag #129: Overnight Library returning NULL or Garbage from a tag

nicwilson58, 2020-01-05:
In August 2019 I implemented the library in a factory to validate carton barcodes.   A barcode scanner stores a barcode in a tag (t_barcode),  then sets a true flag in another tag (t_newdata),  an application running on a windows server uses the library to poll the status if t_newdata every 3 seconds,  and if true,  fetches the barcode tag,  validates the barcode then set another tag (t_bargood) to either true or false depending on the barcode validation.  if bad the carton is kicked off the conveyor,  or if good allows it to pass.

It works fine,  and runs all day from 3:00am to 8:00pm.  After 8:00pm its idle until 3:00am next morning but during this time,  the newdata tag is still being polled every 3 secs.  the next morning it stops working,  the newdata tag still seems to work, but the barcode tag returned garbage (like "YYYYYYYYYYYYYY" see more on this below).   If the library is restarted it instantly works again for a day.

I was using an older V1,  thinking it might be a memory leak or a fixed bug,  I downloaded and compiled your very latest version on 10th December 2019 and installed it into the factory.   This did change its behavior it will now last up to 48 hours before failing and the barcode tag now returns NULL instead of the garbage.   Note the tag  is not NULL as we checked on the PLC controller and it definitely has a good value, but as stated a restart of the library resolved the issue instantly until another 24 - 48 hours.

Regards
Nic Wilson

xinthose, 2020-01-06:
I you read `null` can your code automatically destroy and recreate the tag, successfully recovering?  I use the library in factories too and run into strange issues with PLC's and networks occasionally.

kyle-github, 2020-01-07:
Hi Nic,

Thanks for using the library and the problem report. 

A few questions:

1. What kind of tag is the barcode?   Is it a STRING?  ASCII?  UDT?
2. What do you mean by "reading NULL"?   Can you describe the exact value and from which library function you are getting it?   None of the library functions return NULL.
3. Is this a Control or CompactLogix?  
4. What is the network module?   What firmware revision?
5. Is there any activity happening on the PLC such as programming, software revisions or changes being added or any possible change from RUN to PROGRAM mode and back?
6. Are you using the library with any wrapper?  If so, which language?
7. Are you using timeouts with plc_tag_read() or using it in asynchronous mode?  If there are timeouts in either the library or your code, how long are they?
8. What OS and version is the application running on?

Like @xinthose, I use the library in several factories (mostly using a Java wrapper) and see it run for six months or more at a time without any disconnects that do not have obvious reasons (i.e. maintenance turned off the PLC) and after which the library reconnects automatically.   That is not to say that you have not found a new and exciting edge case where the library is not working correctly!

Is it possible for you to run the application with logging enabled?   To so this, add "&debug=4" to the tag attribute string.   Note that this will generate a _LOT_ of logging output, so do not do this if you have very little disk space.   This will provide detailed logging that shows the packets being sent back and forth as well as the internal state of the library.   

Thanks!

nicwilsonmarel, 2020-01-09:
1.  The barcode tag is a string.
2.  With version 1 of plctag.dll,  when it stops working the result of every read of the barcode tag would always be "ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ",  since upgrading to V2, the result of reading the barcode tag is "" (an empty string)  I understand you may not ever return NULL but that is what is being seen (a quick restart of library resolves it immediately.  The function is from the wrapper "ReadTag" which calls "plc_tag_read"
3.  PLC is an Allen Bradley 1756-LS72S GuardLogix 5570 Safety Controller Revision 30.001, Chassis Type: 1756-A7  7 Slot-ControlLogix Chassis. 
4.  as above item 3.  ??
5.  No changes are being made to the PLC at all.  It works perfectly throughout every production day from early morning to late at night,   the issue occurs during idle time every night,  and a restart of just the library and wrapper immediately resolves it every day.
6.  The wrapper is C# and basically a copy of the example wrapper.
7.  Timeouts are set to 5000
8.  application is running on Windows Server 2012

kyle-github, 2020-01-09:
Thanks for the replies.  

One question I forgot to ask initially: is any error being returned from the library code at any point?   

> 4. as above item 3. ??

This was to determine if you are using an EN2T module or something like that.   

> 5. No changes are being made to the PLC at all. It works perfectly throughout every production day from early morning to late at night, the issue occurs during idle time every night, and a restart of just the library and wrapper immediately resolves it every day.

Interesting. This sounds like something is getting stale.   From what you've said, the tag that is read to see if there is new data is still OK, it is just the string tag that has problems?

> 6. The wrapper is C# and basically a copy of the example wrapper.

Which one of the C# wrappers is being used?   There is one by Mesta Automation and one by Corsinvest.

> 8. application is running on Windows Server 2012

Hmm, isn't that the equivalent of Windows 7?  I am currently building on Windows 10 when I build on Windows, so my DLLs probably won't help.

This feels like a bug in the library.   I will see if I can recreate it.   Unfortunately, I am not a C# programmer, but it sounds like the library is the root of the issue.  At the very least, this should be reconnecting without a problem.  

Are the tags being read all in the same process?   If so, are you using threads?

Would it be possible to run with logging enabled?

I will try to replicate this. I will set up two tags, one boolean and one string and read them in different threads.  I will stop reading one tag for at least 12 hours.   I will probably do this initially on Linux as my Win7 VMs are all trashed right now and I am looking around for my older backups.

This might take a few days for me to get some code doing the right thing.

nicwilsonmarel, 2020-01-09:
Hi,

can you send me your email,  bit easier than here, I can send you the wrapper code and compiled dll  I think the UINT16 tags are ok,  to me it seems its just the string playing up, 
I thought i am calling it more often than usual as its being polled every 2-3 seconds which over a double shift its a lot,  and it goes fine,  it only in the idle period which interesting enough will never do a read of the string,  as no barcode arrives,  yet first thing next morning when the first barcode arrives it returns wrong info,  so I thought of maybe a connection time out or similar as its never failed during produciton that I have seen.

nicwilsonmarel, 2020-01-09:
BTW,  I did not write the wrapper,  so I am not fully across where to put the debugging attribute string.   can you elaborate on that and I can see if the factory will let me implement it.

nicwilsonmarel, 2020-01-09:
To answer your first question,  no error is returned,  it behaves like all is fine.      Everything is being read in one process from a python script to the wrapper.   I will do another message below with the flow of how it works.

nicwilsonmarel, 2020-01-09:
I have the following files:

1. plctag.dll
2.  wrapper code dll
3. host exe program
4. python script.

The host program is running on a server,  that just sits and waits for a message to arrive via a cron messaging system,  when the message arrives (every 2 seconds)  the host program executes the python script.  the python scripts reads/writes pre-determined tags.   it looks to see if newdata has arrived if not it exits.  if there is,  it reads the barcode string,  does a SQL query to validate the barcode,  if valid writes true to a good tag else it writes false then exits.   this continues forever every 2 seconds.  so over night, it runs a little faster as there is never a string read,  just newdata read,  nothing there,  exit over and over.

nicwilsonmarel, 2020-01-09:
one thought.  If we create a 4th tag string that just had a static barcode in it,  and if no new data just read the dummy barcode,  would this keep it alive?

kyle-github, 2020-01-09:
You can attach the files here in GitHub just by dragging and dropping them on a reply pane.   Otherwise, my email is is most of the source headers.

From your description above, is the wrapper and library run continuously?  Is the Python script run continuously?   There is a Python wrapper too, so you could cut out the C# part if you wanted to.

I am not sure if having a static barcode will help.   I suspect what might work would be to always read the barcode string but only check it against the database if the new tag boolean was true.

At this point the only idea I have is that somehow the internal state of the string tag is getting messed up and partially cleaned up or somehow destroyed.   At first I thought it could be a problem with the TCP connection (there should only be one if this is all in a single process), but the fact that the boolean tag is read without a problem means that is not the case.

Perhaps there is something in how I am shutting down the tag that does not work in Windows?

kyle-github, 2020-01-09:
Alternatively, you could destroy and recreate the barcode tag in the library directly if you do not see a barcode when the boolean says you should.   This is along the lines of what @xinthose was proposing.

nicwilsonmarel, 2020-01-09:
[Wrapper code.zip](https://github.com/kyle-github/libplctag/files/4038503/Wrapper.code.zip)

nicwilsonmarel, 2020-01-09:
will give the re-create a shot and see if it helps

kyle-github, 2020-01-09:
Thanks.   Hmm, I see some issues already.   The wrapper is definitely for version 1 of the library.   The function signature for plc_tag_create is wrong as is every function which takes a tag as an argument :-(

This looks a lot like an early version of the Mesta wrapper.   

I would suggest looking at the Corsinvest wrapper here: [cv4ab-api-dotnet](https://github.com/Corsinvest/cv4ab-api-dotnet), or the Mesta wrapper here: [libplctag-csharp](https://github.com/mesta1/libplctag-csharp).   

When you create a Tag object, the last parameter is the debug level.

I think the very first thing to try is to make sure that you are using the updated C# wrapper.   Your wrapper looks like a slightly modified copy of the old Mesta wrapper.  The new one is almost identical, but has the right function signatures for the C library.

Are you certain that you are not accidentally using the version 1 library?   The errors you are getting with tags not working after a while sounds very much like a set of problems that I fixed very early in the 2.0 release.   It is very hard to tell because I made a couple of mistakes: I did not rename the library to plctag2.dll when I changed the API and I did not output the version of the library in the debug output until very recently.

nicwilsonmarel, 2020-01-09:
Hi,

thank you for your time on all this.  the original programmers who did the wrapper etc., are not available aright now and I am left holding the ball.  I am more of a C++ programmer than a C# programmer but I can probably attempt an update of the wrapper, but I am thinking the best way is for me to do a file comparison from the Mesta wrapper you linked too and the current one here and migrate the changes so I can more easily compile it and have work.    There are only four files and not than big so the changes do not seem too difficult.  I did have a quick comparison and they are almost identical mostly with the namespace change to work with the host program amongst other small changes. 

Can you please elaborate on the statement:
"The function signature for plc_tag_create is wrong as is every function which takes a tag", I do not fully understand this, while I can see changes in the two versions of the wrapper I did not see any signature values similar that would need altering.

in the Mesta wrapper you pointed too is a compiled version of plctag.dll is that suitable?, its hard to know what versions the various plctag.dlls are, as there is no build versions attached to them for windows.

Hope this all makes sense.

nicwilsonmarel, 2020-01-09:
Hi,

thank you for your time on all this.  the original programmers who did the wrapper etc., are not available aright now and I am left holding the ball.  I am more of a C++ programmer than a C# programmer but I can probably attempt an update of the wrapper, but I am thinking the best way is for me to do is the use the Mesta wrapper you linked too as its closest to the current one and I can migrate the changes to more easily compile it and have work.    There are only four files and not than big so the changes do not seem too difficult.  I did have a quick comparison and they are almost identical mainly the namespace change to work with the host program, so I will leave the new wrapper as is and just change the namespace. 

your statement:
"The function signature for plc_tag_create is wrong as is every function which takes a tag" are you meaning the function parameters being different?    Given the differences I do not see how V2 is working at all, I checked the server and there is only 1 copy of the dll and just just binary compared it to a compile V2 here and its identical so that is odd.

In the Mesta wrapper you gave a link too is a compiled version of plctag.dll is that suitable?, its hard for me to know what versions the various binary plctag.dlls are, as there is no build versions attached to them for windows.

Hope this all makes sense.

nicwilsonmarel, 2020-01-10:
Hi,

I have implemented the new wrapper and I have altered the script so it always does a read of the barcode tag regardless.  Its installed and running and I will keep an eye on it over the next day or too.

If needed I will add the destroy / re-create tag if it still has issues,  but if the 'keep alive' works then so be it.

kyle-github, 2020-01-10:
Hi Nic,

I would probably not use the DLL that is on the Mesta site.  The last update to the GitHub repo was almost a year ago and I have released a lot of updates and small fixes for the 2.0.x version of the library since them.

If you can, use the DLLs from my GitHub site.  If not trying to build them from scratch using Visual Studio 2019 Community Edition (or the Pro version if you've paid for it) is probably best.   I do not know if I can, but I am going to see if there is a way to make Azure Pipelines give me a Windows 7 VM.

The wrapper probably still works due to quirks in the ABI of Windows on X86-64.   When you call an extern function, you need to either push the parameters on the stack or pass them in registers.  Either way, the parameters end up in 64-bit registers or stack slots.  So the fact that the old wrapper assumed that a tag reference was a pointer and the new library uses int32 gets papered over.   

There will be some differences.   plc_tag_create() returns either a valid positive integer handle or a negative error.   The older version of the library returned a NULL pointer if tag creation failed and you pretty much were left guessing what happened.   Now I can at least tell you something.

I do think you may have found a bug so I am going to try to replicate this over the weekend (we are still on a big project right now so my time is limited).

I think the updated wrapper will help.  Please let me know how it goes!

nicwilson58, 2020-01-10:
Hi,

I used my own dll compile from recent Dec 2019 download,  new wrapper done,  all installed and working and it reads the barcode tag regardless so see how it goes over next few days.

Thanks

Nic

kyle-github, 2020-01-11:
Let me know how it works out.   There is another thread on the forum that also has someone using the Mesta wrapper.   They have started to make some changes to it that sound fairly important.

kyle-github, 2020-01-11:
For a comment about the Mesta wrappers see the forum thread "First and Every other scan throws error..." here: [forum](https://groups.google.com/forum/#!topic/libplctag/-X1Gatn9ilk).

kyle-github, 2020-01-11:
I wrote up a quick PLC program and a test C program to check the basics.   Both are attached.   

The PLC program just loops through a list of strings and copies one to a tag called, creatively, "barcode".   Once it copies the string, it sets a boolean called "new_barcode" to true (latched).   It copies a barcode string once per second.

The C program looks for "new_barcode" to become true and reads the "barcode" tag to get the barcode.  It then sets another boolean tag, "barcode_processed", to true.  

The PLC program clears "new_barcode" when it sees "barcode_processed."

I have a manual boolean controlling the one second timer so that it will stop when I toggle the flag.  That simulates your night-time behavior.

The C program only reads the "new_barcode" tag and only writes the "barcode_processed" tag.  I could have done this with one tag, but from your description it sounded like there were two tags involved.   The C program only reads the "barcode" tag.

This is not production ready code :-)  I never clean up the tags.   Unless it errors, the program does not terminate.   I build it as one of the test/example programs.

Let me know if this is close enough to what you are doing.

[barcode_test.zip](https://github.com/kyle-github/libplctag/files/4049937/barcode_test.zip)

nicwilson58, 2020-01-12:
That sounds very close to what I am doing

kyle-github, 2020-01-12:
I let it sit over night with the barcode not changing and just turned on the enabling bit.  It worked fine.

So something is different materially here.  

I am running this faster than you are.  Instead of every three seconds, I am reading the new_barcode tag every 100ms.   These PLCs have have an internal timeout on connections of about 5 seconds.   The library tries to anticipate that by clean up a connection first if it has not been used in 5 seconds.  

Connections to a PLC are shared by all tags on that PLC, unless you force them not to be. So if any tag on a PLC is read more often than every 5 seconds, the connection will stay up.

What timeouts are you using on your tag reads and writes?   Since the library added automatic reconnection with version 2, I have found that just letting the library do the reconnection is a lot simpler.   I tend to use very long timeouts when I use synchronous mode.

I will try the test again, but with 3 second delays between reads of new_barcode.

kyle-github, 2020-01-12:
Now I am running a test with the barcode changing every ten seconds and the check running every four seconds.   I wanted to make the check a little more spread out than yours in case that is the critical factor.   I will let this run for a few hours.

nicwilsonmarel, 2020-01-12:
> --- For your consideration, this email originates from an external sender ---

Hi,

I will connect to the factory today and see how it went over the weekend with the new wrapper compile.  Timeouts are 5 seconds.   Should I increase that? Or what value do I use to let the library do it?

Nic

kyle-github, 2020-01-12:
There is not a specific value for "never timeout".   I am currently using 15 seconds in the test program I shared above. I was using shorter timeouts.

I have to restart my test.  I closed the lid on my laptop and it suspended in the middle of the test.  Sigh.

nicwilsonmarel, 2020-01-12:
Hi,

I have raised the timeout from 5 to 15  secs too just to see.

One question to be sure it’s not the python script causing this.   When the script runs it creates the three tags,  then checks them etc.  When it ends,  all its doing is a client.Dispose(),  is that enough?  Is there anything else the script should be doing by exiting?

Nic

nicwilsonmarel, 2020-01-12:
Hi,

BTW, I installed the changes at the factory Friday afternoon,  and going by the trace I have installed,  it worked perfectly until  8:31am  Sunday morning.   Hard to trace such intermittent issues  ;)

Nic

nicwilsonmarel, 2020-01-13:
Hi,

I looked into the debug,  however I have not had a reply from the factory if I can implement debugging on a production server but will chase this up,  however in the C# wrapper debug is already built in,  the last parameter is optional and is an integer,  and if supplied the &debug=<int> is automatically appended.

Nic

kyle-github, 2020-01-13:
Ugh, so it worked through one shift and then died the next day at the beginning of the next shift?

In my restarted test, I have not seen a problem but it has not been going on long enough, possibly.

I am not closing the tags or cleaning them up in any way in my test program.  I keep them open the whole time.   Can you outline what the Python program does?   I am not sure I have the right algorithm in my test.   It may be that I'm missing something key to figuring out what is happening.

Debug would really be useful if you can.  Be aware that it will generate a _LOT_ of output.   Make sure that you can redirect it into a log file that you can get.   If the factory does let you turn it on, you will want to use debug level 4.   

Thanks for working with me on this.

nicwilsonmarel, 2020-01-13:
Hi,

The python script does all the work,

1.        a cron sends a msg to a host program that only executes the python script.

2.       The script creates the tags

3.       checks the tag values

4.       reads the newdata tag,

5.       reads the barcode tag

6.       verifies the barcode is a valid barcode if received

7.       sets the gooddata tag either way 0 bad, 1 good

8.       Disposes (C# wrapper, deletes all tags)

9.       exits.

10.   Next cron (2 secs) does it all again.

Nic

kyle-github, 2020-01-13:
Er...   So the Python script just calls C# (IronPython?) to set up the tags?   There is a Python wrapper.  You could dispense with the C# code completely if I understand correctly.

From what you describe, the tags are created and destroyed every two seconds.   Did I understand that correctly?   

If that is the case I am probably testing this incorrectly.   When multiple tags are created in a process, they all share the library internals.  If the tags all point to a single PLC, they will share the same TCP connection with that PLC. 

If the Python process is only called once every two seconds and runs briefly then dies, I wonder if the PLC is running out of resources?   You need to explicitly shut down a PLC connection otherwise it will reserve resources for several minutes.

I think I do not understand enough about the whole flow to really see what could be wrong here :-(

kyle-github, 2020-01-13:
The latest test passed:

> Iteration took 12070ms, total elapsed time is 11168977ms.
> Got barcode: Barcode 7
> Iteration took 8053ms, total elapsed time is 11177030ms.
> Got barcode: Barcode 8
> Iteration took 52433588ms, total elapsed time is 63610618ms.
> Got barcode: Barcode 9
> Iteration took 8052ms, total elapsed time is 63618670ms.
> Got barcode: 
> Iteration took 12066ms, total elapsed time is 63630736ms.
> Got barcode: Barcode 1 - with extra
> Iteration took 8053ms, total elapsed time is 63638789ms.
> Got barcode: Barcode 2

So whatever I am testing, it must not be the same as your code :-(

Does the Python process call the C# code for each tag individually?

nicwilsonmarel, 2020-01-13:
Hi,

Let’s clear some stuff up may make it easier for you to understand the method.

The PLC code is being used with and within a very large food production control software suite running in thousands of factories across the world and is very stable.

A part if this is compiling and executing python scripts handling a lot of unique stuff for individual factories.  We have a built in mechanism for dealing with the scripts and it too has been very stable for the 10 years I have worked for the company.  It’s a  host / msgsender  pair that is designed for the msgsender program to periodically send a custom message with custom parameters to the host program.  The host program uses this custom message and parameters to run a python script (compiled first if necessary, subsequently called from cache).    The entire production software is written in C# and the wrapper now sits nicely in that framework.

Moving to a python wrapper would be much harder getting it to sit within the host framework,  where the Mesta wrapper does nicely.

I look a bit deeper in to the two C# wrappers.  The Metsa wrapper is very simple,  is has very little code,  in fact all it does it expose the C++ function calls in the dll’s and its only other function is Dispose()  which deletes all previously created tags.  I have looked through every line of code and I cannot see the wrapper being an issue,  there is simply not much too it.

Now on the other hand the corsinvest wrapper is much more complex with a heap of code doing all sorts of stuff that I have not investigated,  especially when all we need to do is call the functions in the C++ library,  so it escapes me why it’s so complex.  I do not think moving to that wrapper would solve the issue.

As the python script does a Dispose() (which deletes all created tags) before it returns,   yes,   you are correct in that each 2 seconds the tags are created,  used,  and destroyed.  However it cannot be PLC resources,  as just restarting the library instantly fixes it as it did on Friday and ran till Sunday morning before having issues again.  I saw this yesterday morning and it been running since and still is,  but I will check it again later tonight after production finishes.  The PLC itself has been running for a long time with no reset.

The connection is opened with the call to the wrapper “client = Libplctag()”   and ended with client.Dispose(),  I don’t see any other way in the wrapper to close the connection or am I missing something ?

Regards
Nic

nicwilson58, 2020-01-14:
Hi,it does a get status,  then reads newbarcode,  if false does a read of barcode tag anyway as we agreed.  if true then it does a get status followed by a read barcode tag,  checks if its valid,  if not,  does a get status,  then writes false to goodtag,  and zero to newbarcode,  if barcode is valid,  it does a get status,  followed by a write of goodtag to true,  and zero to newbarcode.

Garbage from a tag (#129)

kyle-github, 2020-01-14:
Thanks for the details!

Let's see if I get this correct.   I'll try to draw a diagram: simple pictures, small words, short sentences :-)

msgsender -> host program (C#, Win 10) -> Python scripts (Win 10) ???

or

msgsender -> host program (Python, Win 10) -> host program (C#, Win 10) ???

In your description you say that the host program is in C# but then you say that the Python script is calling Dispose().   Sorry, this is probably just me being confused here.

If all three tags are in the same process (in the OS sense) and thus should be using a single connection to the PLC.   So as long as any one of them is actively maintained they should all work fine.

So far so good?

The latest code you have runs every two seconds.  It opens all three tags and reads the newbarcode tag and the barcode always.  It will only check the barcode if the newbarcode tag is set.   Yet sometimes you see the newbarcode tag set and the library is returning an empty string for the barcode.   You are recreating the tag connections every time yet sometimes it appears that the barcode connection does not work but only after it has been working for a while.   This is what has me stumped.

How are you determining what the barcode string is?   Are you using some code similar to that I have in the C test program?

Do you always see the empty barcode as the first barcode of a new shift?   One of the last messages you sent made it sound like it might have come a bit into the shift.

nicwilsonmarel, 2020-01-14:
The first one, Like this

CRON - Msgsend - host program (C#) - compiled python - wrapper - plctag

Host program only compiles the python and executes it nothing more,  the compiled python calls all the wrapper functions

The barcode is always a fixed length GS128 so if length is correct, and prefix is ok, then a database search is done for it,  if found its valid,  else it’s a labelled carton with no dB record so the carton is rejected off the conveyor.

The plc host software has a scanner which fills in the barcode tag,  sets new data and reads good tag to reject carton or allows it to pass.

So if new data is true l, barcode = GS128 barcode otherwise barcode tag is empty and new data is false.

kyle-github, 2020-01-14:
How does the Python code execute the wrapper code?  I am trying to understand if this is a wrapper around a wrapper (FFI) or another process.

If this is FFI, are you using CPython or IronPython?   I see that there is a project "Python for .Net" that looks like it allows calling .Net assemblies from CPython.  I did not know that existed, neat!

This might all be moot and it might be something completely different.  The fact that you are recreating all the tags every time and it is still only failing after a long period of inactivity seems to be the only solid line of evidence here...   But I have heard of some really strange behavior with Python when you use a native DLL and multiple processes.

I want to solve this, it is driving me nuts.  Probably not as nuts as you!

nicwilsonmarel, 2020-01-14:
Hi,

I don’t profess to know how that part work but the production suite uses IronPython and compiles it and has includes to link to the wrapper.  Very basic knowledgr there I have not investigated or asked how it actually works the mechanism is built in to the suite.

kyle-github, 2020-01-14:
Ah, OK.  Am I correct in understanding that the whole host system is one large process mostly written in Python?   That the tags etc. are created and destroyed in one process?

If this is not all in one process, here is a possible scenario: Python is creating a new process to read the tags and another one to call Dispose().   If this is all in one Python script it may not be, but this is definitely a much more complicated picture than I first assumed.

I may have found something that we can check.    There is a method in the wrapper, RemoveTag(), that does a more explicit destroy operation on each tag and we can modify it to get back status on the destroy operation.

The old code:

```
        public void RemoveTag(Tag tag)
        {
            plc_tag_destroy(_tags[tag.UniqueKey]);
            _tags.Remove(tag.UniqueKey);
        }
```

New version:

```
        public int RemoveTag(Tag tag)
        {
            int rc = PLCTAG_STATUS_OK;
            rc = plc_tag_destroy(_tags[tag.UniqueKey]);
            _tags.Remove(tag.UniqueKey);
            return rc;
        }
```

Can you modify your code to call this instead of Dispose() and to output the return value if it is negative (an error)?   If we get a return value of 0 (PLCTAG_STATUS_OK) from plc_tag_destroy() then the tag was found and the tag resources were correctly released.   You will need to call RemoveTag() for each tag, individually rather than just calling Dispose() once.   And please log the return value from RemoveTag().   That will help tell us whether the tag was found and destroyed.

We could modify Dispose() to return a combined good/bad count, but that would be less clear.

Is this feasible for you to try?   What I am trying to find out is whether the tags are opened in one process and Dispose() is called in another.  In that case, we would leak open sockets potentially.  The PLC will eventually close them and probably would not run out, but there could be some different behavior on the Windows side.

nicwilsonmarel, 2020-01-14:
Hi,

No that is not correct.  The whole host system is a C# system.  A very small part of it is the ability to run python scripts for customization not in the main program,  but this is a very small part of it, then an even smaller part of it,  it a  host/messaging app,  that allows us to execute a python scripts at regular defined intervals, which is what I am using to execute it every 2-3 seconds.

As everything is done in the python script,  then yes all tags are created, used and destroyed in one process.

I can make the changes and try it and let you know.

Nic

nicwilsonmarel, 2020-01-14:
Hi,

I checked the factory again this morning,  and I must say since putting in the new build of the wrapper and also the keep alive barcode change in the script,  I am not seeing the same failures,  so it’s a lot more stable,  so I think we should just leave this and see how long this lasts,  its now gone for longer than it ever has.

Nic

kyle-github, 2020-01-15:
This is good news for your project and bad news for me debugging it :-)

I suspect that adding in the call to Dispose() may have been one of the key changes.

kyle-github, 2020-01-15:
Oops, I am on my phone and hit comment too soon.  

Let's leave this open for a while.   Thanks for your patience with my questions.

kyle-github, 2020-01-22:
Is it still working?

nicwilsonmarel, 2020-01-22:
Hi,

Yes it has not failed in 9 days now.

kyle-github, 2020-01-22:
OK.   That is good.   

I do wish I could figure out how to replicate this.   My guess is that there are two really important changes:

1. The new version of the DLL.   There were several changes in the last year to help with connection drops.
1. Calling Dispose() directly.  That should free up resources if they were leaking.

I am not sure that reading the barcode every time is important, but it might be in your case.

Shall we close this issue for now?

kyle-github, 2020-01-22:
The Dispose() problem is one I would like to solve in the library in some way.

nicwilsonmarel, 2020-01-22:
Hi Kyle,

I think we should close it,  we can always revisit it should it rear its ugly head again. Thanks for your time and effort on this.  ☺

Cheers
Nic

  2.  Calling Dispose() directly. That should free up resources if they were leaking.

kyle-github, 2020-01-22:
OK.   I wish I had been able to find some sort of smoking gun that I could fix in the library.  It did make me start working on ways to make the library enforce versions more closely!

Thanks for bringing this up and providing all the explanations and code!

I'll close this for now.   Please reopen it if you see the problem again!

closed 2020-01-22 by kyle-github