libplctag/libplctag #131: Build failing with gcc 4.8.3

DJWalker42, 2020-01-14:
The compiler complains about unrecognised option '-Wc99-c11-compat' which I believe is a new flag for gcc version 5. Is version 2 of libplctag only compatible with gcc 5 or has cmake got the options wrong somehow? 

The machine on which I'm building the code I cannot update gcc to latest.

Thanks. 

P.S. libplctag version 1.5 builds successfully

kyle-github, 2020-01-15:
Hi, sorry for the delay!

You can modify the CMakefiles.txt file to remove that option, it is just there to help keep the code clean.   I made the library on Ubuntu 12.04 a long time ago and I do not think that I have added anything that requires C11 since then.   I do not have an Ubuntu 12.04 VM to check anymore as it is long past its support period.

I will see if I can make a VM with Ubuntu 12.04 and try to make it work there. I think that had a version of GCC that was 4.8.

kyle-github, 2020-01-16:
I _think_ I have this working with GCC 4.8 on Ubuntu 12.04.   It took some tweaks to the CMake configuration and I am seeing a few warnings for some of the examples, but tag_rw works.   I am making a release as soon as Azure Pipelines completed the build and gets the results.

DJWalker42, 2020-01-16:
Hi,

The new release builds successfully with gcc 4.8.3 with a few pedantic warnings:

> /sdb1/libplctag/libplctag-2.0/src/protocols/ab/eip_cip.c: In function ‘build_tag_list_request_connected’:
/sdb1/libplctag/libplctag-2.0/src/protocols/ab/eip_cip.c:436:5: warning: missing braces around initializer [-Wmissing-braces]
     uint16_le tmp_u16 = {0,};
     ^
/sdb1/libplctag/libplctag-2.0/src/protocols/ab/eip_cip.c:436:5: warning: (near initialization for ‘tmp_u16.val’) [-Wmissing-braces]

> /sdb1/libplctag/libplctag-2.0/src/examples/barcode_test.c: In function ‘main’:
/sdb1/libplctag/libplctag-2.0/src/examples/barcode_test.c:38:9: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 2 has type ‘int64_t’ [-Wformat=]
         printf("Iteration took %ldms, total elapsed time is %ldms.\n",(new_time-last_read),(new_time - first_read));
         ^
/sdb1/libplctag/libplctag-2.0/src/examples/barcode_test.c:38:9: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 3 has type ‘int64_t’ [-Wformat=]

> /sdb1/libplctag/libplctag-2.0/src/examples/data_dumper.c:54:1: warning: missing braces around initializer [-Wmissing-braces]
 } tags[MAX_TAGS] = {0,};
 ^
/sdb1/libplctag/libplctag-2.0/src/examples/data_dumper.c:54:1: warning: (near initialization for ‘tags[0]’) [-Wmissing-braces]

> /sdb1/libplctag/libplctag-2.0/src/protocols/ab/eip_cip.c: In function ‘build_tag_list_request_connected’:
/sdb1/libplctag/libplctag-2.0/src/protocols/ab/eip_cip.c:436:5: warning: missing braces around initializer [-Wmissing-braces]
     uint16_le tmp_u16 = {0,};
     ^
/sdb1/libplctag/libplctag-2.0/src/protocols/ab/eip_cip.c:436:5: warning: (near initialization for ‘tmp_u16.val’) [-Wmissing-braces]

> /sdb1/libplctag/libplctag-2.0/src/examples/stress_test.c: In function ‘main’:
/sdb1/libplctag/libplctag-2.0/src/examples/stress_test.c:240:12: warning: missing braces around initializer [-Wmissing-braces]
     struct sigaction sigpipe = {0,};
            ^
/sdb1/libplctag/libplctag-2.0/src/examples/stress_test.c:240:12: warning: (near initialization for ‘sigpipe.__sigaction_handler’) [-Wmissing-braces]
/sdb1/libplctag/libplctag-2.0/src/examples/stress_test.c:240:12: warning: missing initializer for field ‘sa_mask’ of ‘struct sigaction’ [-Wmissing-field-initializers]
In file included from /usr/include/signal.h:245:0,
                 from /sdb1/libplctag/libplctag-2.0/src/examples/stress_test.c:29:
/usr/include/bits/sigaction.h:43:16: note: ‘sa_mask’ declared here
     __sigset_t sa_mask;

kyle-github, 2020-01-16:
Yeah, I may fix those, but they are valid C for later versions so I will have to see whether it is worth it.   Interestingly I do not get warnings for them from MSVC.   Can we close this issue?

DJWalker42, 2020-01-17:
Fair enough, maintaining code for old(er) compilers is a pain. 

I'll see if I can get a later (or latest) version of gcc on my build machine, or try it with a relevant VM or docker container instead, or just use the zipped release (so many options!!)

closed 2020-01-17 by DJWalker42