libplctag/libplctag #132: better description of the attribute string

mavenraven, 2020-01-16:
Hey, thank you for sharing your library! It would be great if there was more documentation on the attribute string. For example, I have no idea what "path=1,5" means.

kyle-github, 2020-01-16:
[Tag String Attributes](https://github.com/kyle-github/libplctag/wiki/Tag-String-Attributes)

kyle-github, 2020-01-16:
Hmm, the rest of the comment got erased.  Let's try that again...

Thanks for using the library.  There are associated wiki pages that have more information about alternate builds, tuning, and other aspects of the library.  This includes more information on the tag string attributes.

In this case the path is the path within a ControlLogix chassis (it can be more than that).  The first digit is the port out of the network module.  In this case 1 means the port to the backplane.  The next number is the address on the bus.  So slot 5 in the back plane.  

Most people have the CPU module in slot 0.  

I hope that helps!

kyle-github, 2020-01-22:
Has your question been answered?   If so, can we close this?   Thanks for using the library!

mavenraven, 2020-01-22:
Sorry, haven't had a chance to dig in, I've been sick the last two weeks. I'll take a look in the next week.

kyle-github, 2020-01-22:
Ouch, I hope you get well soon.  The flu that is going around is nasty!

mavenraven, 2020-01-29:
Hey @kyle-github , thanks again for your help!

So what I'm trying to do right now is just to get a hello world working against the `RSLogix Emulate 5000 Chassis Montior` with `EmulLogix5868 Controller` in slot 1. I am trying to retrieve a Program Tag called `TotalPartCount` which is a `DINT`.

I've verified that I can connect to the emulator by doing:
`nc 192.168.99.101 44818 -G 1` 

When using the following attribute string:
`"protocol=ab-eip&gateway=192.168.99.101&path=1,1&cpu=lgx&elem_size=4&elem_count=1&name=TotalPartCount&debug=5"`

and simple.c, I get the out
[out.txt](https://github.com/kyle-github/libplctag/files/4130358/out.txt)

put I've attached as a file..

This suggests to me that the library is connecting to the plc correctly:
`2020-00-29 14:27:06.973 thread(2) tag(0) DETAIL socket_connect_tcp:902 Attempt to connect to 192.168.99.101 succeeded.`

This appears to be the issue:
```
2020-00-29 14:27:06.978 thread(2) tag(0) WARN recv_forward_open_resp:2189 Forward Open command failed, response code: Port Not Available (1)
2020-00-29 14:27:06.978 thread(2) tag(0) WARN recv_forward_open_resp:2210 CIP error code 1!
2020-00-29 14:27:06.978 thread(2) tag(0) INFO recv_forward_open_resp:2230 Done.
2020-00-29 14:27:06.978 thread(2) tag(0) WARN try_forward_open_ex:1938 Unable to use ForwardOpen response!
```

This is also a bit weird:
```
2020-00-29 14:27:06.971 thread(1) tag(0) DETAIL session_create_unsafe:446 Warning: not using passed port 44818
2020-00-29 14:27:06.971 thread(1) tag(0) INFO rc_alloc_impl:99 Starting, called from session_create_unsafe:448
```

I'm very new to PLCs, so I'm probably doing something obvious wrong. Thanks for any help you can lend!

mavenraven, 2020-01-29:
[image]

[what my emulator looks like]

mavenraven, 2020-01-29:
[image]

mavenraven, 2020-01-29:
[image]
this is interesting

mavenraven, 2020-01-29:
Also, FWIW, I changed linux/platform.c in the following way as I'm using OS X:

```
     /* The socket is non-blocking. */
-#ifdef SO_NOSIGPIPE
     /* On *BSD and macOS, the socket option is set to prevent SIGPIPE. */
     rc = (int)write(s->fd, buf, (size_t)size);
-#else
-    /* on Linux, we use MSG_NOSIGNAL */
-    rc = (int)send(s->fd, buf, (size_t)size, MSG_NOSIGNAL);
-#endif
```

I highly doubt that this would cause an issue but figured I'd call it out.

kyle-github, 2020-01-30:
Thanks for the questions and excellent data to show what you have done!  In the future, please start a new issue so that others can search this.

I have had people try with the emulator in the past, but they did not get this far.   There are a couple of things to try.   The error you saw with the ForwardOpen is definitely what is causing the library to stop trying.   

Try different paths.   It does look like the CPU should be in slot 1, so 1,1 would be what I would normally expect.   If the port is not right, then perhaps you can try other numbers.  Generally ports range from 0 to 3.   Usually 1 is the back plane, 2 is Channel A (if there is one), and 3 is Channel B (if there is one).  I would try 0,1 at least.

Perhaps the emulator only handles the older ForwardOpen packets.   Try setting the CPU to "micro800".   This will tell the library that the PLC is a Micro800 family PLC which speaks a limited dialect of the protocol.   I think this is less likely as the error is fairly explicit but I could be translating the error incorrectly.

I seem to remember someone suggesting that you explicitly add a network module to the emulator and then use that IP address to access it.   It is actually the network module that implements a fair amount of the network protocol stack (AB never makes this easy!), so this seems possible.

It looks like this is quite close to working.  You are getting a solid session set up and then ForwardOpenEx is getting a completely valid (if not preferred) response back.

Assuming that we can get you past the connection problem, if the tag you are trying to get to is really a program tag, then you will need to access it with "PROGRAM:<prog name>.<tag name>".   If it is a controller tag, then you just need the tag name.

I hope this helps.

On the code change, did it not compile without that?   I do not have a Mac, so the code is carefully stolen^H^H^H^H^H^Hborrowed from Stack Overflow.   Is SO_NOSIGPIPE not defined on Mac now?   There are actually two places that use the same ifdef, one around line 810 and one around line 990 (the one you changed).

I really want to make sure that the code works fine on Mac, so hopefully you will volunteer to be my guinea pig :-)

mavenraven, 2020-02-11:
Hey @kyle-github, so I've tried other products to try to get this work with no success. At this point, I'm assuming that it's either not possible, or it's not worth my effort. 

This is the actual error:
```
warning: unknown warning option '-Wc99-c11-compat' [-Wunknown-warning-option]
/Users/user/libplctag/src/platform/linux/platform.c:997:46: error: use of undeclared identifier 'MSG_NOSIGNAL'
    rc = (int)send(s->fd, buf, (size_t)size, MSG_NOSIGNAL);
```
I'm going to close this out now. Thanks for your help!

kyle-github, 2020-02-11:
Sorry to hear that it is not working.   I was working on supporting macOS a bit better recently.   I hope to check that in in a day or two.   Would you be willing to give the compilation a try when I get this into GitHub?

I did find the following notes on connecting to Logix 5000 Emulate from a GE website: [How to Configured RSLogix RSEmulate 5000](https://digitalsupport.ge.com/en_US/Article/How-to-configure-RsLogix-Emulator-RSEmulate-5000-with-the-IGS-driver).

There are some interesting things in there.   Apparently, Emulate just does not allow anything other than OPC connections, at least as far as I can tell.   It does appear that SoftLogix may support CIP connections.   There is also a weird note about the configuration of RSLinx on the PC running SoftLogix:

> For proper operation, no Ethernet-based drivers (such as ethernet devices, remote devices via Gateway, and so forth) should be installed in RSLinx on the SoftLogix PC. With one or more Ethernet-based drivers installed, requests will return with CIP Error 0x5 Ext. Error 0x1 and CIP Error 0x8.

mavenraven, 2020-02-11:
Yeah, I read that as well. I don't really understand the difference between the two except that Logix 5000 Emulate is newer. For my purposes, talking to the device at the OPC level isn't useful.

Yeah, I'm happy to test it, just @ me when it's ready.

kyle-github, 2020-02-15:
Sorry for the delay.  Can you check out the branch "macos_fix" and see if it compiled OK on a Mac?  Thanks!

mavenraven, 2020-02-15:
Hey no, that doesn't actually work, but I figured out the issue:

On OSX, <sys/socket.h> has:

```
#if !defined(_POSIX_C_SOURCE) || defined(_DARWIN_C_SOURCE)
#define SO_USELOOPBACK  0x0040          /* bypass hardware when possible */
#define SO_LINGER       0x0080          /* linger on close if data present (in ticks) */
#else
#define SO_LINGER       0x1080          /* linger on close if data present (in seconds) */
#endif  /* (!_POSIX_C_SOURCE || _DARWIN_C_SOURCE) */
#define SO_OOBINLINE    0x0100          /* leave received OOB data in line */
#if !defined(_POSIX_C_SOURCE) || defined(_DARWIN_C_SOURCE)
#define SO_REUSEPORT    0x0200          /* allow local address & port reuse */
#define SO_TIMESTAMP    0x0400          /* timestamp received dgram traffic */
#define SO_TIMESTAMP_MONOTONIC  0x0800  /* Monotonically increasing timestamp on rcvd dgram */
#ifndef __APPLE__
#define SO_ACCEPTFILTER 0x1000          /* there is an accept filter */
#else
#define SO_DONTTRUNC    0x2000          /* APPLE: Retain unread data */
                                        /*  (ATOMIC proto) */
#define SO_WANTMORE     0x4000          /* APPLE: Give hint when more data ready */
#define SO_WANTOOBFLAG  0x8000          /* APPLE: Want OOB in MSG_FLAG on receive */

#endif  /* (!__APPLE__) */
#endif  /* (!_POSIX_C_SOURCE || _DARWIN_C_SOURCE) */
```

The CMake file has:

```
if (CMAKE_C_COMPILER_ID STREQUAL "Clang")
    # using Clang
    set(BASE_RELEASE_FLAGS "${CMAKE_C_FLAGS} -Wall -pedantic -Wextra -Wc99-c11-compat -Wconversion -fms-extensions -fno-strict-aliasing -D__USE_POSIX=1 -D_POSIX_C_SOURCE=200809L")
    set(BASE_DEBUG_FLAGS "${CMAKE_C_FLAGS}  -g -Wall -pedantic -Wextra -Wc99-c11-compat -Wconversion -fms-extensions -fno-strict-aliasing -D__USE_POSIX=1 -D_POSIX_C_SOURCE=200809L")
```

If you disable POSIX compatibility, it builds:
```
# C compiler specific settings
if (CMAKE_C_COMPILER_ID STREQUAL "Clang")
    # using Clang
    set(BASE_RELEASE_FLAGS "${CMAKE_C_FLAGS} -Wall -pedantic -Wextra -Wc99-c11-compat -Wconversion -fms-extensions -fno-strict-aliasing")
    set(BASE_DEBUG_FLAGS "${CMAKE_C_FLAGS}  -g -Wall -pedantic -Wextra -Wc99-c11-compat -Wconversion -fms-extensions -fno-strict-aliasing")
```

FWIW, according to https://pubs.opengroup.org/onlinepubs/9699919799/basedefs/sys_socket.h.html, `SO_NOSIGPIPE` isn't POSIX.

kyle-github, 2020-02-15:
Huh...  Interesting.  I wonder why I was setting that POSIX setting.   Looks like I can either drop the POSIX defines or (maybe and) set _DARWIN_C_SOURCE.   

Alternatively, perhaps I need to just copy the platform.c file and make a macOS version w/out all the defines.   Thanks for checking this.

kyle-github, 2020-02-16:
I was able to add macOS to the build CI system and used that to debug my way out of this.  There is now a macOS built on each release.   It uses whatever is the latest from MS Azure Pipelines.  I think Mojave (10.14).  I built release 2.0.31 based on this.   Thanks very much for all the help and all the research you did into this!   There is no way I would have been able to get this far without that!

kyle-github, 2020-03-30:
I have been checking on the Internet to see if anyone has been able to connect to Emulate 5000 and it looks like the answer is "no."  SoftLogix does look possible.   I am going to close this for now.   I will reopen it or file a new issue if I can find evidence that Emulate works.

Thanks again for the help with macOS!

closed 2020-03-30 by kyle-github