libplctag/libplctag #135: Reference/standard for older versions?

mavenraven, 2020-02-11:
Hey @kyle-github, I was just wondering where code like https://github.com/kyle-github/libplctag/blob/master/src/protocols/ab/eip_plc5_pccc.c came from? Is there a standard or reference floating around somewhere, or did you reverse engineer it?

mavenraven, 2020-02-11:
Probably a better way to ask this question: This library implements two protocols: Ethernet/IP and PCCC. Is that correct?

kyle-github, 2020-02-12:
Hi, sorry for the delay.   Work is very busy right now.

The original doc was from an old paper from Rockwell from 2001 that showed an example of talking to a PLC/5 with EIP/CIP and embedded PCCC.   Then I found the docs for DF1 and started putting the two together.

As far as I can tell, it is just PCCC wrapped in EIP/CIP.   I have found mentions that there are differences but so far I have not found any.   The DF1 docs do indicate that PCCC is not very similar across different PLCs.   MicroLogix and SLC are similar, but not identical.  PLC/5 is its own animal.   There are parts that seem to work but do not fully work.  No idea why.  You can use the PLC/5 protocol commands for many things on a MicroLogix, but not every data type.

Note that this is different from talking with a ControlLogix in PCCC mode!   That is slightly different again.

kyle-github, 2020-02-12:
BTW all these papers can be found with Google and generally you can download them directly from Rockwell.  They are not hiding them (except by their habit of moving things around on their web site).

Here are a few:

[DF1 Protocol Reference 1770-rm516_-en-p.pdf](https://github.com/kyle-github/libplctag/files/4189433/DF1.Protocol.Reference.1770-rm516_-en-p.pdf)

[Allen Bradley Logix Controllers Data Access - 1756-PM020F-EN-P - 2019.pdf](https://github.com/kyle-github/libplctag/files/4189444/Allen.Bradley.Logix.Controllers.Data.Access.-.1756-PM020F-EN-P.-.2019.pdf)

kyle-github, 2020-02-12:
I never really answered your question.  The library implements a few bits of EIP (the Ethernet wrapper), some CIP protocol commands, and some PCCC protocol commands.   PCCC is sent wrapped in a CIP wrapper in EIP.   Why AB needs such a complicated mess of protocols stacked on top of each other, I do not know.

mavenraven, 2020-02-16:
Cool, thanks for the info! My goal is to have a curl like tool that grabs all the tags off a given AB PLC, either with libplctag or something like https://github.com/digitalpetri/ethernet-ip . I just got a real PLC to play with, so I should be able to see what's going a bit better. The problem I have libplctag right now is that the API is very low level for my purposes, since I just want to get all the tags and dump them to stdout.

closed 2020-02-16 by mavenraven