libplctag/libplctag #138: Dll x86

franklupo, 2020-03-05:
Hi,
Are possible to make dll x86 for windows and linux?

Best regards.

kyle-github, 2020-03-06:
Hi @franklupo, 

I have been looking and I do not see a way to make Azure Pipelines spin up a 32-bit VM.   I will keep looking.   Perhaps GitHub has something.

I have no personal VMs with 32-bit Windows. I do have a VM with 32-bit Debian 10 x86.   I will see if it is easy to set that up.

kyle-github, 2020-03-06:
It does not appear that I can get a VM with 32-bit Windows easily in the free tier of Azure Pipelines.   I do have a 32-bit Debian VM on my laptop and I used that to create a 32-bit Debian 10 release.  I added that to the pre-release 2.0.34.  Please try it and let me know if it works for you!

kyle-github, 2020-03-14:
Hmm, I wonder if it is possible to build a 32-bit DLL on 64-bit Windows?   I will see if I can figure this out.   It might also be possible for Linux.  That would allow me to build all these DLLs from Azure.

franklupo, 2020-03-14:
Hi,
I think it is possible to create both in one solution.

Bets regards

kyle-github, 2020-03-16:
It was not as easy as I had hoped, but I was able to create 32-bit DLLs for Windows and Linux.  Unfortunately, I cannot test the Windows DLLs as I have no 32-bit Windows systems.  Can you please check the latest pre-release for v2.0.35 and see if the 32-bit Windows binaries work?  Thanks!

I have this fully integrated into the CI system so every time the master branch is pushed, Azure Pipelines will build all five sets of binaries.

I see that you are in Italy.  You have all our best wishes at this horrible time.  Stay safe!

franklupo, 2020-03-16:
ANDRÀ TUTTO BENE

Grazie

kyle-github, 2020-03-18:
I hope it will all be OK!

Please reopen this if the DLLs are not working!   Thanks for bringing this up.

closed 2020-03-18 by kyle-github