libplctag/libplctag #139: can you force AB plc's with this library?

lyndoncoin, 2020-03-13:
Hi, Is it possible to use this library to connect via ethernet and force outputs on remote IO without any modification to the actual PLC program itself? I'm struggling to find info on this.

kyle-github, 2020-03-13:
The library only can read and write tags exposed within the PLC.  You might be able to write to an output tag.   Unfortunately, I have not tried much with IO tags as my local lab does not have any IO cards in my PLCs.

kyle-github, 2020-04-05:
It might be possible to do some sort of forcing with the new single-bit tags.   Do you have access to a PLC to try it?

lyndoncoin, 2020-04-05:
Hi, I've just bought a compact logix 1768-l43/a of ebay for testing. Im just porting the code to work on windows xp as thats whats on the laptop at work, and ws2_32.dll does not support the functions needed by the .dll. Work in progress, Thanks for the update.

kyle-github, 2020-05-05:
If you report back the changes that you had to make for Windows XP, I can try to fold those into the library or at least make the process less painful.   I already have some tweaks to handle older versions of Visual Studio.

kyle-github, 2020-08-01:
Hi @lyndoncoin, any updates on compiling for Windows XP?   If you have any changes to the build process, I would love to fold those in so that others can get this to work!

DonatelloLTU, 2022-05-26:
I know this has been closed, but is there a solution to force Outputs now directly, or do we have to create tag to turn on the Output?

jaypatel099, 2022-10-03:
hi guys do you any way to force on or off AB PLC tags using python or any other language?

lyndoncoin, 2022-10-11:
> Hi. I'm the original person who started this thread. I tried sending lots
> of messages to turn on an output bit lots of times a second. This didn't
> work properly as the output just flickered on and off as it was fighting
> with the output update table. My eventual workaround was to put a boolean
> tag in the ladder logic that turned on the output bit. And then use the lib
> tag library to externally set or reset that book. It wasn't ideal etc as I
> had to modify the ladder logic and I didn't want to do this. I got
> criticism over why I wanted to do this.  But for my application it was very
> useful. I wanted to make a software tool Joe blogs could use such as a
> mechanical fitter et . To aid machine fault diagnosis on our machines,
> without them having the need to have the knowledge of using Allen Bradley
> software etc

*typo bool

lyndoncoin, 2022-10-11:
Hi. I'm the original person who started this thread. I tried sending lots
of messages to turn on an output bit lots of times a second. This didn't
work properly as the output just flickered on and off as it was fighting
with the output update table. My eventual workaround was to put a boolean
tag in the ladder logic that turned on the output bit. And then use the lib
tag library to externally set or reset that book. It wasn't ideal etc as I
had to modify the ladder logic and I didn't want to do this. I got
criticism over why I wanted to do this.  But for my application it was very
useful. I wanted to make a software tool Joe blogs could use such as a
mechanical fitter et . To aid machine fault diagnosis on our machines,
without them having the need to have the knowledge of using Allen Bradley
software etc