libplctag/libplctag #144: Output Debug Msg to File

rafaelkrause, 2020-03-23:
Hi,
Is it possible to output debug message to a file? 

Ex: plc_tag_set_debugfile(const char* );

xinthose, 2020-03-25:
You can implement your own logging (for example Boost C++ logging), else you can use the command `./plc_tag_test_exe 2>&1 | tee output.txt`.  I use the later with `&debug=3` on the tag path for debugging.

rafaelkrause, 2020-03-30:
OK thanks for reply.

kyle-github, 2020-03-30:
I have been thinking about this request.  This conflicts a bit with the existing goal to make the library "slimmer" so that embedded use is possible.   For now, I think @xinthose proposal is probably better.

That said, I will keep this open in case I come up with a lighter weight idea as logging is quite important for debugging!

kyle-github, 2020-05-03:
Now that I have a callback API anyway, what about another callback for logging?   Something like:
```c
int plc_tag_register_logging_callback(int user_handle, void (*callback)(int user_handle, int debug_level, const char *message));
```
Where:
* user_handle - is a value that you pass to the library when register the callback.
* callback - your function.
* debug_level - the debug level of the debugging output.
* message - the character string of the message, formatted as now.

For each debugging output, your callback would be called.  You would be responsible for using the data provided (such as the user_handle parameter) to figure out what you wanted to do.   You would also be responsible for handling the fact that the logging output comes from potentially several different threads.   And for making sure that it does not block.

As a more detailed alternative, I could break the message down into parts:

* debug_level
* thread_id - unique per thread
* function_name - a string with the function name.
* line_num - an integer of the line number in the source
* tag_id - the ID of the current tag, note that this is the same as the tag ID you get back from plc_tag_create().  There are outputs where there is no direct tag so it could be 0 (zero).

Would that be better?

I do not want to try to handle paths and any more IO than console IO.  That is very environment-specific in how it is handled.   Handling log files is an entire library on its own!
  
* What if the path to the file does not exist?  
  * Should the library create it?   
* How would you know if that fails since the output would be sent to the log?  
*  What if the path, including the log file, does exist?   
   * Should it be overwritten?  
* What if you want to rotate log files?   
* What happens if you call plc_tag_set_debugfile() more than once?
* What if you want to write to syslog?

With a callback, you could hook into whatever native logging system you want.  For instance with an appropriate Java wrapper you could hook into Log4J and all the billion different configuration options it has (some of which are actually useful!).   It is more work on your part, but much, much more flexible if you really want to redirect the logs.

Mostly the logs are to help me debug problems.   They can get pretty big, to the point where turning on the full set, DEBUG_SPEW, will actually slow down the library.

Thinking about this a bit more...

I can also expand the use of the existing (new) event callback API so that it is called with library-internal events too.   That would not change the API at all other than adding to the number of defined events.   

You will already get some indication of internal events just based on the existing calls.  For instance, if you try to open a tag that does not exist, you will get a callback that has a status PLCTAG_ERR_NOT_FOUND or similar on the PLCTAG_EVENT_READ_COMPLETED event.

I do not want to just expand the API without carefully thinking about how much it adds to the pain of using the library.   Callbacks are already a lot more complicated than any other data sent to the library today, so I really want to keep them to a minimum.

kyle-github, 2020-05-05:
The new functions are:

```c
int plc_tag_register_logger(void (*log_callback_func)(int32_t tag_id, int debug_level, const char *message));
int plc_tag_unregister_logger(void);
```

See the post in the forum about how to use these.   GitHub seems to be having some problems right now, so the release is not yet up.

kyle-github, 2020-05-05:
This finally went out in version 2.1.5.

closed 2020-05-05 by kyle-github