libplctag/libplctag #150: Response very slow, I don't know where I'm wrong

andreafasoli, 2020-06-13:
I'll try to read a real array Tag from LGX with C# wrapper library, but the response it's very slow, about 200/300 ms. Ispend a lot of time in this two calls:

1)
while (plcClient.GetStatus(tag) == Libplctag.PLCTAG_STATUS_PENDING)
{
Thread.Sleep(1);
}
// minimum 90 ms elapsed

2)
plcClient.ReadTag(tag, 1000);
// Other 100 ms elapsed

I attach debug output with level = 4, can you hel me? Thanks Andrea.
[esito.txt](https://github.com/libplctag/libplctag/files/4774105/esito.txt)

kyle-github, 2020-06-13:
Hi Andrea,

This is not the forum, but that is fine.   It will be found if someone looks.

I went through the output.   That helps but it is a bit confusing.  I see that the library is connecting and getting a response from the PLC in about 10ms.   Repeatedly.  So it is not the library or the PLC that seems to have a problem.

Looking at the debug output shows that your C# code must be doing something wrong.   I see things like this:

```text

Sending the request to the PLC in the IO thread:

2020-05-13 05:13:37.134 thread(30) tag(0) DETAIL send_eip_request:1696 Sending packet of size 64
2020-05-13 05:13:37.134 thread(30) tag(0) DETAIL send_eip_request:1697  70 00 28 00 6f 0d 64 00 00 00
2020-05-13 05:13:37.134 thread(30) tag(0) DETAIL send_eip_request:1697  00 00 00 00 00 00 00 00 00 00
2020-05-13 05:13:37.134 thread(30) tag(0) DETAIL send_eip_request:1697  00 00 00 00 00 00 00 00 01 00
2020-05-13 05:13:37.134 thread(30) tag(0) DETAIL send_eip_request:1697  02 00 a1 00 04 00 fa 44 b7 ff
2020-05-13 05:13:37.135 thread(30) tag(0) DETAIL send_eip_request:1697  b1 00 14 00 01 00 52 05 91 07
2020-05-13 05:13:37.135 thread(30) tag(0) DETAIL send_eip_request:1697  42 42 5f 54 65 73 74 00 32 00
2020-05-13 05:13:37.135 thread(30) tag(0) DETAIL send_eip_request:1697  00 00 00 00
2020-05-13 05:13:37.135 thread(30) tag(0) DETAIL send_eip_request:1731 Done.

Now waiting for the response.   Note the time, 05:13:37.135, the 135 part is the milliseconds.

2020-05-13 05:13:37.135 thread(30) tag(0) DETAIL recv_eip_response:1751 Starting.

2 ms later we start getting the response:

2020-05-13 05:13:37.137 thread(30) tag(0) DETAIL recv_eip_response:1815 request received all needed data (252 bytes of 252).
2020-05-13 05:13:37.137 thread(30) tag(0) DETAIL recv_eip_response:1817  70 00 e4 00 6f 0d 64 00 00 00
2020-05-13 05:13:37.138 thread(30) tag(0) DETAIL recv_eip_response:1817  00 00 00 00 00 00 00 00 00 00
2020-05-13 05:13:37.138 thread(30) tag(0) DETAIL recv_eip_response:1817  00 00 00 00 00 00 00 00 00 00
2020-05-13 05:13:37.138 thread(30) tag(0) DETAIL recv_eip_response:1817  02 00 a1 00 04 00 9a f0 9c 75
2020-05-13 05:13:37.138 thread(30) tag(0) DETAIL recv_eip_response:1817  b1 00 d0 00 01 00 d2 00 00 00
2020-05-13 05:13:37.138 thread(30) tag(0) DETAIL recv_eip_response:1817  ca 00 45 11 38 43 aa 9a 8f be
2020-05-13 05:13:37.139 thread(30) tag(0) DETAIL recv_eip_response:1817  6e b9 75 3f 5e 51 ca 42 d5 29
2020-05-13 05:13:37.139 thread(30) tag(0) DETAIL recv_eip_response:1817  4f 3f b0 65 16 3f ac ce ab 41
2020-05-13 05:13:37.139 thread(30) tag(0) DETAIL recv_eip_response:1817  ed c3 5e bf 4c 48 fc 3e 1c a1
2020-05-13 05:13:37.140 thread(30) tag(0) DETAIL recv_eip_response:1817  7e 43 9b b7 7c bf 84 76 23 be
2020-05-13 05:13:37.140 thread(30) tag(0) DETAIL recv_eip_response:1817  64 7d 94 42 1e 97 cf 3e 96 03
2020-05-13 05:13:37.140 thread(30) tag(0) DETAIL recv_eip_response:1817  6a bf 5a df 99 43 37 cd 7d 3f
2020-05-13 05:13:37.140 thread(30) tag(0) DETAIL recv_eip_response:1817  e8 e8 05 be b2 54 63 43 be 96
2020-05-13 05:13:37.140 thread(30) tag(0) DETAIL recv_eip_response:1817  d7 3e ee 32 68 3f 32 ab 3c 42
2020-05-13 05:13:37.141 thread(30) tag(0) DETAIL recv_eip_response:1817  9a c2 7f bf 12 43 31 bd 9d 15
2020-05-13 05:13:37.141 thread(30) tag(0) DETAIL recv_eip_response:1817  8b 43 b4 be 0c be 02 92 7d 3f
2020-05-13 05:13:37.141 thread(30) tag(0) DETAIL recv_eip_response:1817  f8 dc 47 43 b6 1c ba 3e f2 7c
2020-05-13 05:13:37.141 thread(30) tag(0) DETAIL recv_eip_response:1817  6e bf ff 11 ee 42 0e e2 70 3f
2020-05-13 05:13:37.141 thread(30) tag(0) DETAIL recv_eip_response:1817  13 56 ad be 88 56 79 43 c4 1f
2020-05-13 05:13:37.141 thread(30) tag(0) DETAIL recv_eip_response:1817  d0 be 3c e5 69 bf 6e cc 2a 43
2020-05-13 05:13:37.141 thread(30) tag(0) DETAIL recv_eip_response:1817  f6 f4 cf 3e c1 ee 69 3f 68 c3
2020-05-13 05:13:37.141 thread(30) tag(0) DETAIL recv_eip_response:1817  b1 42 81 b0 1b 3f 53 37 4b 3f
2020-05-13 05:13:37.141 thread(30) tag(0) DETAIL recv_eip_response:1817  56 0a 0d 41 a8 dd 51 bf 45 9a
2020-05-13 05:13:37.141 thread(30) tag(0) DETAIL recv_eip_response:1817  12 3f d4 43 0d 43 1c 8b 7e bf
2020-05-13 05:13:37.141 thread(30) tag(0) DETAIL recv_eip_response:1817  34 29 da 3d 49 41 72 42 0f 64
2020-05-13 05:13:37.141 thread(30) tag(0) DETAIL recv_eip_response:1817  24 bf
2020-05-13 05:13:37.141 thread(30) tag(0) DETAIL recv_eip_response:1824 Done.

I do not know if this is Windows or something else, but it takes several milliseconds just to get the 200+ byte response.  That is a very slow network.   We start at ms 137 and end at 141.

I have left out some internal processing

Then we have the checking function looking at the response, but this time in the thread that owns the tag.

2020-05-13 05:13:37.143 thread(11) tag(35) DETAIL check_read_status_connected:1720 type byte = 202 (ca)
2020-05-13 05:13:37.143 thread(11) tag(35) DETAIL check_read_status_connected:1771 Increasing tag buffer size to 200 bytes.
2020-05-13 05:13:37.143 thread(11) tag(35) INFO check_read_status_connected:1781 Got 200 bytes of data
2020-05-13 05:13:37.143 thread(11) tag(35) DETAIL rc_dec_impl:230 Calling cleanup functions due to call at check_read_status_connected:1805 for 0xb5d0ccc0.
2020-05-13 05:13:37.143 thread(11) tag(35) INFO refcount_cleanup:244 Starting
2020-05-13 05:13:37.143 thread(11) tag(35) DETAIL request_destroy:2406 Starting.
[... cut ...]
Another way to get good performance is to have a thread per tag.   This does not scale as well (particularly on Windows and macOS) but can make your code more clear.   In that case, you use plc_tag_read() in synchronous mode with a timeout but you have multiple threads each with a tag.   The library will do the same request combination into one packet that it does above and the requests will be processed by the PLC much more quickly.

Can you share your code that is reading the tag?

Best,
Kyle

andreafasoli, 2020-06-14:
Hello Kyle, thanks for your response, I've attached my code in C#.
It's true I not call plc_tag_destroy () or  plc_tag_delete ()

Andrea
[codeExample.txt](https://github.com/libplctag/libplctag/files/4775924/codeExample.txt)

andreafasoli, 2020-06-14:
Important,  the previous code it's in a timer scheduler

Andrea

kyle-github, 2020-06-14:
There are a couple of things that stand out in the code:

1. you wait a maximum of 200ms for the library to connect to the PLC.   Connecting to the PLC is the longest duration operation.   Even on a Logix-class PLC with a good network it can take 50ms when a read takes 7ms.   Change your retry times to something much larger.  I often use 5000ms.  In your case each retry count takes at least one millisecond, so change the retry count.
2. do you see any errors logged from your code?   There are multiple points where errors are checked.
3. you are reading tags serially.   For each tag you call ReadTag with a timeout.   That causes the library to wait up to the timeout time for the PLC to respond.   If you want to read many tags, this is the slowest way to run the code, unless each tag is being read in a different thread.
4. Do you have any code that is using the data coming back from the tag?   In the log I see the result from the read coming back every time within a few milliseconds but then no use of it.   I also do not see anything about plc_tag_read() completing which says to me that perhaps you are not getting that far?
5. which C# wrapper are you using?   There are several and some have an included DLL for the core library that may be out of date.   Please make sure you are using a core DLL that is new from the [libplctag](https://github.com/libplctag/libplctag) GitHub project.  Check the Releases section for Windows binaries.   These will also include some pre-built example programs.

I think that at least one of the C# wrappers calls plc_tag_destroy() when the wrapper object is destroyed.   But I only see that sometimes in the log.

At least the time explains why the thread numbers are so high.   This is clearly not the first part of the log.

I hope this helps point you in some directions to try.

Best,
Kyle

andreafasoli, 2020-06-15:
Hi Kyle, I followed your guidelines and the results have improved a lot, thank you very much; I am attaching a trace of the current processing times, I am reading a single Tag (an array of 30 real data).

START: it is the reading cycle start time

BEFORE READ: it is the time immediately before the data reading call:
rc = plcClient.ReadTag (tag, 5000)

AFTER READ:
is the time immediately after the data read call:
rc = plcClient.ReadTag (tag, 5000)

END:
it is the end of reading cycle time after processing the data

READ TIME:
is the time in ms used for the data reading operation:
rc = plcClient.ReadTag (tag, 5000)

TOTAL TIME:
is the time in ms used for the entire reading cycle.

The time of the call makes the results vary a lot:

rc = plcClient.ReadTag (tag, 5000)

sometimes very fast (4 ms), other times much slower (190 ms), depends on the position in the cycle time of the Plc compared to when the call was made? Or is it the C # wrapper that makes reading time vary? There seems to be a cyclical repetition in the times of the wrapper read call.

Thanks for the always very useful suggestions,
Andrea
[TestReading.txt](https://github.com/libplctag/libplctag/files/4780534/TestReading.txt)

kyle-github, 2020-06-16:
That's a very strange pattern of latency.   

The library is designed to let you hold onto a tag (actually a tag handle) for a long time.   I have programs that keep tags open for weeks.  The library will handle all of the timeouts and reconnection logic internally for you.

Setting up and tearing down a connection to the PLC is very slow process.  It is not like HTTP or other light-weight TCP-based protocols.   You really want to avoid doing this on every access.   I think that the garbage collector is running somewhat asynchronously to your program and sometimes cleans up the tag (triggering the PLC connection teardown) before the next read.

I think it would be good to start with something simpler.   I suggest a different structure to your program.   Once this works then you can extend it.   I am going on the assumption that the C# wrapper object is not thread-safe.  I did not write it so I cannot be sure.  If it thread-safe, then you can keep the current structure, but create the tags in the main part of the program and make sure that they are accessible to the code running when the timer triggers.  

Here is an outline of how I would approach this with a simple threaded design.

1) Create a thread for each tag you want to read, assuming that it is not too many tags.   Windows tends to be less performant with threads than Linux but you should be able to create a few hundred without too much trouble.   There are examples in the source on GitHub of multithreaded programs in C.   While the code will not be the same, hopefully the structure will give you some ideas.

2) each thread will create the plcClient object it will use and the tag it is going to read.  It does this just once when it is created, not on every read cycle.

3) then each thread will go into a loop where it first reads the tag, then processes the data, then sleeps for the required delay.   A new tag is not created in the loop.  The tag is not destroyed.

4) when each thread is terminated, the runtime _should_ clean up the tags (I hope).   However, if you can cleanly signal your threads to terminate, then you can call the tag destructor too.   Either way it will all be cleaned up.

 Here is some pseudo-code (I am not a C# programmer) for the inner loop of the thread:

```c

   // Create tag here outside the loop.
    plcClient = new PLC...
    plcClient.add(tag);

    while(true) {
          startTime = ... get system time in milliseconds ...
          plcClient.Read(tag, 5000);

         ... check status ...

         ... process data ...

         ... calculate delay based on current time and start time ...
         Sleep(delay);
    }
```

I _think_ that I understand what is happening with the strange timings.   But the magnitude of the delay in some cases is a lot larger than I expect.

START 03:06:12.006 BEFORE READ 03:06:12.007 AFTER READ 03:06:12.191  END 03:06:12.191 READ TIME 184.0504 ms TOTAL TIME 184.5298 ms

The numbers above ^^^^ all add up.  So far so good.   If there is some contention on the PLC I would expect to see a much faster connection time than this but Windows keeps surprising me.   I would expect a connection time of around 40-50ms, not 180+.

START 03:06:12.208 BEFORE READ 03:06:12.208 AFTER READ 03:06:12.215  END 03:06:12.216 READ TIME 7.2659 ms TOTAL TIME 7.9262 ms

This is what I would expect to see for all the reads after this, about 7-8ms.   

START 03:06:12.403 BEFORE READ 03:06:12.403 AFTER READ 03:06:12.494  END 03:06:12.495 READ TIME 91.2275 ms TOTAL TIME 91.8482 ms

Hmm, again this is a long time for a connection to the PLC to be set up.   Also, this is 200ms later than the previous read.   So this could be a 200ms timer?

START 03:06:12.607 BEFORE READ 03:06:12.607 AFTER READ 03:06:12.791  END 03:06:12.791 READ TIME 184.1232 ms TOTAL TIME 184.4672 ms

Ah, looks like you are reading every 200ms.   Normally that would not be a problem, but this PLC connection set up time is really long.  On my bad home network I rarely see more than 40ms.   Is your PLC serving a lot of network traffic?   I have seen an ENBT be overrun by traffic before but usually it just drops packets.

My guess is that the garbage collector ran before this and caused the tags to be destroyed.

START 03:06:12.794 BEFORE READ 03:06:12.794 AFTER READ 03:06:12.798  END 03:06:12.799 READ TIME 4.4132 ms TOTAL TIME 4.7278 ms

And again we see a much faster response the second time.  So the GC probably has not run yet. 

One thing that is weird here is that this is close but not quite 200ms apart.  It is less than I expected, but timers are not that accurate.

START 03:06:13.003 BEFORE READ 03:06:13.003 AFTER READ 03:06:13.094  END 03:06:13.095 READ TIME 91.2746 ms TOTAL TIME 91.908 ms

If the GC ran and cleaned everything up, then we get to reconnect to the PLC again.   So this is long but only about half the time of the really long set up time...  Strange.

START 03:06:13.201 BEFORE READ 03:06:13.201 AFTER READ 03:06:13.392  END 03:06:13.393 READ TIME 191.5567 ms TOTAL TIME 192.1805 ms
[... 10 lines cut ...]

20200615_130614001 - START 03:06:13.794 BEFORE READ 03:06:13.794 AFTER READ 03:06:14.000  END 03:06:14.001 READ TIME 206.5047 ms TOTAL TIME 207.1455 ms

This is not good.  It took more than the timer period to set up the PLC connection.   At this point you probably have multiple tags overlapping.

[... cut ...]
Is your PLC's network module overloaded?   You can get a lot of useful network statistics using a browser by navigating to the IP address of the network module (port 80).   On the Diagnostic pages you can find numbers of connections and you should also be able to see the CPU load of the network module.   Anything over 75% or so and you are probably overloading the module.  I prefer to keep the load under 50% otherwise produced/consumed tag performance can suffer.   HMIs can place a huge load on the network module.  I often add a module just for the HMI on a separate IP address and switch port.

Best,
Kyle

andreafasoli, 2020-06-16:
Hello Kyle I find a difference in response time;

I tried to simplify everything as you suggested, I'm trying to make the call from the main, a simple cycle in a single thead. If between a call and the other I use a fixed waiting time, the system always answers with almost equal times, if the waiting time varies the system varies the answers, is there something wrong in the cycle? I also tried on embedded linux by compiling the library on an arm32 processor and I have the same result. I enclose the code of the two call modes with the related logs, .NET core 3.1 development environment (that's what I tried on both linux and Windows). 

[Fixed and dynamic way.txt](https://github.com/libplctag/libplctag/files/4787353/Fixed.and.dynamic.way.txt)

Thanks Andrea

kyle-github, 2020-06-17:
Hi Andrea,

This is quite strange.  The code is simple enough that there should not be anything odd happening.  That pattern of times is almost the same as before, with one outlier that is over 400ms. 

I think we need to get a better idea of what is happening without any other layers first.  Please try to capture all of the logging again with the simple program.   I would like to see where the delays are happening.

Another thing to try is to increase the delay between runs to about 500ms.   I would be interested to see if the delays are all long or if we see the same pattern.

Your code is very simple now so there is no place for thread-based timing problems to hide! 

Have you checked the load on your network module in the PLC?

What is your network topology?   If the network is a bit complicated is it possible to simplify it to eliminate as many other variables as possible?

The fact that you are seeing this with both Linux and Windows seems to point to a common problem between the different systems, but I have not seen one like this before.  I know of other users that have used RPi (ARM) systems, both 32 and 64-bit and with Linux of Windows IoT, so I doubt that there is anything platform specific.

I ran my stress_test.c program to get some recent data.   I have a fairly bad network.   My laptop goes across WiFi (which is being used by my family to stream video right now) and then to an old wired network where I have my lab.   The latency here ranges from 1ms to 50ms with ping.   And there is packet loss on the WiFi network.   While my network latency and variability is not good, I do not have any misconfigured hardware or broken hardware and I test all my network cables.

With 10 threads each reading and writing 10 elements in a tag in the PLC as fast as they can, I get a throughput of  over 2500 iterations (an average of 256 iterations per thread) in 10 seconds.   The longest delay I saw was 178ms, but the vast majority of the time the latency was around 36ms.  In fact the average for every thread was exactly 36ms.

That is what I expect to see with the current version of the library on Linux.

Best,
Kyle

andreafasoli, 2020-06-17:
Hello Kyle, I tested the system on linux arm 32 device, with library compiled for embedded system, as suggested. I have tried the following combinations:

- fixed wait 200 ms in the cycle
- fixed wait 500 ms in the cycle
- dynamic waiting max 200 ms in the cycle
- dynamic waiting max 500 ms in the cycle

For each of the 4 tests I created the following logs that I attach:

- xxxms_y_log.txt
- xxxms_y_ping.txt
- xxxms_y_time.txt

where is it:

xxx = waiting time in ms

y = Fixed (Fixed sleep time) or Dynamic (Max Sleep Time - cycle time)
See attach file example fixed time code (fixed_waiting_code_example.txt) the same of previous test, and example dynamic time code (dynamic_waiting_code_example.txt) the same of previous test

...._ log = output log library with debug = 4

...._ ping = time of ping to the Plc during the test

...._ time = output from code example with times for reading tag (see fixed_waiting_code_example.txt and dynamic_waiting_code_example.txt)

Thsanks for all,  your support is highly appreciated,
Andrea

[dynamic_waiting_code_example.txt](https://github.com/libplctag/libplctag/files/4793010/dynamic_waiting_code_example.txt)

[fixed_waiting_code_example.txt](https://github.com/libplctag/libplctag/files/4793011/fixed_waiting_code_example.txt)

[200ms_Dynamic_log.txt](https://github.com/libplctag/libplctag/files/4793014/200ms_Dynamic_log.txt)

[200ms_Dynamic_ping.txt](https://github.com/libplctag/libplctag/files/4793016/200ms_Dynamic_ping.txt)

[200ms_Dynamic_time.txt](https://github.com/libplctag/libplctag/files/4793017/200ms_Dynamic_time.txt)

[200ms_Fixed_time.txt]
(https://github.com/libplctag/libplctag/files/4793020/200ms_Fixed_time.txt)

[200ms_Fixed_log.txt]
(https://github.com/libplctag/libplctag/files/4793018/200ms_Fixed_log.txt)

[200ms_Fixed_ping.txt]
(https://github.com/libplctag/libplctag/files/4793019/200ms_Fixed_ping.txt)

[500ms_Dynamic_log.txt](https://github.com/libplctag/libplctag/files/4793023/500ms_Dynamic_log.txt)

[500ms_Dynamic_ping.txt](https://github.com/libplctag/libplctag/files/4793024/500ms_Dynamic_ping.txt)

[500ms_Dynamic_time.txt](https://github.com/libplctag/libplctag/files/4793025/500ms_Dynamic_time.txt)

[500ms_Fixed_log.txt]
(https://github.com/libplctag/libplctag/files/4793030/500ms_Fixed_log.txt)

[500ms_Fixed_ping.txt]
(https://github.com/libplctag/libplctag/files/4793031/500ms_Fixed_ping.txt)

[500ms_Fixed_time.txt]
(https://github.com/libplctag/libplctag/files/4793032/500ms_Fixed_time.txt)

kyle-github, 2020-06-18:
Hi Andrea, 

Thanks for being patient on this.

I went through the logs you sent.   Below I am using the log from the C library for the 500ms dynamic test.  I am using the library log because it has many activities in multiple threads so you can see when things are happening.  First we see some lines showing that tag creation is starting up.  These are in thread 1, which is your program.

```text
2020-05-17 13:31:45.005 thread(1) tag(0) INFO find_tag_create_func:82 Matched protocol=ab_eip
2020-05-17 13:31:45.005 thread(1) tag(0) INFO ab_tag_create:189 Starting.
```

Then we have some output that shows the parsing of the tag name and other set up.   I left that out.  Then we have a couple of lines that show the plc_tag_create() routing returning (inside the C# wrapper), a brief delay and then thread 2 starting.  Thread 2 is the IO thread for that PLC.  It will handle all packets going in and out and each tag for that PLC in the library is multiplexed across that IO thread.

We can see the IO thread determining the IP address of the PLC, opening a TCP connection to it and finally registering an EIP session.  Note the timing.   It takes 10ms to do all of that.  So we definitely got a response from the PLC (several in fact) in 10ms.

```text
2020-05-17 13:31:45.008 thread(1) tag(11) INFO plc_tag_create:579 Returning mapped tag ID 11
2020-05-17 13:31:45.008 thread(1) tag(11) INFO plc_tag_create:581 Done.
2020-05-17 13:31:45.009 thread(2) tag(0) INFO session_handler:928 Starting thread for session 0x1b8540
2020-05-17 13:31:45.009 thread(2) tag(0) DETAIL session_handler:947 in SESSION_OPEN_SOCKET state.
2020-05-17 13:31:45.009 thread(2) tag(0) INFO session_open_socket:586 Starting.
2020-05-17 13:31:45.010 thread(2) tag(0) DETAIL socket_create:766 Starting.
2020-05-17 13:31:45.010 thread(2) tag(0) DETAIL socket_create:780 Done.
2020-05-17 13:31:45.011 thread(2) tag(0) DETAIL socket_connect_tcp:799 Starting.
2020-05-17 13:31:45.011 thread(2) tag(0) DETAIL socket_connect_tcp:857 Found numeric IP address: 192.168.1.101
2020-05-17 13:31:45.012 thread(2) tag(0) DETAIL socket_connect_tcp:905 Attempting to connect to 192.168.1.101
2020-05-17 13:31:45.013 thread(2) tag(0) DETAIL socket_connect_tcp:910 Attempt to connect to 192.168.1.101 succeeded.
2020-05-17 13:31:45.013 thread(2) tag(0) DETAIL socket_connect_tcp:949 Done.
2020-05-17 13:31:45.013 thread(2) tag(0) INFO session_open_socket:603 Done.
2020-05-17 13:31:45.014 thread(2) tag(0) DETAIL session_handler:964 in SESSION_REGISTER state.
2020-05-17 13:31:45.014 thread(2) tag(0) INFO session_register:616 Starting.
2020-05-17 13:31:45.014 thread(2) tag(0) DETAIL send_eip_request:1683 Starting.
2020-05-17 13:31:45.014 thread(2) tag(0) DETAIL send_eip_request:1696 Sending packet of size 28
2020-05-17 13:31:45.015 thread(2) tag(0) DETAIL send_eip_request:1697  65 00 04 00 00 00 00 00 00 00
2020-05-17 13:31:45.015 thread(2) tag(0) DETAIL send_eip_request:1697  00 00 00 00 00 00 00 00 00 00
2020-05-17 13:31:45.016 thread(2) tag(0) DETAIL send_eip_request:1697  00 00 00 00 01 00 00 00
2020-05-17 13:31:45.016 thread(2) tag(0) DETAIL send_eip_request:1731 Done.
2020-05-17 13:31:45.016 thread(2) tag(0) DETAIL recv_eip_response:1751 Starting.
2020-05-17 13:31:45.018 thread(2) tag(0) DETAIL recv_eip_response:1815 request received all needed data (28 bytes of 28).
2020-05-17 13:31:45.018 thread(2) tag(0) DETAIL recv_eip_response:1817  65 00 04 00 be 13 5f 00 00 00
2020-05-17 13:31:45.019 thread(2) tag(0) DETAIL recv_eip_response:1817  00 00 00 00 00 00 00 00 00 00
2020-05-17 13:31:45.019 thread(2) tag(0) DETAIL recv_eip_response:1817  00 00 00 00 01 00 00 00
2020-05-17 13:31:45.019 thread(2) tag(0) DETAIL recv_eip_response:1824 Done.
```

Now we move to the next stage of the handshake between us and the PLC: the CIP connection set up.  This is a negotiation of various parameters including how large the request packet can be.

```text
2020-05-17 13:31:45.023 thread(2) tag(0) DETAIL send_eip_request:1683 Starting.
2020-05-17 13:31:45.023 thread(2) tag(0) DETAIL send_eip_request:1696 Sending packet of size 92
2020-05-17 13:31:45.024 thread(2) tag(0) DETAIL send_eip_request:1697  6f 00 44 00 be 13 5f 00 00 00
2020-05-17 13:31:45.024 thread(2) tag(0) DETAIL send_eip_request:1697  00 00 ac f9 cd 00 00 00 00 00
2020-05-17 13:31:45.024 thread(2) tag(0) DETAIL send_eip_request:1697  00 00 00 00 00 00 00 00 01 00
2020-05-17 13:31:45.025 thread(2) tag(0) DETAIL send_eip_request:1697  02 00 00 00 00 00 b2 00 34 00
2020-05-17 13:31:45.025 thread(2) tag(0) DETAIL send_eip_request:1697  5b 02 20 06 24 01 0a 05 00 00
2020-05-17 13:31:45.025 thread(2) tag(0) DETAIL send_eip_request:1697  00 00 78 5a bf 00 40 85 3d f3
2020-05-17 13:31:45.026 thread(2) tag(0) DETAIL send_eip_request:1697  45 43 50 21 01 00 00 00 40 42
2020-05-17 13:31:45.027 thread(2) tag(0) DETAIL send_eip_request:1697  0f 00 a2 0f 00 42 40 42 0f 00
2020-05-17 13:31:45.027 thread(2) tag(0) DETAIL send_eip_request:1697  a2 0f 00 42 a3 03 01 00 20 02
2020-05-17 13:31:45.027 thread(2) tag(0) DETAIL send_eip_request:1697  24 01
2020-05-17 13:31:45.028 thread(2) tag(0) DETAIL send_eip_request:1731 Done.
2020-05-17 13:31:45.028 thread(2) tag(0) INFO send_forward_open_req_ex:2159 Done
2020-05-17 13:31:45.028 thread(2) tag(0) INFO recv_forward_open_resp:2172 Starting
2020-05-17 13:31:45.028 thread(2) tag(0) DETAIL recv_eip_response:1751 Starting.

... snip ...

[... cut ...]
The fact that you are getting the same kinds of results for both the PC and the ARM computer seem to indicate that the problem lies closer to the PLC.  Whatever the problem is, it is not specific to a certain OS or CPU.

One way to ensure that external network factors are removed would be to build an isolated network with just the PLC, a single network switch and the PC on it. No other links.   If that repeatedly shows the problem, then it is most likely the PLC.

Sorry I cannot offer more help here!   This is a very confusing problem.

Best,
Kyle

andreafasoli, 2020-06-19:
Hello Kyle I did another test, if this can help us, I compiled the library with a simple test function inside, what I added to lib.c is in the annex "lib.c with test function.txt", compiled the everything is loaded on the ARM embedded system, I apologize for my C ....

I then changed the main to net core with that shown in "main net.txt", a simple and unique call to the library.

I attach the log out of the printf "log inside library.txt" and a log made at the same time as a ping to the plc "ping at the same time test.txt".

Can this suggest anything?
Thanks Andrea

[log inside library.txt](https://github.com/libplctag/libplctag/files/4803923/log.inside.library.txt)
[main net.txt](https://github.com/libplctag/libplctag/files/4803924/main.net.txt)
[lib.c with test function.txt](https://github.com/libplctag/libplctag/files/4803925/lib.c.with.test.function.txt)
[ping at the same time test.txt](https://github.com/libplctag/libplctag/files/4803926/ping.at.the.same.time.test.txt)

kyle-github, 2020-06-23:
Hi Andrea,

I am not seeing anything obvious in the logs.   It is interesting that the delays are in a different pattern than before.   There are some extremely long delays.  Even one that looks like a full second.

Have you looked at the web page on the network module?   http://192.168.1.101/

Look at the Diagnostics section.  There will be some pages that have data like packet rates, CPU load, number of connections etc.   What do those show?  Is the CPU spiking in load?   Is there anything else that reads your PLC periodically?

Can you capture some logs with debug level 5?   That will output a lot of logs, but it will have outputs from several very low level routines that run often.  If they are delayed or missing, then that may indicate a problem with the library.

You an also modify either the async_stress.c program or the stress_test.c  program to run multiple reads on one thread and output the timing.  That would be all in C.

Also, it may be useful to do a TCP packet capture to see if packets are coming from the PLC to the PC/embedded system and when they are coming.   If there is delay in the incoming packets, the the problem is not in the PC/embedded system.

Best,
Kyle

andreafasoli, 2020-06-23:
Hello Kyle, thanks for the answer, at the same time I had tests done with IBA analyzer ( https://www.iba-ag.com/ibaanalyzer/ ) and the response times are very varied also with iba. A big difference occurred using iba by querying the plc directly with the ethernet IP protocol without going through the LGX module, times with iba went from the average of 160 ms to 10 ms. I don't know if this is of interest, something indicates. I attach the data screens on the plc in the indicated web page.

Now I try to activate the log with debug with value = 5. Then I send the resulting logs.

The library use LGX wrapper schema or ask directly data with ethenet ip protocol to the plc?

Thanks, Andrea
[image]
[image]
[image]
[image]
[image]
[image]

kyle-github, 2020-06-24:
Hi Andrea,

If you are seeing wildly different response times with the other tool, then it seems very likely that it is your network that is the problem.

The library works best when it can connect directly to the PLC.   The AB protocol is EIP (EtherNet/IP, a wrapper on top of TCP/IP) that encapsulates CIP which is the protocol that the PLC actually speaks.

I looked a little bit through your images.   I saw a few things that look like they might be problems:

1.  On the "Ethernet Statistics" tab, I see that the number of MAC errors, "MAC Receive Errors", is not zero.   Is that number changing quickly?   If I understand correctly, that is a measure of the low level error rate of the Ethernet hardware.   On my network modules, that number is zero even after weeks of uptime.
2. On the "Bridge Connections" tab, in the "Missed Rx Pkts" column, the numbers should be very low and they are very high.  This is a measure of how many packets the PLC expected to receive but did not.   It looks like the connection to 192.168.1.123 is very bad.

Perhaps you have a bad Ethernet cable?  Or perhaps the PLC is trying to negotiate full duplex on a half-duplex link or vice versa?

Unfortunately, I have been out all day and have not had time to look over your latest response in detail.   Sorry for the delay. I will try to get to it tomorrow.

Best,
Kyle

andreafasoli, 2020-06-24:
"MAC Receive Errors" changes very frequently for example now it is 45665 now I am testing device 192.168.1.123

kyle-github, 2020-06-25:
Hi Andrea,

I looked through the other screens.   The only other thing I noticed is that your PLC has 25 active TCP connections.   While the CPU load is fairly low, that means that other PLCs or devices are connected to that PLC.   They will add network traffic.  You have 7 active CIP connections.   Those take up bandwidth and internal resources as well.  None of those numbers are particularly bad, but if you have a bad network cable or a bad device in your network (it seems to be set up as a ring?), then all those connections and traffic will be impacted.

When a packet is lost, TCP will resend it.   TCP works really nicely when you have a network with very low packet loss.   When packet loss starts to get higher, TCP may make things worse.

If your network is really in a ring, then it is possible that you have one bad cable.   That is an unfortunate problem with rings, one bad cable may impact everything on the ring.   I prefer star topologies (with a redundant center) for that reason.   If you have a ring with dual connections at each node, that usually works well too.

There are many things that could cause a periodic problem:

1. Duplicate IP addresses.  If two devices have the same IP address, you can have problems where first one device then the other responds to the packets.   Normally this is more irregular in timing and has a longer period, so I think it might not be the problem.
2. A loop in your network spanning tree.   This definitely can cause periodic delays like you see here.   I think the period of the standard spanning tree protocol is several seconds though and this is much faster. I know that AB switches like Stratix switches can use a faster spanning tree protocol takes much less time to reconfigure itself.
3. An Ethernet cable with a bad channel on a port that is full duplex.   Normally a switch will autoconfigure the duplex (full or half) and speed, but if the cable is bad or damaged, the autoconfiguration system can keep retrying and retrying and retrying...   You will see a lot of errors in just one direction (either TX, transmit, or RX, receive).   You do have some errors like that.   

I hope that this helps!

Best,
Kyle

kyle-github, 2020-08-03:
@andreafasoli I thought I would check back in to see if you had found anything new on this problem.

Best,
Kyle

andreafasoli, 2020-08-24:
Hi Kyle, thanks for the report, these days I will try the new library and let you know if the results improve. Thanks!!

kyle-github, 2020-09-26:
HI Andrea,

Have you had a chance to try a newer version to see if it works better?

kyle-github, 2020-12-05:
Please reopen this if you have additional data!

closed 2020-12-05 by kyle-github