libplctag/libplctag #151: Master branch broke MicroLogix 1400 communication

xinthose, 2020-06-22:
FYI, using the latest master branch, I could not connect to my MicroLogix 1400, but with release 2.1.6 I could.  I kept on getting tag timeouts.

kyle-github, 2020-06-22:
Well, that is not good!  Can you attach debug (level 4) output from both?  I am suspicious that one of my changes to support DH+ bridges broke that.

Thanks for testing and reporting this!

kyle-github, 2020-06-23:
I just did another test with my PLC/5 direct via Ethernet and via a DHRIO in a ControlLogix chassis and then DH+ and they both work :-(

Do you have a path element when hitting your MicroLogix?   That will cause the library to think that you are trying to bridge to DH+.

I combined two different changes in this.  One was for support of Ethernet routing and the other was for bridged DH+ support for SLC 500 and MicroLogix.   It could be either one that was the problem.  

If you are able to send me debug logs from both 2.1.6 and 2.1.7, I would really appreciate it!

kyle-github, 2020-06-25:
Another user ran into a problem.  The DH+ addresses were not being handled correctly.   That probably does not impact your bug.  But just in case, version v2.1.8 is now out as a prerelease.

xinthose, 2020-06-25:
Hello.  I have the log output here: [test.log](https://github.com/libplctag/libplctag/files/4831779/test.log).  I also just made you an outside collaborator on my test project which generated the code: https://github.com/EleMech/plctag.  I used the latest master branch again and got the same error `PLCTAG_ERR_TIMEOUT`.

kyle-github, 2020-06-25:
Thanks!  I will dig through this.

kyle-github, 2020-06-25:
I see that you may have a connection path?

Here is where we are trying to set up a connection with the MicroLogix.  Normally that isn't done at all unless there is a path and then we assume that it is via DH+.   What is the path string you are sending?

2020-05-25 08:56:58.851 thread(3) tag(0) WARN recv_forward_open_resp:2215 Forward Open command failed, response code: Unsupported service. (8)

This is normal.   The library tries to set up a large packet using FOpenExtended.  That fails with a response of error code 0x08.   It is odd that the "DETAIL" output is missing at this stage.  That has the packet bytes in it.

2020-05-25 08:56:58.859 thread(3) tag(0) WARN recv_forward_open_resp:2215 Forward Open command failed, response code: Invalid Segment in Connection Path (1)

And this is from the second attempt trying to use the old ForwardOpen packet.   That fails because there is something wrong with the path.

As I mentioned though, there shouldn't be a path unless you are trying to use DH+.

I am confused why the debug level seems to be dropping back to level 3 (I see INFO output but no DETAIL output) after the first few lines.   Are you setting the debug level in your code somewhere?   Maybe in the tag string?

The timeout is coming because there is an error status that is not propagating up to the top so the tag sits there and waits for the background thread to connect.   I need to figure out how to fix that.

Is this PLC on another network that is bridged from the one the PC is on?

xinthose, 2020-06-25:
I am sorry, I do not understand this:
> Here is where we are trying to set up a connection with the MicroLogix. Normally that isn't done at all unless there is a path and then we assume that it is via DH+. What is the path string you are sending?

My path is in the bottom of the log: `protocol=ab_eip&gateway=10.0.0.5&path=1&cpu=mlgx&elem_size=4&elem_count=1&name=F8:0&share_session=1&debug=3`
Is that not correct?

The PLC is on the same network as my Linux PC (Ubuntu 20.04 LTS, 64-bit).

kyle-github, 2020-06-26:
Apologies, I managed to cut and paste the log file such that I left off the bottom few lines. Including the command line etc. :-(

The tag string appears to be: "protocol=ab_eip&gateway=10.0.0.5&path=1&cpu=mlgx&elem_size=4&elem_count=1&name=F8:0&share_session=1&debug=3"

A couple things here:

1. "debug=3" will take effect when the tag is created.  So if you called plc_tag_set_debug_level() earlier than that, this will override that setting :-(
2. You have a path for the MicroLogix. path=1.   I will need to pull my code to see what it did with a path for any PCCC-based PLC.   For PLC/5 it is not needed or used.  I wonder if I just simply left it out.    

Right now, the new logic in the library is simple: if it is a PCCC tag and there is a path, then it is using a DH+ bridge.  That may be too simplistic.   The "docs" are not clear on this and I only have a PLC/5 to test with.   I will need to rethink this because with the new support for Ethernet bridging, you can probably route through a ControlLogix chassis to get to a remote PLC/5 just using Ethernet.   The more I think about it, the more that makes sense.

Let me look at the old code to see if the path just got ignored for MicroLogix.

Thanks for your patience!

kyle-github, 2020-06-26:
I checked out the 2.1.6 code and it just ignores any path for SLC or MicroLogix.   Grrr...

How painful is this for you?   Do you have paths in most of your tag strings?   I can try to make this be ignored again unless it is a DH+ address.

I definitely do not want to break any existing production code!

xinthose, 2020-06-26:
OK, so `&path=1` was never needed for `&cpu=mlgx`?  I did not know that.  None of the examples use a `mlgx`.  It would break all of my previous code until I removed that path.

kyle-github, 2020-06-26:
Yep, paths are not needed (they must be ignored as they will break the protocol if they are included) for PLC/5, SLC 500 and apparently MicroLogix.

I will fix this on my end.

kyle-github, 2020-06-26:
I have rolled back the master and release branches to 2.1.6.  Until I come up with a good way to allow (and ignore) paths that are not DH+ or IP bridging, I will not check back into them.

Again, very sorry about breaking this for you!

xinthose, 2020-06-26:
OK.  Thank you for your help Kyle.  Glad we found the issue.  Should I close this now?

kyle-github, 2020-06-26:
Not yet.  If you could double check that the tip version on both the master and release branches works for you, I would really appreciate it!

While it works for me, so did the change that broke your code!

If that test works then we can close this.

Thanks again for your patience and for reporting this so quickly!

xinthose, 2020-06-29:
You're welcome.  I have verified that master is working for me again.  I would think if the cpu was `mlgx` you could just ignore the `path` parameter.

closed 2020-06-29 by xinthose

kyle-github, 2020-06-30:
Thanks for testing!

The problem is that this work was to enable MicroLogix and SLC 500 PLCs to be used on the other side of a DH+ bridge.   There was support for PLC/5 before for this.   The logic I used was exactly what you stated.  Unfortunately there are now situations where there _can_ be a path with a MicroLogix or SLC 500.   I think I have a solution that lets all of this work in the prerelease branch.

If you do have any spare time to test and could test with the prerelease branch, I would appreciate it.   Thanks again!

xinthose, 2020-06-30:
OK.  I see.  You're welcome Kyle.