libplctag/libplctag #154: plc_tag_get_debug_level()

timyhac, 2020-07-07:
In the documentation it talks about being able to retrieve debug level by calling `plc_tag_get_debug_level()` but this function is not available.

kyle-github, 2020-07-07:
Definitely a doc bug.

kyle-github, 2020-07-13:
I think I have found all the references to plc_tag_get_debug_level() and removed them.   I also updated the code to support both attributes: 'debug_level' and 'debug' with a warning when you use 'debug_level'.

timyhac, 2020-07-13:
Looks good Kyle, thanks for clarifying - I'll update the wrapper with the preferred naming.

kyle-github, 2020-07-13:
The fix is in 2.1.10 IIRC, so it is not released yet.   I did update the API docs.

kyle-github, 2020-07-13:
Yep, in the prerelease branch.   That has the new Modbus stuff in it too.

kyle-github, 2020-07-25:
Fixed in release 2.1.10.

closed 2020-07-25 by kyle-github