libplctag/libplctag #157: Question: how to read the tag from Rockwelll Micro 850 (2080-LC50-24QWB)?

abonent312, 2020-07-16:
Hi! Pls give me a hint on how to correctly build the path string for Rockwelll Micro 850 (2080-LC50-24QWB)?
I used the next form:

kyle-github, 2020-07-17:
Hi,

Thanks for using the library.  The error log is very helpful, thanks for including it.  It is hard to see in the output:

```text
2020-06-16 22:19:11.555 thread(1) tag(0) WARN ab_tag_create:371 A path is not supported for this PLC type.
```

Leave the path off completely, and it should connect if you have all the other parts correct.   The Micro800 series is weird.   These PLCs only support a limited subset of the command protocol used with ControlLogix systems.

```text
protocol=ab-eip&gateway=192.168.5.20&cpu=micro800&elem_count=1&elem_size=4&debug=4&name=L19_0_8_DispActWeight[1]
```

This should probably be a fatal error instead of a warning.   It looks like the code goes on to try to make a connection to the PLC, which then fails some time several milliseconds later.

I am going to mark this as a bug.   The error should be a lot more clear and the code should not continue past the point that the error is logged.

Thanks for reporting this!

kyle-github, 2020-07-17:
BUG: When a path is passed to a Micr800, cause plc_tag_tag() to fail at the point that the path is checked and the warning logged.

abonent312, 2020-07-17:
If I remove "path"-sentence completely and use the string:
`protocol=ab-eip&gateway=192.168.5.20&cpu=micro800&elem_count=1&elem_size=4&debug=4&name=L19_0_8_DispActWeight[1]`
I see another error:
`2020-06-17 07:51:32.727 thread(3) tag(0) WARN recv_forward_open_resp:2193 Forward Open command failed, response code: Invalid Segment in Connection Path (1)`
 (log is attached 
[libplctag_no_path.txt](https://github.com/libplctag/libplctag/files/4936456/libplctag_no_path.txt)
 )

What are the segments of the "Connection path"? What could be probably wrong: "elem_count", "elem_size" or "name" ?

I also read all the available classes by Ethernet/IP (by another library). I assume here is all the settings of PLC... (if it helps)
[plc_5_20_scan_09_08.txt](https://github.com/libplctag/libplctag/files/4936477/plc_5_20_scan_09_08.txt)

thank you for your support!!

kyle-github, 2020-07-17:
This is confusing to me.  There is no connection path in the Forward Open packet!

I will need to experiment with older versions of the library to see if there is something that changed.   I think you have found a bug!

Which version of the library are you using?

abonent312, 2020-07-17:
the lastest commit from master..
d4afe282c49574f5ffec5db345303a552acbe089

kyle-github, 2020-07-18:
This is a regression that was added when I refactored some of the path handling in order to fix some of the DH+ support.   

I think I have this fixed on the prerelease branch.   Can you check that out and try it?   Thanks again for reporting this!

abonent312, 2020-07-20:
Hi Kyle, it works now.
[libplctag_no_path_7b43fd9.txt](https://github.com/libplctag/libplctag/files/4948351/libplctag_no_path_7b43fd9.txt)

`2020-06-20 14:39:48.286 thread(1) tag(11) INFO plc_tag_read:1073 Done
Tag has 1 elements each of 4 bytes.
data[0]=3922`

3922 is a correct value, thank you!

the path I used:
`eip&gateway=192.168.5.20&cpu=micro800&elem_count=1&elem_size=4&debug=4&name=L19_0_8_DispActWeight[1]`

PR #158 (2020-07-18): Fix #157, Micro800 regression
Fix #157.  Caused by mistake in path handling refactor.
commit 524433c8 (2020-07-18): Fix #157, Micro800 regression (#158)

* updating tag count.

* Fix #157, Micro800 connection regression.

* Add Micro800 tests.

* Normalize error returns for use in automated testing.

* Updating tests.

* Added termination flag and signal handling.
commit 2e217798 (2020-07-18): Fix micro800 regression (#159)

* updating tag count.

* Fix #157, Micro800 connection regression.

* Add Micro800 tests.

* Normalize error returns for use in automated testing.

* Updating tests.

* Added termination flag and signal handling.

* Run all tests on both x64 and x86
commit 985d373a (2020-07-25): Add Modbus, fix #155, fix #157, fix #160, fix #163, fix #165 (#167)

Add Modbus.

Fix #155 .
Fix #157.
Fix #160.
Fix #163.
Fix #165.
closed 2020-07-25 by kyle-github