libplctag/libplctag #165: Using a port other than 44818 when there are multiple PLCs behind the router

Manjey73, 2020-07-23:
Good afternoon, can I use the library to access different PLCs located behind the same router ? I didn't find it possible to specify the port in addition to the IP address in the call.

kyle-github, 2020-07-23:
Not at this time.   It would not take much.  I think someone else pointed out that some PLCs do not use port 44818.  I have added this as an enhancement request.   Most of the code is in place, and in fact I have already implemented this for Modbus.

I am not sure what you mean by the same router.   You can route *through* a PLC and out one of its Ethernet modules.

kyle-github, 2020-07-24:
This is implemented in the prerelease branch with commit [2812b29](https://github.com/libplctag/libplctag/commit/2812b29c4b128f8a4c5b54766c01ac62e016d4a6).

It should be released to the release branch in the next few days.  If you can test it, I would appreciate it.

kyle-github, 2020-07-24:
The way to test is to add a port to the gateway, e.g. `gateway=10.1.2.3:44818&path=1,0&...`.

Manjey73, 2020-07-24:
UniqueKey = protocol=ab_eip&gateway=45.157.214.65:44818&cpu=SLC&elem_size=2&elem_count=7&name=N7:0

Downloaded 2.1.10 - https://github.com/libplctag/libplctag/releases/tag/v2.1.10-2
Does not work, does not get the OK status

Perhaps you were talking about some other build ? If you remove the port and enter only the IP address, it works. In the router, dstnat is configured from port 44818 to the internal IP of the PLC and also port 44818. If I change the incoming port to any other than 44818, there is no connection. I am writing a driver for a Scada system based on your library. At the moment, I have in the test the only PLC Allen Bradley MicroLogix1400, which I installed at home behind the router. I thought, what if there are two or more of them ? I changed the router settings and realized that there is no connection if other ports than the regular one are used.

Manjey73, 2020-07-24:
Sorry for my english, i using online translate

Manjey73, 2020-07-24:
Downloaded mingw and installed it. When trying to compile, it immediately throws errors. How to build a dll for Windows and did not understand.

d:\libplctag-release\src\wrappers\cpp>d:\MinGW\bin\make.exe -f Makefile
ccache g++ -Wall -O3 -std=c++11 -DBOOST_LOG_DYN_LINK -m32 -c src/main.cpp -o obj/main.o
process_begin: CreateProcess(NULL, ccache g++ -Wall -O3 -std=c++11 -DBOOST_LOG_DYN_LINK -m32 -c src/main.cpp -o obj/main.o, ...) failed.
make (e=2): Не удается найти указанный файл. (The specified file cannot be found.)
Makefile:70: recipe for target 'obj/main.o' failed
make: *** [obj/main.o] Error 2

d:\MinGW\bin>make.exe -f d:\libplctag-release\src\wrappers\cpp\Makefile
make.exe: *** No rule to make target 'obj/main.o', needed by 'main'.  Stop.

kyle-github, 2020-07-24:
Hi Manjey73,

I have not updated the "official" pre-release yet, so the binaries there do not have the fix.

The prerelease branch can be built with Microsoft Visual C++ (MSVC).   The latest versions of MSVC work directly with the CMake build files in the library.   If you are using Windows and the C library directly, I will be releasing the existing version 2.1.10 code to the release branch and making a release this weekend.

I attached the pre-release builds to this note.   Please let me know if you they work for you.

[libplctag_2.1.10_windows_x86.zip](https://github.com/libplctag/libplctag/files/4972390/libplctag_2.1.10_windows_x86.zip)
[libplctag_2.1.10_windows_x64.zip](https://github.com/libplctag/libplctag/files/4972391/libplctag_2.1.10_windows_x64.zip)
[libplctag_2.1.10_ubuntu_x86.zip](https://github.com/libplctag/libplctag/files/4972392/libplctag_2.1.10_ubuntu_x86.zip)
[libplctag_2.1.10_ubuntu_x64.zip](https://github.com/libplctag/libplctag/files/4972394/libplctag_2.1.10_ubuntu_x64.zip)
[libplctag_2.1.10_macos_x64.zip](https://github.com/libplctag/libplctag/files/4972395/libplctag_2.1.10_macos_x64.zip)

Manjey73, 2020-07-24:
I'm using Windows 10 x64, so I can check it out on Monday at work. It is not necessary so urgently, just wanted to make sure that there is a possibility to change the port. According to the description, it seems possible to build for Raspberry Pi ?, if it works, could you after checking build for Raspberry ? I have just the opportunity to check on it. Ideally, you need to enter either just IP and in this case work on the regular port 44818 as well as enter IP:port if you need to work on another port.

closed 2020-07-25 by kyle-github

kyle-github, 2020-07-25:
The new release covers this.   See release 2.1.10.

Manjey73, 2020-07-27:
Everything is fine, I checked it on release 2.1.10 (Windows x64), it works. Incoming port on router 44820, outgoing port on PLC 44818 (dstnat). Many thanks.

kyle-github, 2020-07-27:
Thanks for testing this.   Glad to hear that it worked for you!

commit 2812b29c (2020-07-24): Add port handling to AB PLCs. Fix #165. (#166)
commit 985d373a (2020-07-25): Add Modbus, fix #155, fix #157, fix #160, fix #163, fix #165 (#167)

Add Modbus.

Fix #155 .
Fix #157.
Fix #160.
Fix #163.
Fix #165.
PR #172 (2020-08-01): Prerelease
This adds more fixes for Omron PLCs and has the following fixes:

- Limit elem_count to 1 on OMRON PLCs. (#171) from @musiguello!
- platform: Fix resource and memory leaks in socket_connect_tcp (#169), thanks to @jubnzv!
- Added Omron CI check, @kyle-github.
- Fix #165, allow alternate port selection for EtherNet/IP PLCs, @kyle-github.
commit eb7949fa (2020-08-01): Prerelease (#172)

* Update a lot of debugging out put to DEBUG_INFO.  It was inconsistent.

* Add Modbus (#156).   v2.1.10

Version 2.1.10, merging in Modbus functionality. This includes:

- Initial Modbus support. Very initial!
- complete refactoring of the byte order handling.
- fixes for a memory found in socket_connect_tcp() when a host name is used. Did not leak in Windows.
- updates and warnings for deprecated tag attributes.  Just a start.
- support for getting/setting debug level using 'debug' attribute rather than 'debug_level'.

* Fix for #155, handle negative timeout.

* updating t [... cut]