libplctag/libplctag #170: No data received from omron_plc ethip

rvipin17, 2020-07-27:
Hi,

I pulled the changed under add_omron branch, and after running simple_cpp.cpp example, no data is received from the plc, shows the following output:

 data[0]=-2147483648

`#include <stdio.h>
#include <stdlib.h>
#include "../lib/libplctag.h"
#include "utils.h"

#define REQUIRED_VERSION 2,1,0

#define TAG_PATH "protocol=ab_eip&gateway=192.168.98.245&path=18,14,49,57,50,46,49,54,56,46,57,56,46,50,52,53&cpu=nx1p2&elem_size=4&elem_count=16&name=Output"
#define ELEM_COUNT 16
#define ELEM_SIZE 4
#define DATA_TIMEOUT 5000

/*
 * There really isn't much to this example other than it being compiled
 * by g++ instead of gcc.  This is just to test that the API functions are
 * correctly exported for C++ code.
 */

int main()
{
    int32_t tag = 0;
    int rc;
    int i;

    /* check the library version. */
    if(plc_tag_check_lib_version(REQUIRED_VERSION) != PLCTAG_STATUS_OK) {
        fprintf(stderr, "Required compatible library version %d.%d.%d not available!", REQUIRED_VERSION);
        exit(1);
    }

    /* create the tag */
    tag = plc_tag_create(TAG_PATH, DATA_TIMEOUT);

    /* everything OK? */
    if(tag < 0) {
        fprintf(stderr,"ERROR %s: Could not create tag!\n", plc_tag_decode_error(tag));
        return 0;
    }

    if((rc = plc_tag_status(tag)) != PLCTAG_STATUS_OK) {
        fprintf(stderr,"Error setting up tag internal state. Error %s\n", plc_tag_decode_error(rc));
        plc_tag_destroy(tag);
        return 0;
    }

    /* get the data */
    rc = plc_tag_read(tag, DATA_TIMEOUT);
    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d: %s\n",rc, plc_tag_decode_error(rc));
        plc_tag_destroy(tag);
        return 0;
    }

    /* print out the data */
    for(i=0; i < ELEM_COUNT; i++) {
        fprintf(stderr,"data[%d]=%d\n",i,plc_tag_get_int32(tag,(i*ELEM_SIZE)));
    }

    /* now test a write */
    /*for(i=0; i < ELEM_COUNT; i++) {
        int32_t val = plc_tag_get_int32(tag,(i*ELEM_SIZE));

        val = val+1;

        fprintf(stderr,"Setting element %d to %d\n",i,val);

        plc_tag_set_int32(tag,(i*ELEM_SIZE),val);
    }

    rc = plc_tag_write(tag, DATA_TIMEOUT);

    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to write the data! Got error code %d: %s\n",rc, plc_tag_decode_error(rc));
        plc_tag_destroy(tag);
        return 0;
    } */

    /* get the data again*/
    /*rc = plc_tag_read(tag, DATA_TIMEOUT);

    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d: %s\n",rc, plc_tag_decode_error(rc));
        plc_tag_destroy(tag);
        return 0;
    }*/

    /* print out the data */
    /*for(i=0; i < ELEM_COUNT; i++) {
        fprintf(stderr,"data[%d]=%d\n",i,plc_tag_get_int32(tag,(i*ELEM_SIZE)));
    }*/

    /* we are done */
    plc_tag_destroy(tag);

    return 0;
}
`

Can you share any working example for omron plc?

rvipin17, 2020-07-27:
```
2020-06-27 16:54:47.552 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=ab_eip
2020-06-27 16:54:47.552 thread(1) tag(0) INFO ab_tag_create:165 Starting.
2020-06-27 16:54:47.552 thread(2) tag(0) INFO tag_tickler_func:188 Starting.
2020-06-27 16:54:47.552 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from ab_tag_create:172
2020-06-27 16:54:47.552 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 0x2352420
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL ab_tag_create:179 tag=0x2352420
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL get_plc_type:885 Found OMRON NJ/NX Series PLC.
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL get_plc_type:885 Found OMRON NJ/NX Series PLC.
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL session_find_or_create:226 Starting
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL session_find_or_create:250 Creating new session.
2020-06-27 16:54:47.552 thread(1) tag(0) INFO session_create_unsafe:432 Starting
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL session_create_unsafe:435 Session should use connected messaging.
2020-06-27 16:54:47.552 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_unsafe:440
2020-06-27 16:54:47.552 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 0x23536f0
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL cip_encode_path:346 Starting
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL match_numeric_segment:464 Starting at position 0 in string 0,2.
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL match_numeric_segment:491 Done.   Found numeric segment 0.
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL cip_encode_path:362 Found numeric segment.
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL cip_encode_path:358 Skipping separator character ','.
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL match_numeric_segment:464 Starting at position 2 in string 0,2.
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL match_numeric_segment:491 Done.   Found numeric segment 2.
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL cip_encode_path:362 Found numeric segment.
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL cip_encode_path:403 PLC needs connection, adding path to the router object.
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL cip_encode_path:435 IOI size before 6
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL cip_encode_path:452 Done
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL add_session_unsafe:306 Starting
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL add_session_unsafe:316 Done
2020-06-27 16:54:47.552 thread(1) tag(0) INFO session_create_unsafe:539 Done
2020-06-27 16:54:47.552 thread(1) tag(0) INFO session_init:555 Starting.
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL thread_create:606 Starting.
2020-06-27 16:54:47.552 thread(1) tag(0) DETAIL thread_create:608 Warning: ignoring stacksize (32768) parameter.
2020-06-27 16:54:47.553 thread(1) tag(0) DETAIL thread_create:628 Done.
2020-06-27 16:54:47.553 thread(3) tag(0) INFO session_handler:965 Starting thread for session 0x23536f0
2020-06-27 16:54:47.553 thread(3) tag(0) DETAIL session_handler:984 in SESSION_OPEN_SOCKET state.
2020-06-27 16:54:47.553 thread(1) tag(0) INFO session_init:570 Done.
2020-06-27 16:54:47.553 thread(3) tag(0) INFO session_open_socket:588 Starting.
2020-06-27 16:54:47.553 thread(1) tag(0) DETAIL session_find_or_create:296 Done
2020-06-27 16:54:47.553 thread(3) tag(0) DETAIL socket_create:792 Starting.
2020-06-27 16:54:47.553 thread(1) tag(0) DETAIL ab_tag_create:267 using session=0x23536f0
2020-06-27 16:54:47.553 thread(1) tag(0) DETAIL ab_tag_create:354 Setting up OMRON NJ/NX Series tag.
2020-06-27 16:54:47.553 thread(3) tag(0) DETAIL socket_create:806 Done.
2020-06-27 16:54:47.553 thread(1) tag(0) DETAIL parse_symbolic_segment:908 Starting with name index=0 and encoded name index=1.
2020-06-27 16:54:47.553 thread(3) tag(0) DETAIL session_open_socket:622 Using default port 44818.
2020-06-27 16:54:47.553 thread(1) tag(0) DETAIL parse_symbolic_segment:947 Parsed symbolic segment "Output" in tag name.
2020-06-27 16:54:47.553 thread(3) tag(0) DETAIL socket_connect_tcp:825 Starting.
2020-06-27 16:54:47.553 thread(1) tag(0) INFO tag_read_common_start:256 Starting
2020-06-27 16:54:47.553 thread(1) tag(0) INFO build_read_request_connected:394 Starting.
2020-06-27 16:54:47.553 thread(3) tag(0) DETAIL socket_connect_tcp:883 Found numeric IP address: 192.168.98.245
2020-06-27 16:54:47.553 thread(1) tag(0) DETAIL session_create_request:2401 Starting.
2020-06-27 16:54:47.553 thread(3) tag(0) DETAIL socket_connect_tcp:933 Attempting to connect to 192.168.98.245
2020-06-27 16:54:47.553 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_request:2410
2020-06-27 16:54:47.553 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2020-06-27 16:54:47.553 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 0x2352b50
2020-06-27 16:54:47.553 thread(1) tag(0) DETAIL session_create_request:2424 Done.
2020-06-27 16:54:47.553 thread(1) tag(0) INFO session_add_request:898 Starting. sess=0x23536f0, req=0x2352b50
2020-06-27 16:54:47.553 thread(1) tag(0) DETAIL session_add_request_unsafe:863 Starting.
2020-06-27 16:54:47.553 thread(1) tag(0) DETAIL session_add_request_unsafe:882 Total requests in the queue: 1
[... cut ...]
2020-06-27 16:54:52.555 thread(1) tag(0) DETAIL thread_destroy:725 Done.
2020-06-27 16:54:52.555 thread(1) tag(0) INFO lib_teardown:163 Tearing down tag lookup mutex.
2020-06-27 16:54:52.555 thread(1) tag(0) INFO lib_teardown:169 Destroying tag hashtable.
2020-06-27 16:54:52.555 thread(1) tag(0) INFO hashtable_destroy:260 Starting
2020-06-27 16:54:52.555 thread(1) tag(0) INFO hashtable_destroy:272 Done
2020-06-27 16:54:52.555 thread(1) tag(0) INFO lib_teardown:176 Done.
2020-06-27 16:54:52.555 thread(1) tag(0) DETAIL plc_tag_unregister_logger:770 Starting
```

kyle-github, 2020-07-27:
Hi,  thanks for the debugging out put.  The key line is here:

```text
2020-06-27 16:54:47.582 thread(3) tag(0) WARN recv_forward_open_resp:2237 Forward Open command failed, response code: Port Not Available (1)
```

That means that your path is not correct.   Your path looks like you are trying to use Ethernet bridging.   Can you first try with a direct connection to the PLC?   

@musiguello, can you check this path and usage?

kyle-github, 2020-07-27:
It may be easier to debug this with the tag_rw tool.  Then you do not have to work to compile it every time you change the tag string.

musiguello, 2020-07-27:
Try using path=18,192.168.98.245

rvipin17, 2020-07-28:
Hi @musiguello 

I tried changing the path, 

My code:

```
#include <stdio.h>
#include <stdlib.h>
#include "../lib/libplctag.h"
#include "utils.h"

#define REQUIRED_VERSION 2,1,0

#define TAG_PATH "protocol=ab_eip&gateway=192.168.98.245&path=18,192.168.98.245&cpu=nx1p2&elem_size=4&elem_count=16&name=Output&debug=4"
#define ELEM_COUNT 16
#define ELEM_SIZE 4
#define DATA_TIMEOUT 5000

/*
 * There really isn't much to this example other than it being compiled
 * by g++ instead of gcc.  This is just to test that the API functions are
 * correctly exported for C++ code.
 */

int main()
{
    int32_t tag = 0;
    int rc;
    int i;

    /* check the library version. */
    if(plc_tag_check_lib_version(REQUIRED_VERSION) != PLCTAG_STATUS_OK) {
        fprintf(stderr, "Required compatible library version %d.%d.%d not available!", REQUIRED_VERSION);
        exit(1);
    }

    /* create the tag */
    tag = plc_tag_create(TAG_PATH, DATA_TIMEOUT);

    /* everything OK? */
    if(tag < 0) {
        fprintf(stderr,"ERROR %s: Could not create tag!\n", plc_tag_decode_error(tag));
        return 0;
    }

    if((rc = plc_tag_status(tag)) != PLCTAG_STATUS_OK) {
        fprintf(stderr,"Error setting up tag internal state. Error %s\n", plc_tag_decode_error(rc));
        plc_tag_destroy(tag);
        return 0;
    }

    /* get the data */
    rc = plc_tag_read(tag, DATA_TIMEOUT);
    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d: %s\n",rc, plc_tag_decode_error(rc));
        plc_tag_destroy(tag);
        return 0;
    }

    /* print out the data */
    for(i=0; i < ELEM_COUNT; i++) {
        fprintf(stderr,"data[%d]=%d\n",i,plc_tag_get_int32(tag,(i*ELEM_SIZE)));
    }

    /* now test a write */
    /*for(i=0; i < ELEM_COUNT; i++) {
        int32_t val = plc_tag_get_int32(tag,(i*ELEM_SIZE));

        val = val+1;
[... 154 lines cut ...]
2020-06-28 15:21:35.770 thread(1) tag(0) WARN thread_destroy:717 null thread pointer.
2020-06-28 15:21:35.770 thread(1) tag(0) INFO ab_teardown:148 Freeing session information.
2020-06-28 15:21:35.770 thread(1) tag(0) INFO ab_teardown:154 Done.
2020-06-28 15:21:35.770 thread(1) tag(0) INFO mb_teardown:1886 Starting.
2020-06-28 15:21:35.770 thread(1) tag(0) DETAIL mb_teardown:1890 Destroying Modbus mutex.
2020-06-28 15:21:35.770 thread(1) tag(0) INFO mb_teardown:1896 Done.
2020-06-28 15:21:35.770 thread(1) tag(0) INFO lib_teardown:151 Tearing down library.
2020-06-28 15:21:35.770 thread(1) tag(0) INFO lib_teardown:156 Tearing down tag tickler thread.
2020-06-28 15:21:35.770 thread(1) tag(0) DETAIL thread_join:673 Starting.
2020-06-28 15:21:35.771 thread(2) tag(0) INFO tag_tickler_func:254 Terminating.
2020-06-28 15:21:35.771 thread(1) tag(0) DETAIL thread_join:685 Done.
2020-06-28 15:21:35.771 thread(1) tag(0) DETAIL thread_destroy:714 Starting.
2020-06-28 15:21:35.771 thread(1) tag(0) DETAIL thread_destroy:725 Done.
2020-06-28 15:21:35.771 thread(1) tag(0) INFO lib_teardown:163 Tearing down tag lookup mutex.
2020-06-28 15:21:35.771 thread(1) tag(0) INFO lib_teardown:169 Destroying tag hashtable.
2020-06-28 15:21:35.771 thread(1) tag(0) INFO hashtable_destroy:260 Starting
2020-06-28 15:21:35.771 thread(1) tag(0) INFO hashtable_destroy:272 Done
2020-06-28 15:21:35.771 thread(1) tag(0) INFO lib_teardown:176 Done.
2020-06-28 15:21:35.771 thread(1) tag(0) DETAIL plc_tag_unregister_logger:770 Starting
```
Is there anything that I'm missing?

@kyle-github tag_rw uses python wrapper which is not supporting the cpu type for omron.

`python tag_rw.py -t uint32 -p "protocol=ab_eip&gateway=192.168.98.245&path=18,192.168.98.245&cpu=LGX&elem_size=4&elem_count=16&name=Output&debug=4"`

rvipin17, 2020-07-28:
The two blockers that I could see in the debug:

`2020-06-28 15:30:16.285 thread(1) tag(0) WARN match_numeric_segment:479 Numeric segment in path at position 0 is out of bounds!`

```
30:16.343 thread(3) tag(0) WARN recv_forward_open_resp:2237 Forward Open command failed, response code: Port Not Available (1)
2020-06-28 15:30:16.343 thread(3) tag(0) WARN recv_forward_open_resp:2258 CIP error code 1!
2020-06-28 15:30:16.343 thread(3) tag(0) INFO recv_forward_open_resp:2274 Done.
2020-06-28 15:30:16.343 thread(3) tag(0) WARN try_forward_open_ex:1986 Unable to use ForwardOpen response!
2020-06-28 15:30:16.343 thread(3) tag(0) DETAIL rc_dec_impl:242 Calling cleanup functions due to call at try_forward_open_ex:1992 for 0x7ffa90000b60.
```

Is this creating the  issue?

musiguello, 2020-07-28:
The second one seems the cause, the first one also appears to me.

That code should work and I would say it's something in the plc settings, I'll test tomorrow on a NX1P2.

rvipin17, 2020-07-28:
@musiguello Thank you so much for helping me out. Meanwhile, I will also look from my end.

kyle-github, 2020-07-28:
Hi @rvipin17,

The tag_rw executable (not the Python version) is either in the ZIP file you can download or if you build it it is in `<build dir>/bind_dist`.   Use the binary tag_rw as it has no dependencies other than the C library.   This minimizes the number of parts we need to debug!

In the second log output, I see the lines:

```text
2020-06-28 15:21:25.658 thread(3) tag(0) DETAIL socket_connect_tcp:933 Attempting to connect to 192.168.98.245
2020-06-28 15:21:30.658 thread(1) tag(0) WARN plc_tag_create:566 Timeout waiting for tag to be ready!
```

Are you sure that 19.2.168.98.245 has a PLC listening on port 44818?  From the lines above, five seconds passes between the time that the attempt to connect is made and the timeout.   Nothing else happens in between.  A TCP connection should be very fast.   Just a few milliseconds at most.   

What does your network look like?   Is there anything between your PC and your PLC?   Any kind of router or NAT gateway?

@musiguello, do you need to have a path when connecting to an Omron PLC?   The path that @rvipin17 is using is a bridging path normally used when you go through a network module in a ControlLogix chassis to another network.   However, in a normal bridging situation, you would have a path from the first network module to the second one.   Suppose that the second network module was in slot 3 (counting from zero).   Then the path would be `1,3,18,192.168.98.245`.   Maybe this is normal with an Omron PLC?

musiguello, 2020-07-29:
@kyle-github,

The PLC seems to always require a path. I have my computer directly connected to the built-in ethernet/ip port of a NX1P2 PLC (without switch).

@rvipin17 

Your problem seems to be of connectivity with the PLC.

I've tested your code and works for me removing elem_size and elem_count from the path. The PLC response returns the correct size for the array (64 in this case) and the internal tag data is resized properly.

`
#include <stdio.h>
#include <stdlib.h>
#include <libplctag/libplctag.h>

#define REQUIRED_VERSION 2,1,0

#define TAG_PATH "protocol=ab_eip&gateway=172.16.1.50&path=18,172.16.1.50&cpu=nx1p2&name=Output&debug=4"
#define ELEM_COUNT 16
#define ELEM_SIZE 4
#define DATA_TIMEOUT 5000

/*
 * There really isn't much to this example other than it being compiled
 * by g++ instead of gcc.  This is just to test that the API functions are
 * correctly exported for C++ code.
 */

int main()
{
    int32_t tag = 0;
    int rc;
    int i;

    /* check the library version. */
    if(plc_tag_check_lib_version(REQUIRED_VERSION) != PLCTAG_STATUS_OK) {
        fprintf(stderr, "Required compatible library version %d.%d.%d not available!", REQUIRED_VERSION);
        exit(1);
    }

    /* create the tag */
    tag = plc_tag_create(TAG_PATH, DATA_TIMEOUT);

    /* everything OK? */
    if(tag < 0) {
        fprintf(stderr,"ERROR %s: Could not create tag!\n", plc_tag_decode_error(tag));
        return 0;
    }

    if((rc = plc_tag_status(tag)) != PLCTAG_STATUS_OK) {
        fprintf(stderr,"Error setting up tag internal state. Error %s\n", plc_tag_decode_error(rc));
        plc_tag_destroy(tag);
        return 0;
    }

    /* get the data */
    rc = plc_tag_read(tag, DATA_TIMEOUT);
    if(rc != PLCTAG_STATUS_OK) {
        fprintf(stderr,"ERROR: Unable to read the data! Got error code %d: %s\n",rc, plc_tag_decode_error(rc));
        plc_tag_destroy(tag);
        return 0;
    }

    /* print out the data */
    for(i=0; i < ELEM_COUNT; i++) {
        fprintf(stderr,"data[%d]=%d\n",i,plc_tag_get_int32(tag,(i*ELEM_SIZE)));
    }

    /* now test a write */
    for(i=0; i < ELEM_COUNT; i++) {
[... 593 lines cut ...]
2389-06-29 14:12:08.302 thread(1) tag(0) INFO ab_teardown:140 Terminating IO thread.
2389-06-29 14:12:08.302 thread(1) tag(0) INFO ab_teardown:148 Freeing session information.
2389-06-29 14:12:08.302 thread(1) tag(0) INFO ab_teardown:154 Done.
2389-06-29 14:12:08.303 thread(1) tag(0) INFO mb_teardown:1886 Starting.
2389-06-29 14:12:08.303 thread(1) tag(0) DETAIL mb_teardown:1890 Destroying Modbus mutex.
2389-06-29 14:12:08.303 thread(1) tag(0) INFO mb_teardown:1896 Done.
2389-06-29 14:12:08.304 thread(1) tag(0) INFO lib_teardown:151 Tearing down library.
2389-06-29 14:12:08.304 thread(1) tag(0) INFO lib_teardown:156 Tearing down tag tickler thread.
2389-06-29 14:12:08.304 thread(1) tag(0) INFO lib_teardown:163 Tearing down tag lookup mutex.
2389-06-29 14:12:08.304 thread(1) tag(0) INFO lib_teardown:169 Destroying tag hashtable.
2389-06-29 14:12:08.305 thread(1) tag(0) INFO hashtable_destroy:260 Starting
2389-06-29 14:12:08.305 thread(1) tag(0) INFO hashtable_destroy:272 Done
2389-06-29 14:12:08.305 thread(1) tag(0) INFO lib_teardown:176 Done.
2389-06-29 14:12:08.305 thread(1) tag(0) DETAIL plc_tag_unregister_logger:770 Starting
2389-06-29 14:12:08.306 thread(1) tag(0) INFO ab_teardown:138 Releasing global AB protocol resources.
2389-06-29 14:12:08.306 thread(1) tag(0) INFO ab_teardown:140 Terminating IO thread.
2389-06-29 14:12:08.306 thread(1) tag(0) INFO ab_teardown:148 Freeing session information.
2389-06-29 14:12:08.307 thread(1) tag(0) INFO ab_teardown:154 Done.
2389-06-29 14:12:08.307 thread(1) tag(0) INFO mb_teardown:1886 Starting.
2389-06-29 14:12:08.307 thread(1) tag(0) DETAIL mb_teardown:1890 Destroying Modbus mutex.
2389-06-29 14:12:08.307 thread(1) tag(0) INFO mb_teardown:1896 Done.
2389-06-29 14:12:08.308 thread(1) tag(0) INFO lib_teardown:151 Tearing down library.
2389-06-29 14:12:08.308 thread(1) tag(0) INFO lib_teardown:176 Done.
2389-06-29 14:12:08.308 thread(1) tag(0) DETAIL plc_tag_unregister_logger:770 Starting
`

kyle-github, 2020-07-29:
Thanks, @musiguello!

Does this work with tag_rw for you?

It looks like the Omron behaves differently than a ControlLogix.   The amount of data returned is not what I expected.   With a ControlLogix, the PLC will return just a single DINT (assuming that is what the array is made of).   Your example above looks as if the Omron PLC is returning the entire tag.  I wish this was what ControlLogix PLCs did!

musiguello, 2020-07-29:
Hi @kyle-github 

yes, it works:

```
Library version 2.1.11.
Arg[0]=tag_rw
Arg[1]=-t
Arg[2]=sint32
Arg[3]=-p
Arg[4]=protocol=ab_eip&gateway=172.16.1.50&path=18,172.16.1.50&cpu=nx1p2&name=Output
Arg[5]=-d
Arg[6]=2
2389-06-29 16:33:06.615 thread(1) tag(0) WARN match_numeric_segment:479 Numeric segment in path at position 0 is out of bounds!
data[0]=1 (1)
data[1]=1 (1)
data[2]=1 (1)
data[3]=1 (1)
data[4]=1 (1)
data[5]=1 (1)
data[6]=1 (1)
data[7]=1 (1)
data[8]=1 (1)
data[9]=1 (1)
data[10]=1 (1)
data[11]=1 (1)
data[12]=1 (1)
data[13]=1 (1)
data[14]=1 (1)
data[15]=1 (1)
Done
```

which is the best place to document specific PLC behaviours? strings for example are different, they use a uint16 type as length.

rvipin17, 2020-07-29:
@musiguello I will try with the changes you have made. It could be the switch connected to the plc causing the issue. Let me try connecting it directly to the plc. Also, I was using Array(0-15) DINT in my plc and mode as "Output", is there anything that is missing from plc end.

musiguello, 2020-07-29:
@rvipin17 I've used the same configuration for the tag.

rvipin17, 2020-07-30:
@musiguello I tried running with the changes, which binary library are you referring to, I think I must be using the wrong library. Can you share the repository details and which libplctag library to be linked in the header? 
Currently, I'm linking the libplctag.h available in the folder /src/lib.
I think wrong library is being linked in the header.

musiguello, 2020-07-30:
@rvipin17 OMRON changes are in the prerelease branch (2.1.11).

kyle-github, 2020-07-30:
The header does not link the library.   That is going to be up to your wrapper (if any) or your build system.   As @musiguello said, the latest Omron updates are in 2.1.11.   The prerelease branch of that (pre-built) is [here](https://github.com/libplctag/libplctag/releases/tag/v2.1.11-pre1).

If you are using Linux, then it is important to make sure that your ldconfig search path will find the right version of the library when your executable starts.  Alternatively, you can use the static library (libplctag.a) and make a static executable.

Hope that helps!

kyle-github, 2020-08-01:
I have merged all the latest Omron fixes from @musiguello into the release branch.   This has been released as version [2.1.11](https://github.com/libplctag/libplctag/releases/tag/v2.1.11).

kyle-github, 2020-08-09:
I updated the Omron support code.   Please try with version [2.1.14](https://github.com/libplctag/libplctag/releases/tag/v2.1.14)!