libplctag/libplctag #183: AB protocol allows dealing with single tag only

GitHubDragonFly, 2020-08-19:
Currently, I have to remove a tag before creating another tag and it's only 1 tag that can be used at the time.

Modbus protocol doesn't have that limitation, it allows creating a list of tags without removing tags.

Is it possible to restore that functionality as it was present in v2.1.6?

kyle-github, 2020-08-21:
Hi, thanks for using the library and reporting this problem.

Can you please describe:

- which OS and compiler you are using?
- which type of PLC you are reading?
- which version of the library you are using?
- how you are getting the library binaries?   Are you building these or are you getting them from one of the releases?
- which, if any, wrapper language you are using?

Thanks.  These things will help us determine the cause of the problem!

GitHubDragonFly, 2020-08-21:
What I stated is applicable to MicroLogix 1100 and v2.1.14 library binary from your release.

I just tried ControlLogix and it did allow creating a tag list, without removing tags, as well as access those from another thread.

Both of the above did work with v2.1.6 library binary from your release, including read/write access from another thread.

Modbus protocol, also from v2.1.14, does allow creating a tag list without removing tags but it just ignored read/write access from another thread.

It is Windows based app that I am using (Windows 10, Visual Studio 2019, modified C# wrapper from Mesta - updated to match some of the features from your latest library).

I am just testing your library and there is no machinery of any kind involved.

On a side note, from my tests and related to your code and some previous topics you had:
- Strings in MicroLogix 1100 are 84 bytes long and have bytes reversed (which might be the case for PLC5 as well)
- Custom length ControlLogix strings should be minimum 8 bytes in size, even if defined as a single character string, and the multiples of 8 afterwards and depending on the custom length So, the elem_size parameter should require user input (by knowing the length of the string as well as the number of bytes it fits in).