libplctag/libplctag #188: Scheduled multi tag read operations results in timeout for all but the last two tags.

chrisdemario, 2020-08-24:
I'm experiencing issues with multi-tag scheduled IO reads using the modbus protocol. Only the last 2 scheduled tag reads complete the IO operation. 

See debug below. I created 3 tags, scheduled 3 reads, only the last 2 complete. If I attempt to read 5 tags, only the last 2 complete.  Any insight into this issue is much appreciated.

**### Environment:**
Client app running on Linux, Ubuntu 15.04 connected to B&R PLC modbus simulator. 
libplctag version: V2.1.15, Same results with V2.1.11

**### Test case source code:**
```
void testMultiTagScheduledReads()
{
  std::vector<int> handles;
  std::vector<std::string> tags;
  std::string tagPath;

  tags.push_back("protocol=modbus-tcp&gateway=192.168.194.183:502&path=192.168.194.183&elem_count=1&name=hr0&debug=2");
  tags.push_back("protocol=modbus-tcp&gateway=192.168.194.183:502&path=192.168.194.183&elem_count=2&name=hr90&debug=2");
  tags.push_back("protocol=modbus-tcp&gateway=192.168.194.183:502&path=192.168.194.183&elem_count=2&name=hr96&debug=2");
  // tags.push_back("protocol=modbus-tcp&gateway=192.168.194.183:502&path=192.168.194.183&elem_count=2&name=hr104&debug=3");
  // tags.push_back("protocol=modbus-tcp&gateway=192.168.194.183:502&path=192.168.194.183&elem_count=2&name=hr94&debug=3");

  for(auto aTag : tags) {
    int tagHandle = plc_tag_create(aTag.c_str(), 5000);

    if(tagHandle < 0) {
      printf("plc_tag_create() Failed for TagPath(%s)\n", aTag.c_str());
    } else {
      printf("plc_tag_create() TagHandle(%d) TagPath(%s)\n", tagHandle, aTag.c_str());
      handles.push_back(tagHandle);
    }
  }

  for(auto handle : handles) {
    int status = plc_tag_read(handle, 0);

    if(status != PLCTAG_STATUS_OK && status != PLCTAG_STATUS_PENDING) {
      printf("plc_tag_read Failed for handle(%d) status(%d)\n", handle, status);
    }

    printf("plc_tag_read scheduleIo for handle(%d) status(%d)\n", handle, status);
  }

  msleep(5000);

  for(auto handle : handles) {
    int status = plc_tag_status(handle);

    if(status != PLCTAG_STATUS_OK) {
      printf("plc_tag_read Failed for handle(%d) status(%d)\n", handle, status);
    } else {
      printf("plc_tag_read Success for handle(%d) status(%d)\n", handle, status);
    }
  }
}
```

### **Output:**
plc_tag_create() TagHandle(11) TagPath(protocol=modbus-tcp&gateway=192.168.194.183:502&path=192.168.194.183&elem_count=1&name=hr0&debug=2)
plc_tag_create() TagHandle(12) TagPath(protocol=modbus-tcp&gateway=192.168.194.183:502&path=192.168.194.183&elem_count=2&name=hr90&debug=2)
plc_tag_create() TagHandle(13) TagPath(protocol=modbus-tcp&gateway=192.168.194.183:502&path=192.168.194.183&elem_count=2&name=hr96&debug=2)
plc_tag_read scheduleIo for handle(11) status(1)
plc_tag_read scheduleIo for handle(12) status(1)
plc_tag_read scheduleIo for handle(13) status(1)
plc_tag_read Failed for handle(11) status(1)
plc_tag_read Success for handle(12) status(0)
plc_tag_read Success for handle(13) status(0)

**### Debug info:**
2020-08-24 12:50:09.203 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=modbus-tcp
[... 70 lines cut ...]
2020-08-24 12:50:09.412 thread(1) tag(13) INFO plc_tag_create:731 Returning mapped tag ID 13
2020-08-24 12:50:09.412 thread(1) tag(13) INFO plc_tag_create:733 Done.
2020-08-24 12:50:09.412 thread(1) tag(11) INFO plc_tag_read:1131 Starting.
2020-08-24 12:50:09.412 thread(1) tag(11) INFO plc_tag_read:1264 Done
2020-08-24 12:50:09.412 thread(1) tag(12) INFO plc_tag_read:1131 Starting.
2020-08-24 12:50:09.412 thread(1) tag(12) INFO plc_tag_read:1264 Done
plc_tag_create() TagHandle(13) TagPath(protocol=modbus-tcp&gateway=192.168.194.183:502&path=192.168.2020-08-24 12:50:09.412 thread(1) tag(13) INFO plc_tag_read:1131 Starting.
2020-08-24 12:50:09.412 thread(1) tag(13) INFO plc_tag_read:1264 Done
194.183&elem_count=2&name=hr96&debug=3)
plc_tag_read scheduleIo for handle(11) status(1)
plc_tag_read scheduleIo for handle(12) status(1)
plc_tag_read scheduleIo for handle(13) status(1)
2020-08-24 12:50:09.413 thread(3) tag(13) INFO create_read_request:1193 Starting.
2020-08-24 12:50:09.413 thread(3) tag(13) INFO create_read_request:1205 preparing read request for 2 registers (of 2 total) from base register 96.
2020-08-24 12:50:09.414 thread(3) tag(12) INFO create_read_request:1193 Starting.
2020-08-24 12:50:09.414 thread(3) tag(12) INFO create_read_request:1205 preparing read request for 2 registers (of 2 total) from base register 90.
2020-08-24 12:50:09.415 thread(3) tag(11) INFO create_read_request:1193 Starting.
2020-08-24 12:50:09.415 thread(3) tag(11) INFO create_read_request:1205 preparing read request for 1 registers (of 1 total) from base register 0.
2020-08-24 12:50:09.458 thread(3) tag(13) INFO check_read_response:1108 Starting.
2020-08-24 12:50:09.458 thread(3) tag(13) INFO check_read_response:1125 Got read response 4 of length 13 with payload of size 4.
2020-08-24 12:50:09.502 thread(3) tag(12) INFO check_read_response:1108 Starting.
2020-08-24 12:50:09.502 thread(3) tag(12) INFO check_read_response:1125 Got read response 5 of length 13 with payload of size 4.
**plc_tag_read Failed for handle(11) status(1)**
plc_tag_read Success for handle(12) status(0)
plc_tag_read Success for handle(13) status(0)

kyle-github, 2020-08-25:
Hi @chrisdemario, thanks for reporting this and providing so much information!  Having the core code is particularly helpful.

I see a couple of things that I am not sure I understand in your code.   The first is that the tag string looks like it is somewhat malformed.   The "path" attribute is the device/unit ID, not the IP address again.   I should be checking the value on that but it looks like I am not doing a careful job of catching format errors!   

The second is the delay.  How long is "msleep(5000)"?   If that is 5000 milliseconds, then the library will possibly close the PLC connection before you check the status.  In the output it looks like each response is taking about 50ms.

I also run against a Modbus simulator (though not a full PLC emulator) and I modified my async_stress.c example program to use 5 tags.  I used the tag string ''protocol=modbus-tcp&gateway=127.0.0.1:5020&path=1&elem_count=4&name=ir4&debug=4''.   That code does not have a specific timeout, but I print out the total time taken (or fail if there is an error).   In that, I see nothing fail and nothing takes too long.

In the output from your program, it looks like the tags that succeed are the first two on which reads are triggered and the third one is still pending (status 0 is OK, status 1 is PENDING, negative statuses are errors).

Would it be possible to run this test again, but set the debug level to 4?   That will show me the packets and when they are sent to and received from the PLC.

From what I can see, it appears that the delay is too short or something is blocking a response from the PLC for the third tag.   If you can get the expanded debug output, I may be able to see if anything is happening within the library during that wait time.

I do not know if it would help, but the example code in data_dumper.c takes an input file with tags (tag names and tag strings and read periods) and reads all the tags then outputs the data at the requested intervals into files.

chrisdemario, 2020-08-25:
@kyle-github, thank you for looking into this issue. I changed the test case to include the following:

- print lib version
- set debug level to PLCTAG_DEBUG_DETAIL
- one second delay before checking for read completion
- updated the tag "path" attribute to device/unit ID

The output shows the first tag(11) still pending while the last two tags(12 and 13) indicate success.
Note, if I create five tags, schedule reads on the the five tags, only the last 2 succeed.  
For some reason, it seems the check_read_response() call for tag 11 is missing. 

**### Updated source:**
```
void testMultiTagScheduledReads()
{
  std::vector<int> handles;
  std::vector<std::string> tags;
  std::string tagPath;

  plc_tag_set_debug_level(PLCTAG_DEBUG_DETAIL);

  int version_major = plc_tag_get_int_attribute(0, "version_major", 0);
  int version_minor = plc_tag_get_int_attribute(0, "version_minor", 0);
  int version_patch = plc_tag_get_int_attribute(0, "version_patch", 0);
  printf("plctag version(%d.%d.%d)\n", version_major, version_minor, version_patch);

  tags.push_back("protocol=modbus-tcp&gateway=192.168.194.183:502&path=1,0&elem_count=2&name=hr90&debug=4");
  tags.push_back("protocol=modbus-tcp&gateway=192.168.194.183:502&path=1,0&elem_count=2&name=hr92&debug=4");
  tags.push_back("protocol=modbus-tcp&gateway=192.168.194.183:502&path=1,0&elem_count=2&name=hr94&debug=4");

  // Create tags
  for(auto aTag : tags) {
    int tagHandle = plc_tag_create(aTag.c_str(), 1000);

    if(tagHandle < 0) {
      printf("plc_tag_create() Failed for tagPath(%s)\n", aTag.c_str());
    } else {
      printf("plc_tag_create() tagHandle(%d) tagPath(%s)\n", tagHandle, aTag.c_str());
      handles.push_back(tagHandle);
    }
  }

  // Schedule reads for all tags
  for(auto tagHandle : handles) {
    int status = plc_tag_read(tagHandle, 0);

    if(status != PLCTAG_STATUS_OK && status != PLCTAG_STATUS_PENDING) {
      printf("plc_tag_read() Failed for tagHandle(%d) status(%d)\n", tagHandle, status);
    }

    printf("plc_tag_read() scheduleIo for tagHandle(%d) status(%d)\n", tagHandle, status);
  }

  printf("Wait 1 second for I/O completion...\n");

  msleep(1000);

  // Check read completion status. Shoule be done after 1 second
  for(auto tagHandle : handles) {
    int status = plc_tag_status(tagHandle);

    if(status != PLCTAG_STATUS_OK) {
      printf("plc_tag_read() Failed for tagHandle(%d) status(%d)\n", tagHandle, status);
    } else {
      printf("plc_tag_read() Success for tagHandle(%d) status(%d)\n", tagHandle, status);
    }
  }
}
```
**### Output:**
[... 283 lines cut ...]
2020-08-25 09:27:04.185 thread(3) tag(13) INFO check_read_response:1108 Starting.
2020-08-25 09:27:04.185 thread(3) tag(13) INFO check_read_response:1125 Got read response 4 of length 13 with payload of size 4.
2020-08-25 09:27:04.185 thread(3) tag(13) DETAIL check_read_response:1126 registers_per_request = 125
2020-08-25 09:27:04.185 thread(3) tag(13) DETAIL check_read_response:1127 register_offset = 0
2020-08-25 09:27:04.185 thread(3) tag(13) DETAIL check_read_response:1128 byte_offset = 0
2020-08-25 09:27:04.185 thread(3) tag(13) DETAIL check_read_response:1129 copy_size = 4
2020-08-25 09:27:04.185 thread(3) tag(13) DETAIL check_read_response:1140 Read is complete.
2020-08-25 09:27:04.185 thread(3) tag(13) DETAIL check_read_response:1179 Done.
2020-08-25 09:27:04.229 thread(3) tag(0) DETAIL read_packet:883 data_needed=7, packet_size=7, read_data_len=6
2020-08-25 09:27:04.229 thread(3) tag(0) DETAIL read_packet:883 data_needed=0, packet_size=7, read_data_len=13
2020-08-25 09:27:04.229 thread(3) tag(0) DETAIL read_packet:908 Received full packet.
2020-08-25 09:27:04.229 thread(3) tag(0) DETAIL read_packet:909  00 05 00 00 00 07 01 03 04 00
2020-08-25 09:27:04.229 thread(3) tag(0) DETAIL read_packet:909  00 00 00
2020-08-25 09:27:04.229 thread(3) tag(12) INFO check_read_response:1108 Starting.
2020-08-25 09:27:04.229 thread(3) tag(12) INFO check_read_response:1125 Got read response 5 of length 13 with payload of size 4.
2020-08-25 09:27:04.229 thread(3) tag(12) DETAIL check_read_response:1126 registers_per_request = 125
2020-08-25 09:27:04.229 thread(3) tag(12) DETAIL check_read_response:1127 register_offset = 0
2020-08-25 09:27:04.229 thread(3) tag(12) DETAIL check_read_response:1128 byte_offset = 0
2020-08-25 09:27:04.230 thread(3) tag(12) DETAIL check_read_response:1129 copy_size = 4
2020-08-25 09:27:04.230 thread(3) tag(12) DETAIL check_read_response:1140 Read is complete.
2020-08-25 09:27:04.230 thread(3) tag(12) DETAIL check_read_response:1179 Done.
plc_tag_read() Failed for tagHandle(11) status(1)
plc_tag_read() Success for tagHandle(12) status(0)
plc_tag_read() Success for tagHandle(13) status(0)
2020-08-25 09:27:09.229 thread(3) tag(0) DETAIL modbus_plc_handler:709 Shutting down socket due to inactivity.

kyle-github, 2020-08-26:
Hmm, I think I see what is happening here.

It looks like I have made assumptions about how the PLC will react when more requests than it expects are sent.   I can see that the requests are sent for all three tags.   Then I see responses come back for two of them and then... nothing.   So I think I am overrunning the internal buffer of the PLC.

It looks like this is a bug in the library at this point for Modbus.   I will need to figure out how to make sure that only one request is in flight at any one time.   I don't think that will take too long but it might be this weekend before I have something I can test.

Thanks for testing and finding this!

chrisdemario, 2020-08-26:
@kyle-github Thanks for working on this. 

Here's a screenshot of a wireshark trace using a Modbus Slave simulator. Note the two read requests in one packet followed by
the response packets. All tags complete without issue. 

[image]

I then ran the same test on a B&R PLC simulator. The two read requests made it into one packet as above but only one response 
made it back. Apparently the Modbus protocol does not support batched read requests?

Not sure why it works on the Modbus simulator but not on the PLC simulator. I was counting on using batched reads to reduce latency.

[image]

kyle-github, 2020-08-26:
I found something late last night that says that the protocol definition is such that you should not pipeline (batch) requests.   I must have missed that before when reading through the specs.  Oh well, I went to some effort to try to batch assuming that the TCP connection itself would buffer them if needed.

I believe that many PLCs support some commands that allow batching to some degree.  However, with only a 256 byte packet size, you are not going to get much back at a time.

First get it right, then make it fast.  I skipped a step.

kyle-github, 2020-08-26:
Now I see conflicting comments.   A couple of threads on some Siemens forums state that you must wait for a response before sending another request on a specific TCP connection (regardless of whether the unit/device ID is the same).   That's coming from Siemens representatives.   However, some docs from Acromag say that you can send multiple requests.  

I suspect that this differs per PLC.   I will implement some interlocking to make sure that only one request is in flight at a time.   That will get something stable.

I think this will work for your situation.   When you create the tags, you use a timeout on each plc_tag_create() call so it is blocking.   This means that effectively you only have one tag request in flight at a time.   Note that when you create a tag, the tag is read implicitly.   I do this so that if you immediately do a write once the tag creation is finished, you only change the values you specifically have changed.  Otherwise, you would write all zeros (except the values you changed).   The tag internally has a status of PLCTAG_STATUS_PENDING until that read completes.

chrisdemario, 2020-08-26:
My main app has over 140 tags and makes use of non-blocking calls to plc_tag_create() at startup.  It currently communicates with a Rockwell PLC using eip. It works quite well with low latency.

I was planning to use the same code for init/read/write to communicate with the B&R PLC via modbus. I have a map that converts tags to holding register numbers.

I used blocking plc_tag_create() calls in the test case to simplify the code. 
Do you kick off reads on plc_tag_create() for the ab_eip protocol or just modbus-tcp?

kyle-github, 2020-08-26:
I kick off initial reads for both AB and Modbus.   Early use of the library showed that users (myself included!) tended to forget that the state of a tag in the program was not the state of the tag in the PLC without this.   With the implicit initial read, the state of the tag starts off with values from the PLC.

I see some hints that there may be some common protocol extensions to Modbus that will allow multiple requests in a single packet.   However, that is definitely PLC-specific.   I may need to eventually create some way to determine which commands a PLC supports, or at least allow the user to specify them.

Each protocol is going to take some tuning.   Modbus is so simple compared to EIP/CIP that the only real challenge will be managing the different commands supported by different types of PLC.

I tried a new test with a modified form of my async_stress.c program on my Modbus simulator.   It worked fine sending multiple requests before any responses were returned :-/   I will probably need to write my own simulator that intentionally does not handle this so that I can catch such problems.

chrisdemario, 2020-08-26:
@kyle-github, let me know when you have a release ready, I will test it on my B&R sim and the B&R PLC.

kyle-github, 2020-08-27:
Actually I may have something to test.   I created a new branch, fix_188, with my attempts at fixing the Modbus overrun problem.   There are some other pending tweaks in the code as well to cover some edge cases I found this evening.

kyle-github, 2020-08-27:
If you can checkout that branch and test, I would appreciate it!

chrisdemario, 2020-08-27:
It works! I ran the test on the B&R modbus simulator. Tomorrow I'll run it on the B&R PLC.

Here's the output:
plctag version(2.1.16)
2020-08-26 23:08:54.479 thread(1) tag(0) INFO plc_tag_create:559 Starting
2020-08-26 23:08:54.479 thread(1) tag(0) INFO initialize_modules:176 Starting.
2020-08-26 23:08:54.479 thread(1) tag(0) INFO initialize_modules:204 Initialized library modules.
2020-08-26 23:08:54.479 thread(1) tag(0) INFO lib_init:122 Setting up global library data.
2020-08-26 23:08:54.479 thread(1) tag(0) INFO lib_init:124 Creating tag hashtable.
2020-08-26 23:08:54.479 thread(1) tag(0) INFO hashtable_create:75 Starting
2020-08-26 23:08:54.479 thread(1) tag(0) INFO hashtable_create:99 Done
2020-08-26 23:08:54.479 thread(1) tag(0) INFO lib_init:130 Creating tag hashtable mutex.
2020-08-26 23:08:54.479 thread(1) tag(0) INFO lib_init:136 Creating tag tickler thread.
2020-08-26 23:08:54.479 thread(1) tag(0) DETAIL thread_create:761 Starting.
2020-08-26 23:08:54.479 thread(1) tag(0) DETAIL thread_create:763 Warning: ignoring stacksize (32768) parameter.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL thread_create:783 Done.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO lib_init:142 Done.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO initialize_modules:207 Initializing AB module.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO ab_init:121 Initializing AB protocol library.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO ab_init:128 Finished initializing AB protocol library.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO initialize_modules:212 Initializing Modbus module.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO mb_init:1928 Starting.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL mb_init:1930 Setting up mutex.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO mb_init:1939 Done.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO initialize_modules:223 Done initializing library modules.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO initialize_modules:228 Done.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=modbus-tcp
2020-08-26 23:08:54.480 thread(1) tag(0) INFO mb_tag_create:211 Starting.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO create_tag_object:260 Starting.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO parse_register_name:1594 Starting.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL parse_register_name:1618 Found holding register type.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO parse_register_name:1630 Done.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL create_tag_object:295 Tag data size is 4 bytes.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from create_tag_object:298
2020-08-26 23:08:54.480 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 0x7f8a0c00bf80
2020-08-26 23:08:54.480 thread(1) tag(0) INFO create_tag_object:317 Done.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO find_or_create_plc:523 Starting.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from find_or_create_plc:546
2020-08-26 23:08:54.480 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 0x7f8a0c00c050
2020-08-26 23:08:54.480 thread(1) tag(0) INFO find_or_create_plc:569 Creating new PLC.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL thread_create:761 Starting.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL thread_create:763 Warning: ignoring stacksize (32768) parameter.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL thread_create:783 Done.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO find_or_create_plc:593 Done.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO set_tag_byte_order:329 Starting.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL set_tag_byte_order:335 int16_byte_order=10
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL check_byte_order_str:426 Starting.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL check_byte_order_str:459 Done.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL set_tag_byte_order:349 int32_byte_order=3210
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL check_byte_order_str:426 Starting.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL check_byte_order_str:459 Done.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL set_tag_byte_order:364 int64_byte_order=76543210
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL check_byte_order_str:426 Starting.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL check_byte_order_str:459 Done.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL set_tag_byte_order:383 float32_byte_order=3210
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL check_byte_order_str:426 Starting.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL check_byte_order_str:459 Done.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL set_tag_byte_order:398 float64_byte_order=76543210
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL check_byte_order_str:426 Starting.
2020-08-26 23:08:54.480 thread(1) tag(0) DETAIL check_byte_order_str:459 Done.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO set_tag_byte_order:415 Done.
2020-08-26 23:08:54.480 thread(1) tag(0) INFO mb_tag_create:244 Done.
2020-08-26 23:08:54.480 thread(2) tag(0) INFO tag_tickler_func:189 Starting.
2020-08-26 23:08:54.480 thread(3) tag(0) INFO modbus_plc_handler:666 Starting.
2020-08-26 23:08:54.480 thread(3) tag(0) DETAIL connect_plc:819 Starting.
[... cut ...]
2020-08-26 23:08:54.819 thread(3) tag(11) DETAIL check_read_response:1155 copy_size = 4
2020-08-26 23:08:54.819 thread(3) tag(11) DETAIL check_read_response:1166 Read is complete.
2020-08-26 23:08:54.819 thread(3) tag(11) DETAIL check_read_response:1205 Done.
plc_tag_read() Success for tagHandle(11) status(0)
plc_tag_read() Success for tagHandle(12) status(0)
plc_tag_read() Success for tagHandle(13) status(0)
Exit testMultiTagScheduledReads

2020-08-26 23:08:59.819 thread(3) tag(0) DETAIL modbus_plc_handler:710 Shutting down socket due to inactivity.

kyle-github, 2020-08-27:
Great to hear!  Let me know if it works equally well with the real PLC.   I am learning to distrust the Modbus emulators :-/

chrisdemario, 2020-08-27:
I tested this release on an APC 2200 B&R PLC. I did not run into any issues. Thanks!

closed 2020-08-28 by kyle-github

kyle-github, 2020-08-28:
Thanks for finding this and testing!   I released this in version 2.1.16.

commit 14f2246a (2020-08-28): Modbus fix - Fix #188, Modbus can only have one request in flight at a time (#192)

* Fix 188 - Modbus can only have one request in flight at a time per TCP connection. (#190) - thanks to @chrisdemario for finding this!

* Bump version for Modbus fixes.

* Fixed RC bug. Cleanup after TCP shutdown.

There was a bug in the ref count handling when an existing PLC was reused in Modbus.   The rc_inc() call was not in place.