libplctag/libplctag #191: v2.1.15 no longer reports an error on invalid CIP path while accessors return INT_MIN

kpreisser, 2020-08-27:
Hi,
we are using libplctag in one our products and it works great.

However, after updating from v2.0.27 to v2.1.15, we found that when using an invalid CIP path, no error is reported, but the accessors (like `plc_tag_set_uint32`) return `INT_MIN`.

For example, when I run a C# program on Windows x64 with the following code (which contains `DllImports` for the native functions from `plctag.dll`) to connect to a AB CompactLogix 5370 L1 PLC and read a Int32 tag:
```c#
    int tag = plc_tag_create("protocol=ab_eip&gateway=192.168.0.71&cpu=logix&name=MyDINTCounter1&elem_size=4&elem_count=1&path=1, 0", 2000);
    Console.WriteLine("Tag: " + tag);

    int status = plc_tag_read(tag, 20000);
    Console.WriteLine("Status: " + status);

    int resultValue = plc_tag_get_int32(tag, 0);
    status = plc_tag_status(tag);

    Console.WriteLine("Result: " + resultValue);
    Console.WriteLine("Status: " + status);
```

with v2.0.27, it prints the following (`18576655` is the actual tag value stored in the PLC):
```
Tag: 11
Status: 0
Result: 18576655
Status: 0
```

With v2.1.15, it prints:
```
Tag: 11
Status: 0
Result: -2147483648
Status: 0
```

So it seems the space in the path is no longer supported. When I change `path=1, 0` to `path=1,0`, the results are as expected (it prints the same as v2.0.27).

However, with the incorrect path, `plc_tag_read` as well as `plc_tag_status` return `0` even though `plc_tag_get_int32` returned `INT_MIN`, in which case I expected that `plc_tag_status` would return a negative value.

For example, when I use `path=1,Z` instead, v2.0.27 actually returns status `PLCTAG_ERR_NOT_FOUND` (-19):
```
Tag: -11
Status: -19
Result: -2147483648
Status: -19
```
whereas v2.1.15 prints the same as when using the space in the path: 
```
Tag: 11
Status: 0
Result: -2147483648
Status: 0
```

Is this expected? Because when the accessor returns `INT_MIN` (`-2147483648`) but `plc_tag_status` returns `0`, we would interpret this as the actual value that is successfully read from the PLC, whereas in fact an error occured.

Thank you!

kyle-github, 2020-08-28:
Hi @kpreisser, thanks for reporting this.   There was a change to the code that processes paths somewhat recently and I suspect that it is not handling some of these errors as well.

I will try to replicate this here and see if I can figure out what is happening with your last example.   Thanks for the detailed descriptions of what is happening.   Definitely there is something wrong if it is accepting "1,Z" as a path!

kyle-github, 2020-08-29:
I think I have a fix for this.   It was not what I expected.   From what I can see, the changed routines I mentioned above are working correctly.  The routines that they replaced and the old logic was not correct.   It returned a tag that was set up but without a session (PLC connection for EIP/CIP).   The new logic correctly fails to create a session and thus the tag's methods (C-based vtable) are still set to the default.  The default methods returned the wrong values for status and a few other things.   

This works when I test.  I was able to recreate your results with the previous code.

I have pushed this to the tip of the prerelease branch.   I have not made a release yet.   Could you please test that to see if it returns correct values for your system?

Thanks for filing this issue on this bug!

kpreisser, 2020-08-31:
Hi @kyle-github,
thanks for fixing the issue!

I retried with the version from `prerelease` branch, and now when I run code with an invalid CIP path, `plc_tag_create` returns an error as expected (in this case, -6 (`PLCTAG_ERR_BAD_GATEWAY`)), and it works successfully when specifying a valid CIP path.

Thank you!

kyle-github, 2020-08-31:
Thanks for checking this!   I will make this a release in the next day or two.

closed 2020-09-06 by kyle-github

kyle-github, 2020-09-07:
Sorry that took longer than I had hoped.  I put this together with the fix for #195.

commit 92ec62d0 (2020-08-29): Fix #191 - incorrect status reported when path is malformed (#193)

* Default vtable functions returned the wrong status.

In the case that the session object could not be created, the default vtable is used.

In that case the default status, abort etc. methods all returned PLCTAG_STATUS_OK rather than an error.

* Bump version for fix of #191.
commit b6a15583 (2020-09-06): Release 2.1.17 - Fix #191, Fix #195 (#196)

* Fix #191 - incorrect status reported when path is malformed (#193)

* Default vtable functions returned the wrong status.

In the case that the session object could not be created, the default vtable is used.

In that case the default status, abort etc. methods all returned PLCTAG_STATUS_OK rather than an error.

* Bump version for fix of #191.

* Fixed export for plc_tag_abort()

* fixed long standing typo in LGPL license file name and renamed license files.

* Make the contact info more easy to find.

* Almost complete rewrite of async_stress.c.  [... cut]