libplctag/libplctag #194: could not create tag

leowoki, 2020-08-31:
Hi!
I used library to connect micro850 and micrologix1400, but plc_tag_create() function returned an error value. Is there anything wrong with my path:"path=protocol=ab_eip&gateway=192.168.2.11&path=1,1&cpu=micrologix800&elem_size=2&elem_count=1&name=A2, timeout=1000"?

leowoki, 2020-08-31:
> Hi!
> I used library to connect micro850 and micrologix1400, but plc_tag_create() function returned an error value. Is there anything wrong with my path:"protocol=ab_eip&gateway=192.168.2.11&path=1,1&cpu=micrologix800&elem_size=2&elem_count=1&name=A2, timeout=1000"?

GitHubDragonFly, 2020-09-01:
You should try:

1. "protocol=ab_eip&gateway=192.168.2.11&cpu=micro800&elem_size=2&elem_count=1&name=A2"
2. "protocol=ab_eip&gateway=192.168.2.10&cpu=micrologix&elem_size=2&elem_count=1&name=B3:0"

leowoki, 2020-09-01:
> You should try:
> 
> 1. "protocol=ab_eip&gateway=192.168.2.11&cpu=micro800&elem_size=2&elem_count=1&name=A2"
> 2. "protocol=ab_eip&gateway=192.168.2.10&cpu=micrologix&elem_size=2&elem_count=1&name=B3:0"

Thanks. You're right

GitHubDragonFly, 2020-09-01:
You should close this issue then.

leowoki, 2020-09-02:
> You should close this issue then.
If I want to get b3:0/15, what is the path?

GitHubDragonFly, 2020-09-02:
One of these might work:

"protocol=ab_eip&gateway=192.168.2.10&cpu=micrologix&elem_size=1&elem_count=1&name=B3:0/15"
"protocol=ab_eip&gateway=192.168.2.10&cpu=micrologix&elem_size=1&elem_count=1&name=B3:0.15"
"protocol=ab_eip&gateway=192.168.2.10&cpu=micrologix&elem_type=bool&elem_size=1&elem_count=1&name=B3:0/15"
"protocol=ab_eip&gateway=192.168.2.10&cpu=micrologix&elem_type=bool&elem_count=1&name=B3:0/15"
"protocol=ab_eip&gateway=192.168.2.10&cpu=micrologix&elem_type=bool&elem_size=1&elem_count=1&name=B3:0.15"
"protocol=ab_eip&gateway=192.168.2.10&cpu=micrologix&elem_type=bool&elem_count=1&name=B3:0.15"

leowoki, 2020-09-02:
> One of these might work:
>
> "protocol=ab_eip&gateway=192.168.2.10&cpu=micrologix&elem_size=1&elem_count=1&name=B3:0/15" [...]

None of these paths can be successfully created... o(╥﹏╥)o

GitHubDragonFly, 2020-09-02:
Then try using the original path for B3:0 and then just extract the bit with plc_tag_get_bit(tag, bitIndex)), where bitIndex should be 15.

kyle-github, 2020-09-03:
Hi, sorry for the delay.   Was buried with work.

The library currently only supports 16-bit values (N and B data file entries), 32-bit values (F and L on some PLCs) and strings on PCCC PLCs such as PLC/5, SLC and MicroLogix.   You can also read some aggregate data files like counters and timers.  Single bits are not yet supported for PCCC-based PLCs.

As @GitHubDragonFly said, right now the best way to do this is to read in the whole 16-bit word and then extract out the bit you want with the plc_tag_get/set_bit() functions.

I recently added support for single bit tags to the CIP PLC side of the library, so it would be possible to see if I can port that code over to PCCC PLCs.   It would be lower priority than ensuring that you can read whole data files at once for PCCC PLCs.

leowoki, 2020-09-03:
> Then try using the original path for B3:0 and then just extract the bit with plc_tag_get_bit(tag, bitIndex)), where bitIndex should be 15.

Thank you. According to your method, I have successfully read/write a single bit

leowoki, 2020-09-03:
> Hi, sorry for the delay. Was buried with work.
>
> The library currently only supports 16-bit values (N and B data file entries), 32-bit values (F and L on some PLCs) and strings on PCCC PLCs such as [...]

Thank you for your reply. I used his @GitHubDragonFly method to achieve the function. I'm looking forward to the future version supporting single bit reading and writing.

GitHubDragonFly, 2020-09-03:
As a temporary fix, if you have availability in your B3 table then just move the bit to non-critical word of the table, where you can write without thinking of affecting other bits.

There is PLC-5 Bit Write command/function of 0F/02 that could possibly allow independent bit writing, but I'm not sure if it's implemented in the libplctag library and whether it would work for MicroLogix PLCs as well.

leowoki, 2020-09-06:
> As a temporary fix, if you have availability in your B3 table then just move the bit to non-critical word of the table, where you can write without thinking of affecting other bits.
>
> There is PLC-5 [...]

I'm sorry to have been delayed these days. This is a good proposal, but it needs PLC program cooperation, which is not under my control.

kyle-github, 2020-09-06:
BTW, please note issue #82.

There is single bit tag support for CIP-based PLCs, but not for PCCC-based PLCs.   That has not been a priority.   I will see if I can find any solid documentation for the Read-Modify-Write command for PCCC and whether it is fully supported across PLC/5, SLC and MicroLogix systems.   Unfortunately, AB did not support the same commands across all of those PLCs so there is a big split between PLC/5 and SLC+MicroLogix.

kyle-github, 2020-09-26:
As far as I can tell, the remaining problem is covered by issue #82.   If there is still another question, please reopen this issue or open another issue.   Thanks!

closed 2020-09-26 by kyle-github