libplctag/libplctag #195: Modbus doesn't recover after loss of communication

GitHubDragonFly, 2020-09-03:
As tested on Android, the same code structure is used for polling AB (MicroLogix 1100) and Modbus (simulator).

After loss of communication, Modbus did re-establish the connection, was polling the device but no values were returned.

After loss of communication, AB did re-establish the connection, was polling the device and values were returned.

I could blame either Android or the simulator but that wouldn't explain why AB was working.

kyle-github, 2020-09-04:
Hmm.   That is not good. Thanks for bringing this up.  I will investigate as soon as I can.  I will mark this as a bug until it is fixed.

GitHubDragonFly, 2020-09-05:
Just some additional info for Modbus, after re-establishing the connection it did start sending requests but it was painfully slow.

kyle-github, 2020-09-06:
Thanks for the update.  I am modifying the async_stress.c test to try to reproduce this.

How are you causing communications to fail?   I am going to kill the Modbus simulator and restart it.

kyle-github, 2020-09-06:
I was able to reproduce this.   It was due to an incomplete set of code that handled drop outs.  A flag was reset, but not a buffer size and so it keep returning the same response packet over and over without ever clearing it.

This is fixed in 2.1.17.

closed 2020-09-06 by kyle-github

GitHubDragonFly, 2020-10-25:
Just so you know, your README.md file has an extra letter "m" on the "releases" link within "How to Contribute" paragraph.
It might just confuse people if they are to click it.
Your other "releases" link works fine (within "How to Get The Code" paragraph).

There was no need to start a new issue because of this, so I will delete this post later on when you manage to correct it.

kyle-github, 2020-10-25:
Thanks! I fixed it directly in GitHub.   I will go through the other links when I can in the morning.  Off to bed now!

commit 7116e6ba (2020-09-06): Early clearing of response ready flag caused skip of actual response clearing.   Fix #195.
commit b6a15583 (2020-09-06): Release 2.1.17 - Fix #191, Fix #195 (#196)

* Fix #191 - incorrect status reported when path is malformed (#193)

* Default vtable functions returned the wrong status.

In the case that the session object could not be created, the default vtable is used.

In that case the default status, abort etc. methods all returned PLCTAG_STATUS_OK rather than an error.

* Bump version for fix of #191.

* Fixed export for plc_tag_abort()

* fixed long standing typo in LGPL license file name and renamed license files.

* Make the contact info more easy to find.

* Almost complete rewrite of async_stress.c.  [... cut]