libplctag/libplctag #197: Accessing A file and ST file data on micrologix1400

saltratrix, 2020-09-07:
Am having trouble accessing ASCII file and String file data on micrologix1400. Just need to read these values but the syntax is unclear. 
[image]

using permutations of this as my connection string (also have tried ST52:0.0 ST52:0.0.DATA ....

#define TAG_PATH "protocol=ab-eip&gateway=PLC_IP&cpu=micrologix&elem_count=1&elem_size=30&name=ST52:0"
Can create the tag and read data back but using both simple.c and string the translation to character string is failing. thanks

saltratrix, 2020-09-07:
I was able to solve this by reading the data bitwise in reverse.

closed 2020-09-07 by saltratrix

kyle-github, 2020-09-07:
I was just going to post a response.   All the PCCC-based PLCs seem to byteswap the character data.

I am working on a new set of API functions to hide almost all of this from the end user because it is such a pain.

What do the ASCII files look like?   Do they have a count field and then the (byteswapped) character data?  I only have a PLC/5 so I cannot easily look at examples.

GitHubDragonFly, 2020-09-10:
It appears to be 2 bytes length + 82 reversed bytes of data.