libplctag/libplctag #199: Read Template Service Request (Logix 5000)

GitHubDragonFly, 2020-09-15:
If this is not already a part of the library, would it be possible to implement it?

It is described on page 55 of this document:

https://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm020_-en-p.pdf

and it allows reading the structure information.

It could supplement getting the tag list from the controller.

kyle-github, 2020-09-15:
It is not part of the library today.  That is the document I have and mentioned in the issue in the libplctag4android.  If you look on page 49 you can figure out how to decode the data type.

The tag information was able to be done as a shared responsibility.   Part of it is done in the library and part of it is done in user code.   With the data structures (UDTs and some systems tags), it is less clear how that would work as the processing is a lot trickier than for tag entries.   For instance figuring out how bit arrays are packed within structures and what fields those are.   

The other part of the problem is that this kind of structured data is really not a happy pairing with C code.

I think that some sort of shared responsibility where part of this is done in the C library and part of this is done in Java code could work.

Given the number of different CIP commands that will need to be done eventually, I may just make the C library handle nothing but a very basic directly CIP call via names that are just hex bytes or something.   

That said, there is a discussion happening in the forum right now about using the tag listing to get the tag IDs so that a different set of CIP commands can be used to read and write tags using a binary ID for the tag.    There are several hassles with this, not the least of which is the large memory overhead that in the C library.   Of course, if you are using C# or Java, the C overhead is pathetically small :-)   One of my key goals is trying to avoid any memory allocation.   I have a pending reworking of the core which will let me do everything the library does today with no allocations outside of creating a tag and about 4k of overhead per PLC.

While I do not want to complicate the C library more the possible performance advantages of embedding all of the tag listing code within it make it quite appealing.  And if I do that...

So this bears more thought.   I think it is something that could be reasonable to do as half-in, half-out of the library, certainly at first.  I'll keep looking at the docs to see if there is some reasonable point that makes sense to do in C and split the rest into Java for this because I think this is valuable functionality.   

For example, I have the special name @tags right now to get all the tags in raw form.  I could also possibly do one like @udt:<id> to get all the raw data for a UDT entry.   Then you'd have to parse all the raw data in user code, but that is at least feasible as a prototype.

GitHubDragonFly, 2020-09-15:
I am happy with your explanations and, since this was posted as a question, you can decide when to close this issue.

kyle-github, 2020-09-15:
I will leave this open as it is an enhancement to the library.

I think that it is probably better to start with some sort of new syntax that allows creation of raw CIP calls.   I might make it easier to handle CIP's class/instance values but probably just raw hex bytes at first.   This will let me prototype all kinds of things to get an idea of where the right balance is.

Something like `name=@cip:03,03,20,ac,25,00,...` and then you get the raw CIP response back.

kyle-github, 2020-09-24:
I have expanded issue #200 to cover this.  Closing this as it is part of that larger enhancement.   You need all three parts explained there to get full use of this feature.

closed 2020-09-24 by kyle-github