libplctag/libplctag #201: could not create tag when the elem_count greater than 124?

eslinan, 2020-09-18:
protocol=ab_eip&gateway=192.168.127.221&path=1,0&cpu=LGX&elem_size=4&elem_count=600&name=SV_STACKS_INFO

kyle-github, 2020-09-18:
Hi, thanks for using the library.   

First, a few questions so that I can narrow down the source of the problem.  What version are you using?   What OS and compiler?  What model of PLC are you attempting to read?

With the latest prerelease, I get:

```
./tag_rw -t sint32 -p 'protocol=ab-eip&gateway=10.206.1.39&path=1,0&plc=ControlLogix&elem_size=4&elem_count=1000&name=TestBigArray'
Library version 2.1.18.
Arg[0]=./tag_rw
Arg[1]=-t
Arg[2]=sint32
Arg[3]=-p
Arg[4]=protocol=ab-eip&gateway=10.206.1.39&path=1,0&plc=ControlLogix&elem_size=4&elem_count=1000&name=TestBigArray
data[0]=0 (0)
data[1]=0 (0)
data[2]=0 (0)
data[3]=0 (0)
data[4]=0 (0)
data[5]=0 (0)
data[6]=0 (0)
data[7]=0 (0)
data[8]=0 (0)
data[9]=0 (0)
data[10]=0 (0)
data[11]=0 (0)
data[12]=0 (0)
...
data[993]=0 (0)
data[994]=0 (0)
data[995]=0 (0)
data[996]=0 (0)
data[997]=0 (0)
data[998]=0 (0)
data[999]=0 (0)
Done
```

At least one fairly recent release had broken large tags due to a change to support Omron PLCs, but that was fixed in release version [2.1.15](https://github.com/libplctag/libplctag/releases/tag/v2.1.15).

If you are using an earlier version, please try the latest release (2.1.17) and see if that fixes the problem!

If that is not the issue, then please run your test, but with debug turned on to level 4:

`protocol=ab_eip&gateway=192.168.127.221&path=1,0&cpu=LGX&elem_size=4&elem_count=600&name=SV_STACKS_INFO&debug=4`

Then capture the output (warning, it could be fairly long!) and attach it to this issue and I can try to see where things are going wrong.

Again, thanks for bringing this up and hopefully I can get you up and running soon!

eslinan, 2020-09-19:
thanks a lot, i used the source code from latest trunk code 2.1.9, compiled by msvc2017, but now it is ok when i used 2.1.17.

kyle-github, 2020-09-19:
Glad to hear that it is working for you!

Is it OK to close this issue now?

kyle-github, 2020-09-24:
Closing this.  Please reopen if this is still a problem.  Thanks for bringing it up!

closed 2020-09-24 by kyle-github