libplctag/libplctag #204: PLCTAG_ERR_BAD_PARAM on System Tags

g3n10z, 2020-10-12:
Hi,
i'm trying to read this system tag: __SYSVA_KVBPERR as uint8 but i receive error: PLCTAG_ERR_BAD_PARAM on micro800 plc.
i can read other bool tag as uint8 without problems.
Any ideas?
thanks a lot

kyle-github, 2020-10-12:
Hi,

Thanks for using the library and bringing this issue up.   Can you run your test with debugging set to level 4 and capture the output?   Then please attach it to this issue.   Using that I can determine whether the error is reported by the PLC or whether it is being reported inside the library.

System tags are very often unusual sizes or types.   That might be the issue here.   It also might be that the library is having a problem parsing the tag name.   With the debug output I should be able to help you more.

Thanks for bringing this up.

g3n10z, 2020-10-12:
[log.txt](https://github.com/libplctag/libplctag/files/5365815/log.txt)

kyle-github, 2020-10-12:
Thanks for the fast response.   I see the following:

```text
2389-09-12 14:42:21.985 thread(1) tag(11) WARN check_read_status_connected:1110 CIP read failed with status: 0x5 Path Destination Error
2389-09-12 14:42:21.985 thread(1) tag(11) INFO check_read_status_connected:1111 The Class, Instance or Attribute value specified in the Unconnected Explicit Message request is incorrect or not supported in the Target device. Check the manufacturer's documentation for the correct codes to use.
```

So the PLC is returning an error.   It is a little hard to decipher the error, but I think it is indicating that this tag cannot be read using the protocol commands the library uses.

This is a very old version of the library: 2.0.35.   We are currently on 2.1.19.   It may be worth trying a newer version to see if that changes anything.  It might not help, but we might be lucky!   There are some bug fixes and other enhancements when running and building on Windows in later versions.   It certainly won't hurt!

From the error returned it looks like the PLC will not allow the library to read the tag in the same way as other tags.   Perhaps the PLC does not expose this tag to outside access?

The Micro800 series is quite limited in its implementation of the AB CIP protocol.   It may be that system tags like this are simply not accessible from outside the PLC or that there is some other way that they must be accessed than standard AB CIP commands.   I will look around to see if I can find any hints on the Internet.

kyle-github, 2020-10-18:
I have not found anything that mentions special behavior of system tags on Micro800.   It looks like this is simply not supported by the PLC.   If you know of other libraries that support these tags on Micro800, please let me know!

Please reopen if using the 2.1.19 version of the library changes the output.  I will add a note to the README and find a place in the wiki to mention system tags on Micro800.

closed 2020-10-18 by kyle-github