libplctag/libplctag #207: How do I read tag that contain different data types?

niuchenfei, 2020-10-16:
For example, there is a tag named A that has one BOOL type of data, two DINT type of data, three REAL type of data. I want to read all the data for that tag at once.
How do I configure this：
protocol=ab-eip&gateway=192.168.16.205&path=1,0&cpu=contrologix&elem_size=?&elem_count=?&name=A&debug=3
I tried several times but failed. Error:PLCTAG_ERR_OUT_OF_BOUNDS。
Can anyone answer that？

xinthose, 2020-10-16:
What are you talking about?  Every tag has its own unique name and a separate data type in a PLC.

GitHubDragonFly, 2020-10-17:
He was talking about structured tags like Timer, which is 3 DINTs or 12 bytes containing PRE/ACC as DINT values and EN/TT/DN as BOOL values (bits of the 3rd DINT). The same would also apply to UDT tags which can contain different data types.

It appears that he resolved this issue by using elem_size=12 and elem_count=1, which he posted on another website, so I guess this issue can be closed.

kyle-github, 2020-10-17:
@GitHubDragonFly yeah, I finally saw this and I think you are right.  Looks like a UDT or some system type.   That said, how you read the whole tag and then extract the individual fields varies depending on the type.   Things like Timer have BOOLs in explicit places (as you mention, the 3rd DINT) whereas a UDT would pack them in the first hidden field.

Generally, if you want to figure out how big a tag is, use the list_tags program and check the output.   Especially with UDTs there can be surprises.

Please reopen if you need more information!

closed 2020-10-17 by kyle-github

xinthose, 2020-10-18:
I would not recommend reading/writing tags directly to things like timers.  My PLC programmer co-workers always point the finger at my code instead of the ladder logic when things do not work right.  We always use a separate array tag for our own internal SCADA communications.

kyle-github, 2020-10-18:
+1 on this.   There are no guarantees in the protocol or in the implementations on the PLC side that prevent you from getting data "tearing."   The PLC may be in the process of updating a UDT or array when you read it.   You can read partially updated data.   Similarly when you write anything other than an atomic data type (DINT, INT, REAL), the PLC program can run before you have written all of the data into the tag's memory in the PLC.

When you are reading or writing aggregate data structures such as UDTs and arrays, you should provide for some mechanism to signal that a new value is ready.   We often use DINT sentinel values at the front and back of UDTs or arrays as this signal.   In the PLC code we update the first sentinel when we start updating the tag.   Once we are done updating it, we update the second sentinel value.   When we read the tag externally, we look a the sentinel values.  If they are the same, then we know we got a good snapshot of the tag data.  If they are not the same, then we know that the PLC was modifying values within the tag while we read it and we ignore the read and do another.