libplctag/libplctag #211: libplctag for centOS

vinuthap333, 2020-10-27:
Hello,

I am trying to use libplctag for centOS 7. But, when I run cmake, the build is not successfull. I have followed the same steps as mentioned in the readme file. But the same is working in ubuntu 20.04.Could you please help me to achieve the same in centOS 7.

Regards,
Vinutha

kyle-github, 2020-10-28:
Is this CentOS x86 or x64?

What version of CMake and the C compiler are you using?

I am installing CentOS 7 into a VM, but it will be the latest update, so I do not know if it will match your versions!

kyle-github, 2020-10-28:
I was able to get cmake to run by installing the package gcc-c++ as well as gcc and cmake.  

I used the latest version of the release branch via a git checkout.

Thanks for reporting this!  I will update the build instructions and see if there is a way to make the CMake build configuration more resilient to this.

kyle-github, 2020-11-10:
It looks like CMake really wants there to be a C++ compiler.   I updated the build instructions in the prerelease branch.  When that goes out I will close this.

kyle-github, 2020-11-22:
The BUILD.md file is now updated in the release branch.

closed 2020-11-22 by kyle-github