libplctag/libplctag #213: 1756-L83E

ibjones69, 2020-11-09:
Has anyone tried this software on a 1756-L83E. I see the L82 was tried. I have run all of the sample on this plc and it times out.
Here is the output when I try to read a tag. I know the tag is good because I can monitor and change using other software.

plc_tag_check_lib_version(): success
2020-11-09 11:37:46.627 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=ab-eip
2020-11-09 11:37:46.627 thread(2) tag(0) INFO tag_tickler_func:189 Starting.
2020-11-09 11:37:46.628 thread(1) tag(0) INFO ab_tag_create:165 Starting.
2020-11-09 11:37:46.628 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from ab_tag_create:172
2020-11-09 11:37:46.628 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2020-11-09 11:37:46.629 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 00B27478
2020-11-09 11:37:46.629 thread(1) tag(0) DETAIL ab_tag_create:178 tag=00B27478
2020-11-09 11:37:46.629 thread(1) tag(0) DETAIL get_plc_type:904 Found ControlLogix/CompactLogix PLC.
2020-11-09 11:37:46.629 thread(1) tag(0) DETAIL get_plc_type:904 Found ControlLogix/CompactLogix PLC.
2020-11-09 11:37:46.629 thread(1) tag(0) DETAIL session_find_or_create:230 Starting
2020-11-09 11:37:46.629 thread(1) tag(0) DETAIL session_find_or_create:254 Creating new session.
2020-11-09 11:37:46.629 thread(1) tag(0) INFO session_create_unsafe:446 Starting
2020-11-09 11:37:46.629 thread(1) tag(0) DETAIL session_create_unsafe:449 Session should use connected messaging.
2020-11-09 11:37:46.630 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_unsafe:454
2020-11-09 11:37:46.630 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2020-11-09 11:37:46.630 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 00B28000
2020-11-09 11:37:46.630 thread(1) tag(0) DETAIL cip_encode_path:346 Starting
2020-11-09 11:37:46.630 thread(1) tag(0) DETAIL match_numeric_segment:468 Starting at position 0 in string 2,0.
2020-11-09 11:37:46.630 thread(1) tag(0) DETAIL match_numeric_segment:495 Done.   Found numeric segment 2.
2020-11-09 11:37:46.630 thread(1) tag(0) DETAIL cip_encode_path:362 Found numeric segment.
2020-11-09 11:37:46.631 thread(1) tag(0) DETAIL cip_encode_path:358 Skipping separator character ','.
2020-11-09 11:37:46.631 thread(1) tag(0) DETAIL match_numeric_segment:468 Starting at position 2 in string 2,0.
2020-11-09 11:37:46.631 thread(1) tag(0) DETAIL match_numeric_segment:495 Done.   Found numeric segment 0.
2020-11-09 11:37:46.631 thread(1) tag(0) DETAIL cip_encode_path:362 Found numeric segment.
2020-11-09 11:37:46.631 thread(1) tag(0) DETAIL cip_encode_path:403 PLC needs connection, adding path to the router object.
2020-11-09 11:37:46.631 thread(1) tag(0) DETAIL cip_encode_path:435 IOI size before 6
2020-11-09 11:37:46.631 thread(1) tag(0) DETAIL cip_encode_path:456 Done
2020-11-09 11:37:46.631 thread(1) tag(0) DETAIL add_session_unsafe:310 Starting
2020-11-09 11:37:46.631 thread(1) tag(0) DETAIL add_session_unsafe:320 Done
2020-11-09 11:37:46.632 thread(1) tag(0) INFO session_create_unsafe:553 Done
2020-11-09 11:37:46.633 thread(1) tag(0) INFO session_init:569 Starting.
2020-11-09 11:37:46.633 thread(1) tag(0) DETAIL mutex_create:632 Starting.
2020-11-09 11:37:46.634 thread(1) tag(0) DETAIL mutex_create:659 Done.
2020-11-09 11:37:46.634 thread(1) tag(0) DETAIL thread_create:790 Starting.
2020-11-09 11:37:46.636 thread(1) tag(0) DETAIL thread_create:824 Done.
2020-11-09 11:37:46.638 thread(1) tag(0) INFO session_init:584 Done.
2020-11-09 11:37:46.639 thread(3) tag(0) INFO session_handler:979 Starting thread for session 00B28000
2020-11-09 11:37:46.639 thread(1) tag(0) DETAIL session_find_or_create:300 Done
2020-11-09 11:37:46.640 thread(3) tag(0) DETAIL session_handler:998 in SESSION_OPEN_SOCKET state.
2020-11-09 11:37:46.641 thread(1) tag(0) DETAIL ab_tag_create:266 using session=00B28000
2020-11-09 11:37:46.641 thread(3) tag(0) INFO session_open_socket:602 Starting.
2020-11-09 11:37:46.642 thread(1) tag(0) DETAIL ab_tag_create:317 Setting up Logix tag.
2020-11-09 11:37:46.643 thread(3) tag(0) DETAIL socket_create:1013 Starting.
2020-11-09 11:37:46.643 thread(1) tag(0) DETAIL setup_tag_listing:2389 Starting.
2020-11-09 11:37:46.645 thread(1) tag(0) INFO setup_tag_listing:2479 Tag is not a tag listing request.
2020-11-09 11:37:46.645 thread(1) tag(0) DETAIL get_tag_data_type:614 Done.
2020-11-09 11:37:46.646 thread(1) tag(0) DETAIL parse_symbolic_segment:912 Starting with name index=0 and encoded name index=1.
2020-11-09 11:37:46.647 thread(1) tag(0) DETAIL parse_symbolic_segment:951 Parsed symbolic segment "TZ1_JobModel" in tag name.
2020-11-09 11:37:46.648 thread(1) tag(0) DETAIL parse_numeric_segment:962 Starting with name index=13 and encoded name index=15.
2020-11-09 11:37:46.648 thread(1) tag(0) DETAIL parse_numeric_segment:1022 Parsed 1-byte numeric segment of value 1.
2020-11-09 11:37:46.649 thread(1) tag(0) DETAIL parse_numeric_segment:1025 Done with name index=14 and encoded name index=17.
2020-11-09 11:37:46.649 thread(3) tag(0) DETAIL socket_create:1032 Done.
2020-11-09 11:37:46.649 thread(1) tag(0) INFO tag_read_start:239 Starting
2020-11-09 11:37:46.650 thread(3) tag(0) DETAIL session_open_socket:636 Using default port 44818.
2020-11-09 11:37:46.650 thread(1) tag(0) INFO build_read_request_connected:357 Starting.
2020-11-09 11:37:46.651 thread(3) tag(0) DETAIL socket_connect_tcp:1052 Starting.
2020-11-09 11:37:46.651 thread(1) tag(0) DETAIL session_create_request:2334 Starting.
2020-11-09 11:37:46.652 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_request:2343
2020-11-09 11:37:46.653 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2020-11-09 11:37:46.654 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 00B30020
2020-11-09 11:37:46.654 thread(1) tag(0) DETAIL session_create_request:2357 Done.
2020-11-09 11:37:46.655 thread(1) tag(0) INFO session_add_request:912 Starting. sess=00B28000, req=00B30020
2020-11-09 11:37:46.655 thread(1) tag(0) DETAIL session_add_request_unsafe:877 Starting.
2020-11-09 11:37:46.655 thread(3) tag(0) DETAIL socket_connect_tcp:1101 Found numeric IP address: 192.168.200.241
2020-11-09 11:37:46.656 thread(1) tag(0) DETAIL session_add_request_unsafe:896 Total requests in the queue: 1
2020-11-09 11:37:46.658 thread(1) tag(0) DETAIL session_add_request_unsafe:898 Done.
[... 414 lines cut ...]
2020-11-09 11:38:16.681 thread(1) tag(0) INFO refcount_cleanup:256 Starting
2020-11-09 11:38:16.681 thread(1) tag(0) INFO session_destroy:773 Starting.
2020-11-09 11:38:16.682 thread(1) tag(0) DETAIL remove_session:371 Starting.
2020-11-09 11:38:16.682 thread(1) tag(0) DETAIL remove_session_unsafe:347 Starting
2020-11-09 11:38:16.683 thread(1) tag(0) DETAIL remove_session_unsafe:362 Done
2020-11-09 11:38:16.685 thread(1) tag(0) DETAIL remove_session:379 Done.
2020-11-09 11:38:16.685 thread(1) tag(0) INFO session_destroy:784 Session sent 12 packets.
2020-11-09 11:38:16.686 thread(1) tag(0) DETAIL session_destroy:790 Destroying session thread.
2020-11-09 11:38:16.687 thread(3) tag(0) DETAIL session_handler:1216 Critical block.
2020-11-09 11:38:16.687 thread(1) tag(0) INFO session_unregister:741 Starting.
2020-11-09 11:38:16.687 thread(1) tag(0) INFO session_unregister:745 Done.
2020-11-09 11:38:16.688 thread(1) tag(0) DETAIL session_destroy:838 Destroying session mutex.
2020-11-09 11:38:16.688 thread(1) tag(0) DETAIL mutex_destroy:747 destroying mutex 00B29050
2020-11-09 11:38:16.688 thread(1) tag(0) DETAIL mutex_destroy:760 Done.
2020-11-09 11:38:16.689 thread(1) tag(0) DETAIL session_destroy:844 Cleaning up allocated memory for paths and host name.
2020-11-09 11:38:16.689 thread(1) tag(0) INFO session_destroy:860 Done.
2020-11-09 11:38:16.690 thread(1) tag(0) INFO refcount_cleanup:268 Done.
2020-11-09 11:38:16.690 thread(1) tag(0) DETAIL mutex_destroy:747 destroying mutex 00B2748C
2020-11-09 11:38:16.690 thread(1) tag(0) DETAIL mutex_destroy:760 Done.
2020-11-09 11:38:16.691 thread(1) tag(0) DETAIL mutex_destroy:747 destroying mutex 00B27490
2020-11-09 11:38:16.691 thread(1) tag(0) DETAIL mutex_destroy:760 Done.
2020-11-09 11:38:16.692 thread(1) tag(0) INFO ab_tag_destroy:837 Finished releasing all tag resources.
2020-11-09 11:38:16.692 thread(1) tag(0) INFO ab_tag_destroy:839 done
2020-11-09 11:38:16.693 thread(1) tag(0) INFO refcount_cleanup:268 Done.
ERROR PLCTAG_ERR_TIMEOUT: Could not create tag!

xinthose, 2020-11-09:
I have libplctag successfully on the following: 

- CompactLogix 1769-L35E
- CompactLogix 1769-L30ER
- CompactLogix 1769-L24ER
- CompactLogix 1769-L18ER

kyle-github, 2020-11-10:
Hi,

Thanks for filing this with the debugging output.  That is very helpful!   It looks like your tag string is not correct.  The path looks suspicious to me:

```text
...
2020-11-09 11:37:46.630 thread(1) tag(0) DETAIL cip_encode_path:346 Starting
2020-11-09 11:37:46.630 thread(1) tag(0) DETAIL match_numeric_segment:468 Starting at position 0 in string 2,0.
...
```
So your path element is 2,0.   The first number is the port out which the connection will go.   Port 1 is the backplane.  The second number is the node/address/slot of the target.  In this case, you are telling the PLC to connect to the module at address/slot zero.

The process of connecting to this kind of PLC goes through a few steps:

1. open a TCP connection to the PLC.  This succeeds.
2. set up a CIP connection (CIP is tunneled over EIP which is layered on top of TCP).   This uses a Forward Open request.  This fails:

``` text
2020-11-09 11:37:46.676 thread(3) tag(0) DETAIL session_handler:1030 in SESSION_SEND_FORWARD_OPEN state.
2020-11-09 11:37:46.676 thread(3) tag(0) INFO send_forward_open_request:1942 Starting
2020-11-09 11:37:46.677 thread(3) tag(0) DETAIL send_forward_open_request:1952 Setting initial payload size guess to 4002.
2020-11-09 11:37:46.677 thread(3) tag(0) INFO send_extended_forward_open_request:2068 Starting
2020-11-09 11:37:46.677 thread(3) tag(0) INFO send_eip_request:1763 Starting.
2020-11-09 11:37:46.678 thread(3) tag(0) INFO send_eip_request:1776 Sending packet of size 92
2020-11-09 11:37:46.680 thread(3) tag(0) INFO send_eip_request:1777 6f 00 44 00 00 01 00 09 00 00
2020-11-09 11:37:46.680 thread(3) tag(0) INFO send_eip_request:1777 00 00 dd 66 00 00 00 00 00 00
2020-11-09 11:37:46.681 thread(3) tag(0) INFO send_eip_request:1777 00 00 00 00 00 00 00 00 01 00
2020-11-09 11:37:46.681 thread(3) tag(0) INFO send_eip_request:1777 02 00 00 00 00 00 b2 00 34 00
2020-11-09 11:37:46.682 thread(3) tag(0) INFO send_eip_request:1777 5b 02 20 06 24 01 0a 05 00 00
2020-11-09 11:37:46.682 thread(3) tag(0) INFO send_eip_request:1777 00 00 99 50 00 00 c7 12 3d f3
2020-11-09 11:37:46.682 thread(3) tag(0) INFO send_eip_request:1777 45 43 50 21 01 00 00 00 40 42
2020-11-09 11:37:46.683 thread(3) tag(0) INFO send_eip_request:1777 0f 00 a2 0f 00 42 40 42 0f 00
2020-11-09 11:37:46.683 thread(3) tag(0) INFO send_eip_request:1777 a2 0f 00 42 a3 03 02 00 20 02
2020-11-09 11:37:46.684 thread(3) tag(0) INFO send_eip_request:1777 24 01
2020-11-09 11:37:46.684 thread(3) tag(0) INFO send_eip_request:1811 Done.
2020-11-09 11:37:46.685 thread(3) tag(0) INFO send_extended_forward_open_request:2127 Done
2020-11-09 11:37:46.685 thread(3) tag(0) INFO send_forward_open_request:1969 Done
2020-11-09 11:37:46.686 thread(3) tag(0) DETAIL session_handler:1036 Send Forward Open succeeded, going to SESSION_RECEIVE_FORWARD_OPEN state.
2020-11-09 11:37:46.686 thread(3) tag(0) DETAIL session_handler:1042 in SESSION_RECEIVE_FORWARD_OPEN state.
2020-11-09 11:37:46.687 thread(3) tag(0) INFO receive_forward_open_response:2140 Starting
2020-11-09 11:37:46.687 thread(3) tag(0) INFO recv_eip_response:1831 Starting.
2020-11-09 11:37:46.690 thread(3) tag(0) INFO recv_eip_response:1895 request received all needed data (54 bytes of 54).
2020-11-09 11:37:46.690 thread(3) tag(0) INFO recv_eip_response:1897 6f 00 1e 00 00 01 00 09 00 00
2020-11-09 11:37:46.690 thread(3) tag(0) INFO recv_eip_response:1897 00 00 dd 66 00 00 00 00 00 00
2020-11-09 11:37:46.691 thread(3) tag(0) INFO recv_eip_response:1897 00 00 00 00 00 00 00 00 00 00
2020-11-09 11:37:46.691 thread(3) tag(0) INFO recv_eip_response:1897 02 00 00 00 00 00 b2 00 0e 00
2020-11-09 11:37:46.692 thread(3) tag(0) INFO recv_eip_response:1897 db 00 04 00 c7 12 3d f3 45 43
2020-11-09 11:37:46.692 thread(3) tag(0) INFO recv_eip_response:1897 50 21 03 00
2020-11-09 11:37:46.693 thread(3) tag(0) INFO recv_eip_response:1904 Done.
2020-11-09 11:37:46.694 thread(3) tag(0) WARN receive_forward_open_response:2164 Forward Open command failed, response code: IOI could not be deciphered or tag does not exist. (4)
2020-11-09 11:37:46.695 thread(3) tag(0) WARN receive_forward_open_response:2189 CIP error code IOI could not be deciphered or tag does not exist. (The path segment identifier or the segment syntax was not understood by the target device.)!
```
Here we see that the PLC is sending back an error (4).   This error means that the PLC cannot understand the path or tag name.   Since this is occurring in the CIP connection set up, it is the path that is the most likely problem.

If your CPU module is in slot 0, then you want to use path 1,0 rather than 2,0.

One of my test systems is a 1756 L81.  I have multiple CPU modules and multiple network modules in a single chassis so that I can test out different paths.  In my case the L81 is in slot 4, so my path is 1,4 for that.  I also have an old L55 CPU in slot zero, so that path is 1,0.

Even on the built-in network connection on the L81 CPU module, the path is still 1,4.

Hopefully that helps.  Please let me know if it works!

ibjones69, 2020-11-10:
Thank you for the quick response. You were correct this was the problem. I was able to read a write tags and I also ran the list_tags example and this worked as well. 

Model: 1756-L83E      
Firmware Version 32.014

kyle-github, 2020-11-10:
Glad to hear it worked!

closed 2020-11-10 by kyle-github