libplctag/libplctag #217: High CPU usage on low-end devices

heppth, 2020-11-26:
First of all I would like to compliment you for this first class Lib. It is very easy to use. I could connect my Micro820 very quickly. Great work!

I usually work with embedded devices with small CPUs. Here I notice that the CPU usage is quite high. As far as I could analyze it so far, the lib opens 2 threads. It seems to be irrelevant how many tags are created. These 2 threads consume >25% CPU on low end devices.

Unfortunately I lacked the time to analyze the source code deeper. Are there any parameters which lead to a reduction of the CPU load?

Something else: I am always interested in keeping the software's footprint quite small. On some embedded devices there is only 1-2 MB memory available. Have you considered to make the lib smaller with build parameters? For example, we could exclude the modbus functionality from the build if it is not needed.

kyle-github, 2020-11-26:
Hi @heppth, thanks for using the library.

There are other users who have wondered about reducing the CPU load and about reducing the memory footprint.

There are potentially several threads created:

- one for "tickling" all tags.   This enables the auto-sync functionality.
- one per PLC.   I found out the hard way that there are failure cases in DNS resolution that can block this thread for up to 90 seconds.  I had a few choices: 1) implement non-blocking DNS, 2) have at least a thread per PLC, 3) start a thread for blocking operations.   I considered 3 at length, but starting a thread has a lot of overhead.

So I feel that the complexity is at a local minimum now with 1+X threads where X is the number of PLCs.

The code uses calls to `sleep_ms()` to give up the CPU for a period of time.  On POSIX systems, this is based on calling select() with a timeout.   There is another user who was asking the same kinds of questions recently on the [forum](https://groups.google.com/g/libplctag).  There I suggested increasing the delays from 1ms to 5ms.   The balance here is that increasing the delays too much results in lower throughput/performance.

On the use of memory, there is already an open issue about embedded systems, see issue #68.  There are two aspects to this.   The first is the binary size.  The second is the amount of memory used at runtime.  

There are a few things I can try for binary size:

1. Add a compilation flag to remove all the debugging code.   There is a _lot_ of that and all those strings and calls to `printf()` take up space.   However, this would eliminate all the ability to get debug output.
2. Add a compilation flag to compile for small size.   In my experience this helps, but not a huge amount.
3. (as you noted) make some parts of the library optional.

Runtime is more complicated.   I experimented with a different method of buffer allocation and use in Modbus; there is only _one_ buffer at any time and no allocation at all during normal use.   I am pleased with that result.  However, AB protocol use results in an allocation per request.   That allocation is carefully tracked and freed, but it is done and so having a lot of requests in flight will result in significantly more memory use.   Unfortunately rewriting the AB code to use the same pattern as Modbus is not a trivial undertaking.

One approach to the runtime memory cost would be to rewrite the AB code alongside the existing code rather than replacing it and make it optional.

Unfortunately I do not have any embedded hardware on which to test.   I gave away my Raspberry Pi, but even that had multiple cores and multiple GB of RAM.  Can you suggest any free emulators that run under Linux or Windows?

I will keep this issue open and immediately look into at least some compilation flags to reduce the binary size and remove the debugging code and increase the timeout values.   Due to the lack of real or emulated hardware I will not be able to test.  Are you willing/able to ~~be a guinea pig~~ help test?

Thanks for bringing this up.  I will see what I can do as noted above.

heppth, 2020-11-26:
Wow, thanks for the detailed answer. 
**Threading:** I think your approach is totally fine. 
**sleep_ms():** Probably the easiest solution would be to make the number of milliseconds via compile time variables overwritable.
**binary size:** Your suggestions sound very good. This would surely improve things a lot.
**runtime memory:** I didn't see any problems here. Also on embedded devices I see no problem here.

I am willing to test changes on my embedded devices. No problem!

kyle-github, 2020-11-29:
I am tracking work here in the [forum thread](https://groups.google.com/g/libplctag/c/lFlpUvZZdgE).

kyle-github, 2021-01-08:
Still working on this.   The approach is going to be:

- Build out a core event-driven flow.
- Bring up PCCC-on-EIP PLCs first as that is a good forcing function.
- Add the Modbus to exercise a different stack.
- Add the other Rockwell PLC types one at a time.
- Automate more of my local testing so that I can exercise this as much as possible to check for performance, memory leaks etc.

I am still on the first step with a prototype of the second step, but I have hit the knee in the curve.   I am still doing rewrites of the new event-logic, but each time I am covering more edge cases and the code is getting smaller.

kyle-github, 2021-12-04:
I am going to close this a fixed.  If you see any regressions, please let me know and reopen it!

closed 2021-12-04 by kyle-github