libplctag/libplctag #218: Add ARM as a CI target

timyhac, 2020-11-29:
As discussed on https://github.com/libplctag/libplctag.NET/issues/111, using libplctag on ARM devices is becoming increasingly popular. This issue requests that runtimes are produced for ARM.

Github Actions does not have free hosted ARM runners, but cross-compiling might be possible.

kyle-github, 2020-11-29:
Yep.   Now a high priority.   I will start w/Raspian/Debian since I think I can build that under Ubuntu.   After that, I'll try to see if I can build ARM and ARM64 under Windows.   

First, solve the CPU load problem...

timyhac, 2021-02-20:
I don't have the full answer yet, but I have been able to cross-compile for ARM 32bit ubuntu running on Raspberry Pi 3. The compilation happened on x64bit Ubuntu running in a Virtual Machine on my Windows 10 laptop.

I don't know at this stage how this would fit into CMakeLists.txt, so I just plonked the following on line 80 (after all other commands to configure the build tools).

```
set(CMAKE_SYSTEM_PROCESSOR arm)
set(CMAKE_C_COMPILER arm-linux-gnueabihf-gcc)
set(CMAKE_CXX_COMPILER arm-linux-gnueabihf-g++)
```

The relevant build tools need to be installed:
```sudo apt-get install g++-arm-linux-gnueabihf```

Relevant information:
* https://askubuntu.com/questions/250696/how-to-cross-compile-for-arm
* https://cmake.org/cmake/help/latest/manual/cmake-toolchains.7.html#cross-compiling-for-linux

kyle-github, 2021-02-20:
I found another slightly higher-level way to do it and added that into the CI process.   It should be building for Arm v6hf (hard float), Arm 7l (no idea what the "L" is for, little endian?), and Aarch64 now.  I also started building for Windows for Arm 32-bit and Arm 64-bit.

Instead of including the compiler itself, Debian and Ubuntu have another package that pulls in all the basics:

```text
sudo apt install crossbuild-essential-armhf crossbuild-essential-arm64
```
That gets the compiler and all the base libraries and include files.   The C libraries builds like a charm with that.   

I updated the CMake file to take a new optional parameter, `CROSS_BUILD_TYPE`, that lets you select between `Linux-Arm6`, `Linux-Arm7`, and `Linux-Aarch64`.   One thing I still need to work on is that if you run _once_ with the cross build type set, another run will result in building with the host platform tools, not the cross tools.   You need to remove your build directory and start over to get a new build type to work.   

Thanks for all the work on this!   I made another release, v2.3.3, with the new CI process and the new binaries.

closed 2021-02-20 by kyle-github

timyhac, 2021-02-21:
Nice work @kyle-github! I've added these to libplctag.NativeImport (v1.0.8).